clash-lang/clash-compiler #351: Clash generates invalid Verilog when an output port is named "result"

adamwalker, 2018-09-11:
Given the code below:
```
import Clash.Prelude

{-# ANN topEntity
  (Synthesize
    { t_name     = "bug"
    , t_inputs   = [
        ]
    , t_output   = PortProduct "" [
            PortName "result",
            PortName "out2"
        ]
    }) #-}
topEntity :: (Signal System Bool, Signal System Bool)
topEntity = (pure True, pure True)
```
Clash 0.9.3 generates the following Verilog with an output signal and internal wire both named "result":
```
module bug
    ( // No inputs

      // Outputs
      output wire  result 
    , output wire  out2 
    );
  wire [1:0] result;

  assign result = {1'b1,1'b1};

  assign result = result[1:1];

  assign out2 = result[0:0];
endmodule
```

martijnbastiaan, 2018-09-20:
Thanks for the report! I've fixed this in https://github.com/clash-lang/clash-compiler/pull/360, which will hopefully soon land in master.

martijnbastiaan, 2018-09-25:
(Renamed commit message to conform to project policy.)

commit d5b34f02 (2018-09-25): Account for topentity annotation when generating port names. Fix #351

Users can specify custom port names for their top entity. Clash did not
account for this possibility when generating unique argument / result
names thus generating name collisions whenever some commonly used
variable name was used as a port name (such as "result").
commit 9e722d62 (2018-10-08): Account for topentity annotation when generating port names. Fix #351

Users can specify custom port names for their top entity. Clash did not
account for this possibility when generating unique argument / result
names thus generating name collisions whenever some commonly used
variable name was used as a port name (such as "result").
commit 212320da (2018-10-08): Account for topentity annotation when generating port names. Fix #351

Users can specify custom port names for their top entity. Clash did not
account for this possibility when generating unique argument / result
names thus generating name collisions whenever some commonly used
variable name was used as a port name (such as "result").
commit b597f3da (2018-10-09): Account for topentity annotation when generating port names. Fix #351

Users can specify custom port names for their top entity. Clash did not
account for this possibility when generating unique argument / result
names thus generating name collisions whenever some commonly used
variable name was used as a port name (such as "result").
closed 2018-10-09 by martijnbastiaan