clash-lang/clash-compiler #352: Exception: Cannot reduce an integer

Jhana1, 2018-09-12:
I've attempted to make a minimal reproducible example of this failure. The key issue is around `resizeUnpack`.

Directly specifying the size of the intermediate vector instead of `DVec` also fixes the issue. I'll comment that below.
```
module Test where

import           Clash.Prelude              hiding (lift, min)
import           Clash.Signal               (exposeClockReset)
import           Control.Lens               hiding (Index, index)
import           Control.Monad              (when)
import           Control.Monad.State.Strict
import           Data.Word

type Constraint t = (KnownNat (BitSize t), BitPack t)
type DVec t = Vec (DivRU (BitSize t) (BitSize Word32)) Word32

extract :: ( Constraint t ) => State (DVec t) (Maybe t) -- Fails
extract = do
  s <- get
  return $ Just . resizeUnpack $ s

resizeUnpack :: Constraint t => DVec t -> t
resizeUnpack dv = unpack . resize $ (pack dv)

topEntity :: Clock System Source -> Reset System Asynchronous -> Signal System (Word32) -> Signal System (Maybe Word32)
topEntity = exposeClockReset $ (`stateMachine` def) $ \i -> do
  extract

-- Utility
stateMachine :: (HiddenClockReset dom gate sync) => (i -> State s o) -> s -> Signal dom i -> Signal dom o
stateMachine f iS = \i -> let (o, s') = unbundle $ runState . f <$> i <*> s
                              s       = register iS s'
                           in o
```
```
$ stack exec clash -- Test.hs --vhdl                                                                                                                   
Loading dependencies took 1.395947602s
Compiling: Test.topEntity

Test.hs:19:1: error:
    Clash.Normalize(380): Clash.Core.Util(459): Cannot reduce an integer: AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.TypeNats.Div3674937295934325082, nameLoc = UnhelpfulSpan "<wired into compiler>"}))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.TypeNats.+3674937295934325068, nameLoc = UnhelpfulSpan "<wired into compiler>"}))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = Clash.Class.BitPack.BitSize8214565720323788855, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.Word.Word323674937295934324864, nameLoc = UnhelpfulSpan "<no location info>"}))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.TypeNats.-3674937295934325076, nameLoc = UnhelpfulSpan "<wired into compiler>"}))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = Clash.Class.BitPack.BitSize8214565720323788855, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.Word.Word323674937295934324864, nameLoc = UnhelpfulSpan "<no location info>"}))))) (LitTy (NumTy 1))))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = Clash.Class.BitPack.BitSize8214565720323788855, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.Word.Word323674937295934324864, nameLoc = UnhelpfulSpan "<no location info>"}))))
    
    NB: The source location of the error is not exact, only indicative, as it is acquired after optimisations.
    The actual location of the error can be in a function that is inlined.
    To prevent inlining of those functions, annotate them with a NOINLINE pragma.
   |
19 | resizeUnpack dv = unpack . resize $ (pack dv)
   | ^^^^^^^^^^^^
```

Jhana1, 2018-09-12:
changing the signature of `resizeUnpack` 'fixes' the issue. However I want to be polymorphic (as my actual program is significantly larger/more complex)
```
module Test where

import           Clash.Prelude              hiding (lift, min)
import           Clash.Signal               (exposeClockReset)
import           Control.Lens               hiding (Index, index)
import           Control.Monad              (when)
import           Control.Monad.State.Strict
import           Data.Word

type Constraint t = (KnownNat (BitSize t), BitPack t)
type DVec t = Vec (DivRU (BitSize t) (BitSize Word32)) Word32

extract :: ( Constraint Word32 ) => State (Vec 1 Word32) (Maybe Word32)
extract = do
  s <- get
  return $ Just . resizeUnpack $ s

resizeUnpack :: Constraint t => Vec 1 t -> t
resizeUnpack dv = unpack . resize $ (pack dv)

topEntity :: Clock System Source -> Reset System Asynchronous -> Signal System (Word32) -> Signal System (Maybe Word32)
topEntity = exposeClockReset $ (`stateMachine` def) $ \i -> do
  extract

-- Utility
stateMachine :: (HiddenClockReset dom gate sync) => (i -> State s o) -> s -> Signal dom i -> Signal dom o
stateMachine f iS = \i -> let (o, s') = unbundle $ runState . f <$> i <*> s
                              s       = register iS s'
                           in o
```

Jhana1, 2018-09-12:
Manually inlining `resizeUnpack` gives us a different error.
```
module Test where

import           Clash.Prelude              hiding (lift, min)
import           Clash.Signal               (exposeClockReset)
import           Control.Lens               hiding (Index, index)
import           Control.Monad              (when)
import           Control.Monad.State.Strict
import           Data.Word

type Constraint t = (KnownNat (BitSize t), BitPack t)
type DVec t = Vec (DivRU (BitSize t) (BitSize Word32)) Word32

extract :: ( Constraint t ) => State (DVec t) (Maybe t)
extract = do
  s <- get
  return $ Just . unpack . resize . pack $ s

topEntity :: Clock System Source -> Reset System Asynchronous -> Signal System (Word32) -> Signal System (Maybe Word32)
topEntity = exposeClockReset $ (`stateMachine` def) $ \i -> do
  extract

-- Utility
stateMachine :: (HiddenClockReset dom gate sync) => (i -> State s o) -> s -> Signal dom i -> Signal dom o
stateMachine f iS = \i -> let (o, s') = unbundle $ runState . f <$> i <*> s
                              s       = register iS s'
                           in o
```
```
stack exec clash -- Test.hs --vhdl
Loading dependencies took 1.369135217s
Compiling: Test.topEntity
Applied 244 transformations
Normalisation took 0.442111096s
Netlist generation took 0.00192669s

<no location info>: error:
    Clash error call:
    Clash.Netlist.BlackBox(122): Forced to evaluate untranslatable type: (Clash.Sized.Vector.concatBitVector#
  @(GHC.TypeNats.Div
      (GHC.TypeNats.+
         (Clash.Class.BitPack.BitSize GHC.Word.Word32)
         31)
      32)
  @(Clash.Class.BitPack.BitSize GHC.Word.Word32)
  1
  32
  (Clash.Sized.Vector.map @GHC.Word.Word32
     @(Clash.Sized.Internal.BitVector.BitVector
         (Clash.Class.BitPack.BitSize GHC.Word.Word32))
     @(GHC.TypeNats.Div
         (GHC.TypeNats.+
            (Clash.Class.BitPack.BitSize GHC.Word.Word32)
            31)
         32)
     (λ(ds4 :: GHC.Word.Word32) ->
     letrec
       #case_alt :: Clash.Sized.Internal.BitVector.BitVector
                      32
       = Clash.Sized.Internal.BitVector.fromInteger#
           @32
           32
           (GHC.Integer.Type.wordToInteger
              x#)
       x# :: GHC.Prim.Word#
       = case ds4 of
           GHC.Word.W32#  (#sel :: GHC.Prim.Word#) ->
             #sel
     in #case_alt)
     x1) :: Clash.Sized.Internal.BitVector.BitVector
  (GHC.TypeNats.*
     (GHC.TypeNats.Div
        (GHC.TypeNats.+
           (Clash.Class.BitPack.BitSize GHC.Word.Word32)
           31)
        32)
     (Clash.Class.BitPack.BitSize GHC.Word.Word32)))
```

Jhana1, 2018-09-12:
Altering the original failing `Test` module such that it compiles with Clash 0.7.2.
```
extra-deps:
- git: git://github.com/clash-lang/clash-compiler
  commit: 33b6e62e9304e3bef8cb48ea7374847b3ca8d8f5
  subdirs:
  - clash-ghc
  - clash-lib
  - clash-vhdl
  - clash-verilog
  - clash-systemverilog
- git: git://github.com/clash-lang/clash-prelude
  commit: f8cd10a96bd23428aa79b13d453b8dc5c0b3a66e
```
```
module Test where

import           CLaSH.Prelude              hiding (lift, min)
import           Control.Monad              (when)
import           Control.Monad.State.Strict
import           Data.Word

type Constraint t = (KnownNat (BitSize t), BitPack t)
type DVec t = Vec (DivRU (BitSize t) (BitSize Word32)) Word32

extract :: ( Constraint t ) => State (DVec t) (Maybe t)
extract = do
  s <- get
  return $ Just . resizeUnpack $ s

resizeUnpack :: Constraint t => DVec t -> t
resizeUnpack dv = unpack . resize $ (pack dv)

topEntity :: Signal (Word32) -> Signal (Maybe Word32)
topEntity = (`stateMachine` def) $ \i -> do
  extract

-- Utility
stateMachine :: (i -> State s o) -> s -> Signal i -> Signal o
stateMachine f iS = \i -> let (o, s') = unbundle $ runState . f <$> i <*> s
                              s       = register iS s'
                           in o
```
```
$ stack exec clash -- Test.hs --vhdl
Loading dependencies took 1.21657s
Applied 92 transformations
Normalisation took 0.291741s
Netlist generation took 0.007501s
Testbench generation took 0.000159s
Total compilation took 1.519681s
```

Jhana1, 2018-09-12:
Hopefully that can help find the issue.

Jhana1, 2018-09-12:
Reduced it even further. You get three different behaviours depending on what you uncomment.
```
module Test where

import           Clash.Prelude
import           Clash.Signal  (exposeClockReset)
import           Data.Word

type Constraint t = (KnownNat (BitSize t), BitPack t)
type DVec t = Vec (DivRU (BitSize t) (BitSize Word32)) Word32

--{-# INLINE resizeUnpack #-}
resizeUnpack :: Constraint t => DVec t -> t
resizeUnpack = unpack . resize . pack

topEntity :: Clock System Source -> Reset System Asynchronous -> Signal System (DVec Word32) -> Signal System Word32
topEntity = exposeClockReset $ fmap resizeUnpack
--topEntity = exposeClockReset $ fmap (unpack . resize . pack)
```
As Written:
```
$ stack exec clash -- Test.hs --vhdl
Loading dependencies took 1.342921833s
Compiling: Test.topEntity

<no location info>: error:
    Clash.Normalize(380): Clash.Core.Util(459): Cannot reduce an integer: AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.TypeNats.Div3674937295934325082, nameLoc = UnhelpfulSpan "<wired into compiler>"}))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.TypeNats.+3674937295934325068, nameLoc = UnhelpfulSpan "<wired into compiler>"}))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = Clash.Class.BitPack.BitSize8214565720323788855, nameLoc = UnhelpfulSpan "<no location info>"}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = GHC.Word.Word323674937295934324864, nameLoc = UnhelpfulSpan "<no location info>"}))))) (LitTy (NumTy 31)))) (LitTy (NumTy 32))
    
    NB: The source location of the error is not exact, only indicative, as it is acquired after optimisations.
    The actual location of the error can be in a function that is inlined.
    To prevent inlining of those functions, annotate them with a NOINLINE pragma.
```
With Inline pragma:
```
$ stack exec clash -- Test.hs --vhdl
Loading dependencies took 1.372963915s
Compiling: Test.topEntity
Applied 195 transformations
Normalisation took 0.353286303s
Netlist generation took 0.000857445s

<no location info>: error:
    Clash error call:
    Clash.Netlist.BlackBox(122): Forced to evaluate untranslatable type: (Clash.Sized.Vector.concatBitVector#
  @(GHC.TypeNats.Div
      (GHC.TypeNats.+
         (Clash.Class.BitPack.BitSize GHC.Word.Word32)
         31)
      32)
  @(Clash.Class.BitPack.BitSize GHC.Word.Word32)
  (Clash.Transformations.removedArg
     @GHC.Natural.Natural)
  32
  (Clash.Sized.Vector.map @GHC.Word.Word32
     @(Clash.Sized.Internal.BitVector.BitVector
         (Clash.Class.BitPack.BitSize GHC.Word.Word32))
     @(GHC.TypeNats.Div
         (GHC.TypeNats.+
            (Clash.Class.BitPack.BitSize GHC.Word.Word32)
            31)
         32)
     Clash.Class.BitPack.$fBitPackWord58214565720323810544
     \#arg_1\) :: Clash.Sized.Internal.BitVector.BitVector
  (GHC.TypeNats.*
     (GHC.TypeNats.Div
        (GHC.TypeNats.+
           (Clash.Class.BitPack.BitSize GHC.Word.Word32)
           31)
        32)
     (Clash.Class.BitPack.BitSize GHC.Word.Word32)))
```
Manually Inlined:
```
$ stack exec clash -- Test.hs --vhdl
Loading dependencies took 1.200792483s
Compiling: Test.topEntity
Applied 37 transformations
Normalisation took 0.104005961s
Netlist generation took 0.001552395s
Total compilation took 1.308164839s
```

Jhana1, 2018-09-12:
And here you can see it working in Clash 0.7.2
```
module Test where

import           CLaSH.Prelude
import           Data.Word

type Constraint t = (KnownNat (BitSize t), BitPack t)
type DVec t = Vec (DivRU (BitSize t) (BitSize Word32)) Word32

resizeUnpack :: Constraint t => DVec t -> t
resizeUnpack = unpack . resize . pack

topEntity :: Signal (DVec Word32) -> Signal (Word32)
topEntity = fmap resizeUnpack
```
```$ stack exec clash -- Test.hs --vhdl
Loading dependencies took 0.907398s
Applied 15 transformations
Normalisation took 0.241541s
Netlist generation took 0.003535s
Testbench generation took 0.000131s
Total compilation took 1.155322s
```

leonschoorl, 2018-10-03:
Clash isn't solving the `Div`.

`ghcTypeToHWType` tries to lookup the size of the `Vec` using  `tyNatSize` and fails.

Minimal testcase:
```haskell
topEntity :: Vec (Div 4 2) Bit -> Vec (Div 4 2) Bit
topEntity = id
```

martijnbastiaan, 2018-10-04:
I'd be fine with fixing this, but I'm not sure where to start. Should this happen in one of the plugins, or in `clash-compiler`?

leonschoorl, 2018-10-04:
@christiaanb said: [reduceTypeFamily](https://github.com/clash-lang/clash-compiler/blob/22605c6bdf0ea3f8c5f241c9e09deefb14a57a73/clash-lib/src/Clash/Core/Type.hs#L458) is responsible for that reduction

commit 935d7860 (2018-10-04): Account for more type-level operations in `GHC.TypeNats`. Fix #352
commit b03abebb (2018-10-08): Account for more type-level operations in `GHC.TypeNats`. Fix #352
commit d587dbdd (2018-10-08): Account for more type-level operations in `GHC.TypeNats`. Fix #352
closed 2018-10-08 by martijnbastiaan