clash-lang/clash-compiler #354: Feature request: multi-core compilation

gergoerdi, 2018-09-14:
I am running `clash --vhdl` to generate VHDL, and I just noticed that it is pegging exactly one core at 100% and nothing else. Compilation took almost 3 minutes, so it would matter a lot if I could decrease that 8-fold on my 8-core machine.

christiaanb, 2018-09-14:
I’ll keep it in mind. On a compile-time improvement note, I’m working on dropping `unbound-generics` and using the same substitution method as GHC. It’s currently on a private branch, but I plan to release it quite soon, hopefully before icfp. You can see the speed up at the bottom of this page: https://github.com/clash-lang/clash-compiler/wiki/Normalization-benchmark

alex-mckenna, 2022-01-14:
Concurrency work has started in #2034, and will continue with #2035. Since that issue is more specific, this issue can be closed.

closed 2022-01-14 by alex-mckenna