clash-lang/clash-compiler #356: "Signal is defined more than once"

gergoerdi, 2018-09-15:
I am getting the following error from the Xilinx ISE consuming CLaSH-generated VHDL:

> Line 6134: Signal `inst_chip8/clash_explicit_mealy_mealy_cpuout/_` is defined more than once.

I don't have any idea on how to even start cutting this down to a minimal repro, but my full repro is at https://github.com/gergoerdi/chip8-clash/tree/6702fa2c6979647e295712691cf45d0e686b2209.

gergoerdi, 2018-09-15:
I've put all the VHDL (hand-written and CLaSH-generated) to https://github.com/gergoerdi/clash-issue-356/tree/d3ea7a5ef9d96ab0314d942603c6d1bae47de567

Funnily, none of the VHDL files even has 6134 lines..

gergoerdi, 2018-09-15:
I don't have a way of minimizing the code yet, but I did notice that commenting out some opcode implementations in `CHIP8/CPU.hs` makes synthesis go through. In particular, commenting out the following opcode implementations is enough:

* `DrawSprite`
* `SkipKey`
* `StoreRegs`
* `LoadRegs`

gergoerdi, 2018-09-15:
> It’s been tested on quite large code bases, but apparently not large enough.

I guess the big difference is that those were large code bases written by people who knew what they were doing :)

gergoerdi, 2018-09-17:
[According to the *people of StackOverflow*](https://stackoverflow.com/q/52359821/477476), the error could be coming from around [`clash_explicit_mealy_mealy.vhdl:3116`](https://github.com/gergoerdi/clash-issue-356/blob/master/vhdl/CHIP8/CHIP8/clash_explicit_mealy_mealy.vhdl#L3116), simply overloading the synthesizer and causing it to run out of stack or heap.
 
If that's correct, then the relevant CLaSH code would be something around the handling of the CHIP-8 register file [as a `Vec 16 Word8`](https://github.com/gergoerdi/clash-issue-356/blob/master/vhdl/CHIP8/CHIP8/clash_explicit_mealy_mealy.vhdl#L3116) and [indexing into it with an input-dependent index](https://github.com/gergoerdi/chip8-clash/blob/6702fa2c6979647e295712691cf45d0e686b2209/src-clash/CHIP8/CPU.hs#L124).

christiaanb, 2018-09-18:
I'll run it through quartus, and see if I can get a more sensible error message than complaining about a line that doesn't exist.

expipiplus1, 2018-09-19:
Gergő and I ran it through Yosys without problems yesterday, albeit with
Verilog output; I think ISE failed with an error on Verilog too though.

gergoerdi, 2018-09-19:
Yes, I've tried running it in ISE via Verilog instead of VHDL; that gave a similar error message ("signal is defined more than once) but with a completely different location/instantiation path.

christiaanb, 2018-09-19:
Running the clash-generated files from https://github.com/gergoerdi/clash-issue-356/tree/d3ea7a5ef9d96ab0314d942603c6d1bae47de567 through both Quartus 18.0 and Vivado 2018.1 doesn't give any of the issues you described.

Might be that ISE isn't handling the Extended / Escaped identifiers produced by Clash very well.

christiaanb, 2018-09-19:
Flagging it up as a request for an "enhancement" to produce code accepted by ISE.

gergoerdi, 2018-09-19:
I don't think it's the extended identifiers; if it was, then nothing produced by CLaSH would have worked with the ISE for me. For example, [an earlier version of my code without the full-featured CPU](https://github.com/gergoerdi/chip8-clash/tree/bbe1bc2b9e9311b3254aa250bf3dfd96defcc039) synthesized with ISE without issues.

gergoerdi, 2018-09-20:
I've tried some things yesterday.

First off, I've hacked the emitted VHDL not to contain any blocks, by lifting all block-local signal declarations into the component declaration. This merely changed slightly the bogus location where ISE reports "duplicate signal definition". 

Then I [rewrote the CPU in the most direct style possible](https://github.com/gergoerdi/chip8-clash/blob/ise/src-clash/CHIP8/CPU.hs), as a `CPUState -> CPUIn -> (CPUState, CPUOut)` function (no `RWS`, no accumulation of `Endo`morphisms, ...), since I was worried that maybe CLaSH has a deficiency in turning these abstractions into efficient VHDLS, which would be sad and surprising, considering how good a job GHC usually does at that. This simplified version still either crashes XST (with the bogus "duplicate signal definition" error showing up at various places) or causes it to loop: I've let it run for 10 hours, it is using one core 100% and not consuming any memory (~300M) with no result.  This looping also happens in the 'low-level synthesis' step, same step that would crash otherwise.

I'm running out of ideas here. One remaining thing I can think of is to investigate the component usage of the generated VHDL -- I noticed XST reports 600+ multiplexers which seems a lot for such a simple CPU.

bgamari, 2018-10-18:
For what it's worth, I use VHDL with Vivado as well. I have encountered blatantly incorrect synthesis using Vivado given Verilog.