clash-lang/clash-compiler #358: Change type of `NetlistState._seenIds` to set

martijnbastiaan, 2018-09-20:
`NetlistState._seenIds` and `NetlistState._seenComps` are currently lists. As they are mostly used for lookups (`elem`), a (hash)set would be a more suitable data structure. Although this is a quite simple fix, I will postpone implementing it until https://github.com/clash-lang/clash-compiler/wiki/Normalization-benchmark is merged due to the complexity of that work.

christiaanb, 2018-09-20:
You should probably count the number of lookups and inserts. Because the list version has a cheaper insert than (Hash)Set.

martijnbastiaan, 2018-09-26:
They're basically balanced. Even for our small examples the lists approach 100 members, so I think we should be much better off changing it. I'll leave the final decision to you.

> . Although this is a quite simple fix, I will postpone implementing

I forgot that I said that, so I'll rebase the patch if needed..

martijnbastiaan, 2019-02-22:
Added in https://github.com/clash-lang/clash-compiler/commit/b55c6fb03e8e28fcad3250aa5c24ec42db948732.

I think the general rule here is to avoid using sets *anytime* you use `elem` on it, unless you can guarantee the list will always be very small (say <10). While it's true that lists have the benefit of `O(1)` insertion as opposed to `O(logn)` for sets, this will always lose against having `O(n)` lookup. For an arbitrary ratio of *M* inserts versus a single lookup:

lists: `O(M * 1) + O(n)`  ~ `O(1) + O(n)` ~ `O(n)`
ordsets: `O(M * log(n)) + O(log(n))`  ~ `O(log(n)) + O(log(n))` ~ `O(2log(n))` ~ `O(log(n))`

It should be noted that it might take a while for the constants to take over if comparison/hashing is expensive. So, to put it all together; if you use `elem` you should:

* Make sure your list remains very small, or..
* .. replace it by `HashSet.member` if hashing is cheap (if your key is a `String` or `Text`, consider [FastString](https://hackage.haskell.org/package/ghc-8.4.3/docs/FastString.html))
* .. replace it by `Set.member` if comparisons are cheap

commit ebbe6a3f (2018-10-08): Use `Set` to keep track of seen ids. Fix #358
closed 2019-02-22 by martijnbastiaan