clash-lang/clash-compiler #359: Result type of TopEntity fails to synthesize when it's a newtype of a function type

christiaanb, 2018-09-20:
This fails:

```haskell
{-# LANGUAGE DataKinds #-}

import           Clash.Prelude

newtype Circuit a b c d = Circuit ((a,b) -> (c, d))

topEntity
  :: Clock System 'Source
  -> Reset System 'Asynchronous
  -> Circuit (Signal System Int) () () (Signal System Int)
topEntity c r = withClockReset c r $ Circuit $ \(i,()) -> ((), register 0 i)
```

because of:

https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/Clash/Netlist.hs#L199-L211

specifically, `getFunResult` doesn't look through newtypes. We have `splitCoreFunForallTy` but that also looks through annotations, which is the thing we want to keep. So we must make something like `splitCoreFunForallTy` that keeps annotations.

martijnbastiaan, 2018-09-25:
> We have `splitCoreFunForallTy` but that also looks through annotations, which is the thing we want to keep.

Are you sure? I don't think it does. It's just that GHC throws away annotation information at some point, which is what we're counteracting at that point. I'll provide a patch in a sec.

martijnbastiaan, 2018-09-25:
^ See commit ^

commit 818b4b15 (2018-10-08): Account for newtypes in `genComponentT`. Fix #359
commit 855f5b12 (2018-10-08): Account for newtypes in `genComponentT`. Fix #359
closed 2018-10-08 by martijnbastiaan