clash-lang/clash-compiler #364: Simple design fails to compile

adamwalker, 2018-09-30:
Using Clash 0.99.3, compilation of the design below fails with the error:

```
<no location info>: error:
    Clash error call:
    Clash.Normalize(189): Expr belonging to bndr: x1536 not found
```
```
module Mod where

import Clash.Prelude

newtype Thing dom = Thing {unThing :: Signal dom (BitVector 8) -> Signal dom (BitVector 32)}

instance Semigroup (Thing dom) where 
   Thing x <> Thing y = Thing $ \o -> liftA2 (.|.) (x o) (y o)

instance Monoid (Thing dom) where
    mempty = Thing (const (pure 0))

func
    :: BitVector 8
    -> BitVector 8
    -> BitVector 32
func d c
    | c == 0    = resize d
    | otherwise = 0

mod'
    :: forall dom gated sync. HiddenClockReset dom gated sync
    => Signal dom (BitVector 32)
mod' = o
    where

    x :: Signal dom (BitVector 8) = pure 0 
    y :: Signal dom (BitVector 8) = (+1) <$> z
    z :: Signal dom (BitVector 8) = pure 0

    f :: Thing dom
    f = foldl1 (<>) 
        $  Thing (liftA2 func x)
        :> Thing (liftA2 func y)
        :> Thing (liftA2 func z)
        :> Nil

    o = unThing f (pure 0)

topEntity clk rst = withClockReset @System @Source @Synchronous clk rst mod'
```

Jhana1, 2018-10-01:
I think there's something broken with `resize`
See #352

adamwalker, 2018-10-02:
I'm not sure. If I remove the `resize` the bug is still there.

leonschoorl, 2018-10-02:
So far I've reduced the test case to this:
```haskell
module Mod where

import Clash.Prelude

func
    :: BitVector 8
    -> BitVector 8
    -> BitVector 32
func d c = 0

mod'
    :: forall dom gated sync. HiddenClockReset dom gated sync
    => Signal dom (BitVector 32)
mod' = o
    where

    x :: Signal dom (BitVector 8) = pure 0

    f :: Signal dom (BitVector 8) -> Signal dom (BitVector 32)
    f = foldl1 (\x y q -> liftA2 (.|.) (x q) (y q))
        $  liftA2 func x
        :> liftA2 func x
        :> liftA2 func x
        :> Nil

    o :: Signal dom (BitVector 32)
    o = f (pure 0)

topEntity clk rst = withClockReset @System @Source @Synchronous clk rst mod'
```

There seems to be something weird going on in Clash.Core.Evaluator during a caseCon transformation. (Adding `-fclash-debug DebugApplied` will make it complain about free variables being added)

It seems to add a reference to this `x1536` but forgets to add the actual binding.

I'll dig some further tomorrow

commit a1a30fa8 (2018-11-05): Bind heap variables returned by the whnf' evaluator

Fixes #364
Fixes #383
closed 2018-11-05 by leonschoorl