clash-lang/clash-compiler #365: Vector update results in unknown ('XXXXXX') value

gergoerdi, 2018-10-02:
Here is my minimized CLaSH source:

```
module CHIP8 where

import Clash.Prelude

data Phase
    = Init
    | Exec
    | Halt

{-# NOINLINE topEntity #-}
{-# ANN topEntity
  (Synthesize
    { t_name   = "CHIP8"
    , t_inputs =
          [ PortName "CLK"
          , PortName "RESET"
          ]
    , t_output = PortName "OUTPUT"
    }) #-}
topEntity
    :: Clock System Source
    -> Reset System Asynchronous
    -> Signal System (Unsigned 8)
topEntity = exposeClockReset board
  where
    board = (!!0) . snd <$> s

    s = register (Init, pure 0) $ f <$> s
      where
        f :: (Phase, Vec 16 (Unsigned 8)) -> (Phase, Vec 16 (Unsigned 8))
        f (Init, regs) = (Exec, regs)
        f (Halt, regs) = (Halt, regs)
        f (Exec, regs) = (Halt, replace 0 (regs!!0 + 1) regs)

```
So all it does is it updates the 0th value in a vector by adding 1 to it.

Running the resulting VHDL in a simulator, it seems the 0th element of the vector becomes `XXXXXXXX` after the update. In the CLaSH simulator, `sampleN 10 $ hideClockReset topEntity` shows `[0, 0, 1, 1, 1..]` as I would expect.

There are several small changes one can make to the program so that it does not corrupt the register's value:
* if I don't use the old value in computing the new one, e.g. `replace 0 1 regs` instead of `replace 0 (regs!!0 + 1) regs`. Perhaps notably, using the old value from a different index, `replace 0 (regs!!1 + 1) regs`, does NOT fix the corruption.
* Changing the program to us a single `Unsigned 8` not in a `Vec`tor. Something like `Vec 1 (Unsigned 8)` is not enough, it has to be an unwrapped `Unsigned 8`.

gergoerdi, 2018-10-02:
Here's the emitted VHDL (compiled with `-fclash-intwidth=32` for unrelated reasons):

```
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.chip8_types.all;

entity CHIP8 is
  port(-- clock
       CLK    : in std_logic;
       -- asynchronous reset: active high
       RESET  : in std_logic;
       OUTPUT : out unsigned(7 downto 0));
end;

architecture structural of CHIP8 is
  signal \#s_app_arg\    : chip8_types.array_of_unsigned_8(0 to 0);
  signal ds1             : std_logic_vector(1 downto 0);
  signal \#s_case_alt\   : chip8_types.tup2;
  signal s               : chip8_types.tup2;
  signal y               : chip8_types.array_of_unsigned_8(0 to 0);
  signal \#case_alt_rec\ : unsigned(7 downto 0);
begin
  -- replace begin
  replacevec : block
    signal vec_index : integer range 0 to 1-1;
  begin
    vec_index <= to_integer(to_signed(0,32))
    -- pragma translate_off
                 mod 1
    -- pragma translate_on
                 ;

    process(vec_index,y)
      variable ivec : chip8_types.array_of_unsigned_8(0 to 0);
    begin
      ivec := y;
      ivec(vec_index) := (\#case_alt_rec\ + to_unsigned(1,8));
      \#s_app_arg\ <= ivec;
    end process;
  end block;
  -- replace end

  ds1 <= s.tup2_sel0;

  with (ds1) select
    \#s_case_alt\ <= ( tup2_sel0 => "01"
                     , tup2_sel1 => y ) when "00",
                     ( tup2_sel0 => "10"
                     , tup2_sel1 => \#s_app_arg\ ) when "01",
                     (tup2_sel0 => "10", tup2_sel1 => y) when others;

  -- register begin 
  chip8_register : process(CLK,RESET)
  begin
    if RESET = '1' then
      s <= ( tup2_sel0 => "00", tup2_sel1 => chip8_types.array_of_unsigned_8'(0 to 1-1 =>  to_unsigned(0,8) ) )
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    elsif rising_edge(CLK) then
      s <= \#s_case_alt\
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    end if;
  end process;
  -- register end

  y <= s.tup2_sel1;

  -- index begin
  indexvec : block
    signal vec_index_0 : integer range 0 to 1-1;
  begin
    vec_index_0 <= to_integer(to_signed(0,32))
    -- pragma translate_off
                 mod 1
    -- pragma translate_on
                 ;
    \#case_alt_rec\ <= y(vec_index_0);
  end block;
  -- index end

  OUTPUT <= \#case_alt_rec\;
end;

```

gergoerdi, 2018-10-02:
[As some have noticed in my related SO question](https://stackoverflow.com/q/52604935/477476), the generated code is missing `\#case_alt_rec\` from the sensitivity list of the process in the `replacevec` block.

leonschoorl, 2018-10-02:
Thanks for the report.

This is actually already fixed by 8aa875274c6972260e1911291ed2973328b70bf2
But there is no release yet that containing that fix.

You could try manually building from git with something like:
```sh
ghc-pkg unregister clash-ghc clash-lib clash-prelude
git clone https://github.com/clash-lang/clash-compiler.git
cd clash-compiler/clash-prelude
cabal install
cd ../clash-lib
cabal install
cd ../clash-ghc
cabal install
```

martijnbastiaan, 2018-10-03:
An alternative (cleaner?) way of running clash is:

```
git clone https://github.com/clash-lang/clash-compiler.git --recursive
cd clash-compiler
cabal new-run clash -- <your arguments here>
```

You'd have to stay in (a subdirectory of) the `clash-compiler` directory .

gergoerdi, 2018-10-03:
Hrm, any ideas how I'd integrate that into my build workflow, which is assuming the source files are in the current directory? I.e. my Shake rules all pass source files in the form `src/Foo.hs` to `clash`, so I won't be able to run it via a wrapper script that would run `cabal new-run clash` in the Clash source dir.

gergoerdi, 2018-10-03:
OK so I managed to hand-compile my repro code with 22605c6bdf0ea3f8c5f241c9e09deefb14a57a73, and the sensitivity list of `replacevec`'s process is bigger, but still not correct:

```
    process(vec_index,y,\x#\)
      variable ivec : chip8_types.array_of_unsigned_8(0 to 0);
    begin
      ivec := y;
      ivec(vec_index) := (resize((resize(\#w\(7 downto 0),32)),8));
      \#s_app_arg\ <= ivec;
    end process;
```
It should contain `\#w\`. So it seems this bug is still open.

leonschoorl, 2018-10-03:
@martijnbastiaan I suggested the old style cabal installation method precisely it doesn't require you to run clash its source dir.

@gergoerdi That's weird `\x#\` isn't even used in the process.
Are you still using minimal code from your initial report?
Because if I try compiling that code using the current master(22605c6bdf0ea3f8c5f241c9e09deefb14a57a73), compiled with GHC-8.4.3, I get:
```VHDL
    process(vec_index,y,\#case_alt_rec\)
      variable ivec : chip8_types.array_of_unsigned_8(0 to 15);
    begin
      ivec := y;
      ivec(vec_index) := (\#case_alt_rec\ + to_unsigned(1,8));
      \#s_app_arg\ <= ivec;
    end process;
```

gergoerdi, 2018-10-04:
@leonschoorl no you're right it does work with 22605c6 with my code from the original question! (except of course I had to add an `Undefined` instance for `Phase`).

However. It seems I had an accidental local change which is causing this wrong sensitivity list! Instead of `Unsigned 8`s, I have used `Word8` everywhere:

```
{-# LANGUAGE DeriveAnyClass #-}
module CHIP8 where

import Clash.Prelude
import Data.Word

data Phase
    = Init
    | Exec
    | Halt
    deriving Undefined

{-# NOINLINE topEntity #-}
{-# ANN topEntity
  (Synthesize
    { t_name   = "CHIP8"
    , t_inputs =
          [ PortName "CLK"
          , PortName "RESET"
          ]
    , t_output = PortName "OUTPUT"
    }) #-}
topEntity
    :: Clock System Source
    -> Reset System Asynchronous
    -> Signal System Word8
topEntity = exposeClockReset board
  where
    board = (!!0) . snd <$> s

    s = register (Init, pure 0) $ f <$> s
      where
        f :: (Phase, Vec 1 Word8) -> (Phase, Vec 1 Word8)
        f (Init, regs) = (Exec, regs)
        f (Halt, regs) = (Halt, regs)
        f (Exec, regs) = (Halt, replace 0 (regs!!0 + 1) regs)

```

And it is with this version that I am getting the following VHDL:

```
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.chip8_types.all;

entity CHIP8 is
  port(-- clock
       CLK    : in std_logic;
       -- asynchronous reset: active high
       RESET  : in std_logic;
       OUTPUT : out unsigned(7 downto 0));
end;

architecture structural of CHIP8 is
  signal \x#\            : unsigned(31 downto 0);
  signal \#s_app_arg\    : chip8_types.array_of_unsigned_8(0 to 0);
  signal ds1             : std_logic_vector(1 downto 0);
  signal \#s_case_alt\   : chip8_types.tup2;
  signal s               : chip8_types.tup2;
  signal y               : chip8_types.array_of_unsigned_8(0 to 0);
  signal \#case_alt_rec\ : unsigned(7 downto 0);
  signal \#w\            : unsigned(31 downto 0);

begin
  \x#\ <= resize(\#case_alt_rec\,32);

  \#w\ <= (\x#\ + to_unsigned(1,32));

  -- replace begin
  replacevec : block
    signal vec_index : integer range 0 to 1-1;
  begin
    vec_index <= to_integer(to_signed(0,32))
    -- pragma translate_off
                 mod 1
    -- pragma translate_on
                 ;

    process(vec_index,y,\x#\)
      variable ivec : chip8_types.array_of_unsigned_8(0 to 0);
    begin
      ivec := y;
      ivec(vec_index) := (resize((resize(\#w\(7 downto 0),32)),8));
      \#s_app_arg\ <= ivec;
    end process;
  end block;
  -- replace end

  ds1 <= s.tup2_sel0;

  with (ds1) select
    \#s_case_alt\ <= ( tup2_sel0 => "01"
                     , tup2_sel1 => y ) when "00",
                     ( tup2_sel0 => "10"
                     , tup2_sel1 => \#s_app_arg\ ) when "01",
                     (tup2_sel0 => "10", tup2_sel1 => y) when others;

  -- register begin 
  chip8_register : process(CLK,RESET)
  begin
    if RESET = '1' then
      s <= ( tup2_sel0 => "00", tup2_sel1 => chip8_types.array_of_unsigned_8'(0 to 1-1 =>  (resize(to_unsigned(0,32),8)) ) )
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    elsif rising_edge(CLK) then
      s <= \#s_case_alt\
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    end if;
  end process;
  -- register end

  y <= s.tup2_sel1;

  -- index begin
  indexvec : block
    signal vec_index_0 : integer range 0 to 1-1;
  begin
    vec_index_0 <= to_integer(to_signed(0,32))
    -- pragma translate_off
                 mod 1
    -- pragma translate_on
                 ;
    \#case_alt_rec\ <= y(vec_index_0);
  end block;
  -- index end

  OUTPUT <= \#case_alt_rec\;
end;
```

gergoerdi, 2018-10-09:
Here's a much-trimmed-down version that still exhibits the bug as of d3ff8f8b41f864cc1effaa810dbbb2fb37f2e0ba:

```
import Clash.Prelude
import Data.Word

type Byte = Word8
-- type Byte = Unsigned 8

{-# NOINLINE topEntity #-}
topEntity
    :: Clock System Source
    -> Reset System Asynchronous
    -> Signal System Byte
topEntity = exposeClockReset $ head <$> r
  where
    r = register (pure 0) $ f <$> r
      where
        f :: Vec 1 Byte -> Vec 1 Byte
        f regs = replace 0 (regs!!0 + 1) regs
```

This will generate a process for the vector update that is missing the new element value in its sensitivity list. However, if I change the definition of the type synonym `Byte` to `Unsinged 8`, the generated sensitivity list seems correct.

martijnbastiaan, 2018-10-11:
I *think* the following should fix the issue your having: 

```diff
diff --git a/clash-lib/src/Clash/Netlist/BlackBox/Util.hs b/clash-lib/src/Clash/Netlist/BlackBox/Util.hs
index b173837..1c8b04c 100644
--- a/clash-lib/src/Clash/Netlist/BlackBox/Util.hs
+++ b/clash-lib/src/Clash/Netlist/BlackBox/Util.hs
@@ -57,6 +57,8 @@ import           Clash.Signal.Internal           (ClockKind (Gated),
                                                   ResetKind (Synchronous))
 import           Clash.Util
 
+import Debug.Trace
+
 -- | Determine if the number of normal/literal/function inputs of a blackbox
 -- context at least matches the number of argument that is expected by the
 -- template.
@@ -472,20 +474,12 @@ renderTag b e@(TypElem _)   = let ty = lineToType b [e]
                               in  renderOneLine <$> getMon (hdlType Internal ty)
 renderTag _ (Gen b)         = renderOneLine <$> genStmt b
 renderTag _ (GenSym [C t] _) = return t
-renderTag b (Vars n)        =
-  let (e,_,_) = bbInputs b !! n
 
-      go (Identifier i _) = [i]
-      go (DataCon _ _ es) = concatMap go es
-      go (DataTag _ e')   = [either id id e']
-      go (Literal {})     = []
-      go (ConvBV _ _ _ e') = go e'
-      go (BlackBoxE _ _ _ _ t b' _) =
-        let usedArgs = mapMaybe (indexMaybe (bbInputs b')) (usedArguments t)
-        in  concatMap (\(e',_,_) -> go e') usedArgs
-
-      vars    = go e
-  in  case vars of
+-- Determine variables used in argument /n/.
+renderTag b (Vars n)        =
+  let (e,_,_) =  bbInputs b !! n
+      vars    = sensitivityList e
+  in  traceShow (e, vars) $ case vars of
         [] -> return Text.empty
         _  -> return (Text.concat $ map (Text.cons ',' . Text.fromStrict) vars)
 renderTag b (IndexType (L n)) =
@@ -727,8 +721,58 @@ prettyElem (Template bbname source) = do
                                   <> brackets (string $ Text.concat bbname')
                                   <> brackets (string $ Text.concat source'))
 
-usedArguments :: N.BlackBox
-              -> [Int]
+-- |
+sensitivityList :: Expr -> [Identifier]
+sensitivityList (Identifier i _)  = [i]
+sensitivityList (DataCon _ _ es)  = concatMap sensitivityList es
+sensitivityList (DataTag _ e')    = [either id id e']
+sensitivityList (Literal {})      = []
+sensitivityList (ConvBV _ _ _ e') = sensitivityList e'
+sensitivityList (BlackBoxE _ _ _ _ t b' _) = nub $ sList ++ sList'
+  where
+    -- Get input holes:
+    usedInputHoles = go
+      where
+        go = \case
+          D (Decl _ args) -> concatMap (\(a,b) -> concatMap go a ++ concatMap go b) args
+          I _ i -> [i]
+          Var _ _ -> []
+          IndexType e -> go e
+          FilePath e -> go e
+          Template bbname source -> concatMap go bbname ++ concatMap go source
+          IF b esT esF -> go b ++ concatMap go esT ++ concatMap go esF
+          SigD es _ -> concatMap go es
+          BV _ es _ -> concatMap go es
+          GenSym es _ -> concatMap go es
+          DevNull es -> concatMap go es
+          _ -> []
+
+    -- Recurse to get all explicitely mentioned variables
+    usedVars = go
+      where
+        go = \case
+          D (Decl _ args) -> concatMap (\(a,b) -> concatMap go a ++ concatMap go b) args
+          Var [C v] _ -> [Text.toStrict v]
+          IndexType e -> go e
+          FilePath e -> go e
+          Template bbname source -> concatMap go bbname ++ concatMap go source
+          IF b esT esF -> go b ++ concatMap go esT ++ concatMap go esF
+          SigD es _ -> concatMap go es
+          BV _ es _ -> concatMap go es
+          GenSym es _ -> concatMap go es
+          DevNull es -> concatMap go es
+          _ -> []
+
+    t' =
+      case t of
+        (N.BBFunction _ _ _) -> []
+        (N.BBTemplate es)    -> es
+
+    usedInputHoles' = mapMaybe (indexMaybe (bbInputs b')) (concatMap usedInputHoles t')
+    sList  = concatMap (\(e',_,_) -> sensitivityList e') usedInputHoles'
+    sList' = concatMap usedVars t'
+
+usedArguments :: N.BlackBox -> [Int]
 usedArguments (N.BBFunction _nm _hsh (N.TemplateFunction k _ _)) = k
 usedArguments (N.BBTemplate t) = nub (concatMap go t)
   where
```

I'll test the patch more thoroughly tomorrow and post an explanation :)

commit ec56a614 (2018-11-02): Collect variables (not inputs) in body of BlackBox in ~VARS. Fix #365
commit 2f818b07 (2018-11-02): Collect variables (not inputs) in body of BlackBox in ~VARS. Fix #365
commit 3c41a716 (2018-11-02): Collect variables (not inputs) in body of BlackBox in ~VARS. Fix #365
closed 2018-11-02 by martijnbastiaan