clash-lang/clash-compiler #372: Unsigned misbehaves at zero width

bgamari, 2018-10-08:
```
$ clashi -isrc
CLaSHi, version 0.99 (using clash-lib, version 0.99):
http://www.clash-lang.org/  :? for help
Clash.Prelude> 100 :: Unsigned 0
[hangs and starts chewing up memory]
```

martijnbastiaan, 2018-10-08:
I can't reproduce this on Clash 0.99.1 or master. The definition of `Unsigned` also looks quite simple, which leads me to believe that this might be a GHC issue. I'm running 8.4.3, what are you running?

martijnbastiaan, 2018-10-08:
GHC 8.2.2:

```
> Clash.Prelude> 100 :: Unsigned 0
0
```

Not good, but no sign of hanging.

bgamari, 2018-10-08:
Hmm, alright then, I'll clearly need to look at this again. I'm also using 8.4.3.

martijnbastiaan, 2018-10-10:
Are you still able to reproduce it? How did you install GHC? How did you obtain Clash?

martijnbastiaan, 2018-10-17:
I'm going to close this issue for now as no one seems to be able to reproduce it. If you encounter this issue again, please reopen this report.

closed 2018-10-17 by martijnbastiaan