clash-lang/clash-compiler #374: Fixed: fromInteger and fromRational don't saturate (correctly)

leonschoorl, 2018-10-09:
While looking at Ben's on fix for signum Fixed: c2d064f741d4d13280f2a14b812e8e458c027c70
I noticed that even with that fix:
```haskell
Clash.Prelude> signum (42 :: SFixed 4 2) == signum (42.0 :: SFixed 4 2)
False
```
It's because `fromInteger`(used for the literal `42`) and `fromRational`(used for the literal `42.0`) produce very different results when the input is out of range:
```haskell
Clash.Prelude> 42.0 :: SFixed 4 2
7.75
Clash.Prelude> 42 :: SFixed 4 2
-6.0
```
But our documentation states that Fixed saturates, so both of these should produces 3 times the same tuple:
```haskell
Clash.Prelude> [(minBound,maxBound), (-42.0, 42.0), (-42, 42)] :: [(SFixed 4 2,SFixed 4 2)]
[(-8.0,7.75),(-7.75,7.75),(6.0,-6.0)]
Clash.Prelude> [(minBound,maxBound), (-42.0, 42.0), (-42, 42)] :: [(UFixed 4 2,UFixed 4 2)]
[(0.0,15.75),(0.25,15.75),(6.0,10.0)]
```

martijnbastiaan, 2018-10-10:
So I guess this is because `Signed` and `Unsigned` don't actually saturate? That is:

```haskell
Clash.Prelude> 100 :: Signed 2
0
```

and:

```haskell
  fromInteger i    = let fSH = fromInteger (natVal (Proxy @frac))
                         res = Fixed (fromInteger i `shiftL` fSH)
                     in res
```

martijnbastiaan, 2018-10-11:
`fromInteger` does not know anything about its underlying representation, so unless we convert `res` back to an integer and compare it to the original input I'm not sure how we can fix this.

commit 085b3a41 (2020-01-17): Literals of Fixed saturate correctly

Needs `-XNegativeLiterals` to fully saturate to `minBound`,
because without that extension `-42` is desugared to
`negate (fromInteger 42)`, and `-42.0` is desugared to
`negate (fromRational (42 % 1))`. Which means it's first
saturated to the upper bound and then negated.

Fixes #374
PR #1015 (2020-01-17): Literals of Fixed saturate correctly
Needs `-XNegativeLiterals` to fully saturate to `minBound`, because without that extension `-42` is desugared to `negate (fromInteger 42)`, and `-42.0` is desugared to `negate (fromRational (42 % 1))`. Which means it's first saturated to the upper bound and then negated.

Fixes #374
commit 8171fccf (2020-01-17): Literals of Fixed saturate correctly

Needs `-XNegativeLiterals` to fully saturate to `minBound`,
because without that extension `-42` is desugared to
`negate (fromInteger 42)`, and `-42.0` is desugared to
`negate (fromRational (42 % 1))`. Which means it's first
saturated to the upper bound and then negated.

Fixes #374
closed 2020-01-17 by christiaanb