clash-lang/clash-compiler #381: DataFlow : "stepLock" fails when reducing input synchronization granularity for multi-cycle mealy type dataflow elements

syadav1994, 2018-10-16:
This is the code for multi-cycle mealy machine dataflow element :

mealyDF' :: forall domain s gated synchronous o i. 
           Clock domain gated
        -> Reset domain synchronous
        -> (s -> Maybe i -> (s, (Bool,Maybe o)))
        -> s
        -> DataFlow domain Bool Bool i o
mealyDF' clk rst f iS = DF dfFun
  where
    dfFun i iV oR = (o ,oV ,backpressure)
      where
        en :: Signal domain Bool
        en     = (iV .&&. (not <$> working)) .||.
                 (working .&&. (not <$> producing)) .||.
                 (producing .&&. oR)

        producing :: Signal domain Bool
        producing = oV

        i'     = combineI <$> iV <*> i         
        temp1 = (f <$> s <*> i') :: Signal domain (s, (Bool, Maybe o))    
        (s', temp2) = unbundle temp1 :: (Signal domain s, Signal domain (Bool, Maybe o))
        (backpressure, oMaybe) = unbundle temp2 :: (Signal domain Bool, Signal domain (Maybe o))
        
        o :: Signal domain o
        (o,oV) = unbundle $ splitO <$> oMaybe

        s      = E.register# (E.clockGate clk en) rst iS s'

        working,working' :: Signal domain Bool
        working = E.register# clk rst False working'
        working' = mux working (not <$> producing) iV

    combineI :: Bool -> i -> Maybe i
    combineI False _ = Nothing
    combineI True  i = Just i
    splitO :: Maybe o -> (o,Bool)
    splitO (Just x) = (x,True)
    splitO Nothing = (errorX "mealyDF' splitO Nothing", False)

-----------------------------------------------------------------------------------------------------------------------------------
Using the definition of mealyDF' to implement a multi-cycle floating point operation, like :
where lat_fpadd :: latency of floating point addition
and , Idle, Add a b n are the states of operation

-- Floating Point Addition
-- Input_State -> Input_arguments -> (Output_State , (can_accept_new_data, Output_Value))
fpadd :: Float_State -> Maybe (Float , Float) -> (Float_State , (Bool , Maybe Float))
fpadd Idle Nothing      = (Idle,                     (True,Nothing))
fpadd Idle (Just (a,b)) = (Add a b (lat_fpadd - 1), (False,Nothing))
fpadd (Add a b 0) _     = (Idle,                 (True, Just (a+b)))
fpadd (Add a b n) _     = (Add a b (n-1) ,         (False, Nothing))

-------------------------------------------------------------------------------------------------------------------------------------

If we do following things, then dataflow model fails and printing values on output gets stuck

fpadd' = (hideClockReset S.mealyDF') fpadd Idle

a = pureDF id
b = pureDF id

res = idDF `seqDF` stepLock `seqDF` ((a `parDF` b) `seqDF` lockStep fpadd') 

******** Printing output values for res results into terminal getting stuck
Initial stepLock is used for reducing granularity of input synchronisation channels

Any solution????

martijnbastiaan, 2018-11-06:
@syadav1994 , you've solved this right? Could you comment on this issue and (if applicable) close it?

syadav1994, 2018-11-06:
The stepLock and lockStep functions provided to reduce and expand the granularity of dataflow elements do not go hand in hand when we operate on multi-cycle operations.

If we read the assumptions on which dataflow module of ClaSh is based, which says : 
a) Not consume data when validity is deasserted.
b) Only update its output when readiness is asserted

And looking at the figures mentioned for stepLock and lockStep circuitary, we observe that only stepLock functions as per the assumptions mentioned above. 

Lockstep works differently by discarding the assumptions. For lockstep, output should be updated as soon as it is available without looking at the readiness of the dataflow element connected ahead of it.
So, A redesign of lockstep is required which follows the assumptions mentioned above.

In case of a multi-cycle environment, stepLock can be implemented as a state-machine which will perform the synchronization between dataflow elements.

closed 2018-11-06 by syadav1994