clash-lang/clash-compiler #383: fir filter from front page does not sythesize if len of fir is 3

ra1u, 2018-10-16:
```haskell
import Clash.Prelude

fir coeffs x = dotp coeffs (window x)
  where
    dotp as bs = sum (zipWith (*) as bs)

v = 1 :> 2 :> 3 :> Nil -- NOK
-- v = 1 :> 2 :> 3 :> 4 :> Nil -- OK

topEntity
  :: Clock System Source
  -> Reset System Asynchronous
  -> Signal System (Signed 8)
  -> Signal System (Signed 8)
topEntity = exposeClockReset ( fir v)

v = 1 :> 2 :> 3 :> Nil -- NOK
-- v = 1 :> 2 :> 3 :> 4 :> Nil -- OK
```

I am using clash-ghc from cabal `CLaSH, version 0.99.3 (using clash-lib, version: 0.99.3)`

command and output

    clash fir.hs --verilog
    Loading dependencies took 0.886579595s
    Compiling: Main.topEntity

    <no location info>: error:
        Clash error call:
        Clash.Rewrite.Util(409): Name {nameSort = Internal, nameOcc = x6658, nameLoc = UnhelpfulSpan "<no location info>"} not found

verbose start of clash
    -> % clash -v                           
    Glasgow Haskell Compiler, Version 8.4.2, stage 2 booted by GHC version 8.0.2
    Using binary package database: /usr/local/lib/ghc-8.4.2/package.conf.d/package.cache
    Using binary package database: /home/ralu/.ghc/x86_64-linux-8.4.2/package.conf.d/package.cache
    package flags []
    loading package database /usr/local/lib/ghc-8.4.2/package.conf.d
    loading package database /home/ralu/.ghc/x86_64-linux-8.4.2/package.conf.d
    wired-in package ghc-prim mapped to ghc-prim-0.5.2.0
    wired-in package integer-gmp mapped to integer-gmp-1.0.2.0
    wired-in package base mapped to base-4.11.1.0
    wired-in package rts mapped to rts
    wired-in package template-haskell mapped to template-haskell-2.13.0.0
    wired-in package ghc mapped to ghc-8.4.2
    wired-in package dph-seq not found.
    wired-in package dph-par not found.
    *** Deleting temp files:
    Deleting: 
    *** Deleting temp dirs:
    Deleting: 
    clash: no input files
    Usage: For basic information, try the `--help' option.

martijnbastiaan, 2018-10-17:
Interesting pattern..

| N  | Status                                   |
|----|------------------------------------------|
| **1**  | **OK**                                       |
| **2**  | **OK**                                       |
| 3  | Expr belonging to bndr: x12345 not found |
| **4**  | **OK**                                       |
| 5  | Expr belonging to bndr: x12345 not found |
| 6  | Expr belonging to bndr: x12345 not found |
| 7  | Expr belonging to bndr: x12345 not found |
| **8**  | **OK**                                       |
| 9  | Can't translate non-tycon type           |
| 10 | Expr belonging to bndr: x12345 not found |
| 11 | Can't translate non-tycon type           |
| 12 | Expr belonging to bndr: x12345 not found |
| 13 | Expr belonging to bndr: x12345 not found |
| 14 | Expr belonging to bndr: x12345 not found |
| 15 | Expr belonging to bndr: x12345 not found |
| **16** | **OK**                                       |
| 17 | Can't translate non-tycon type           |
| .. | ..                                       |
| **32** | **OK**                                       |
| 33 | Can't translate non-tycon type           |

martijnbastiaan, 2018-10-17:
This might be related to https://github.com/clash-lang/clash-compiler/issues/364. I'm marking it as duplicate for now, and revert the label if it ends up not being the case.

commit a1a30fa8 (2018-11-05): Bind heap variables returned by the whnf' evaluator

Fixes #364
Fixes #383
closed 2018-11-05 by leonschoorl