clash-lang/clash-compiler #407: romFile expects incorrect size of address bits

GuzTech, 2018-11-08:
The first example where romFile is used [here](http://hackage.haskell.org/package/clash-prelude-0.99.3/docs/Clash-Prelude-ROM-File.html) does not work with Clash 0.99.3 (and maybe with earlier versions, but I don't have them here):

```haskell
f :: HiddenClock domain => Signal domain (Unsigned 3) -> Signal domain (Unsigned 9)
f rd = unpack <$> romFile d7 "memory.bin" rd
```

A 3-bit address is enough to address the 7 elements of the memory, but Clash expects it to be the same number of bits as the amount of element (7 bits in this case). So this does compile:

```haskell
f :: HiddenClock domain Source => Signal domain (Unsigned 7) -> Signal domain (Unsigned 9)
f rd = unpack <$> romFile d7 "memory.bin" rd
```

Explicit.romFile does work correctly.

martijnbastiaan, 2018-11-08:
Bug confirmed, thanks! I'll have a look at it tomorrow.

martijnbastiaan, 2019-06-17:
This has been fixed in `master`. Thanks for the report :).

dramforever, 2020-07-03:
It seems that this is still not fixed in 1.2.2? The signature still has `n` as both the ROM size in words and address width: http://hackage.haskell.org/package/clash-prelude-1.2.2/docs/Clash-Prelude.html#v:romFile

```haskell
romFile
  :: (KnownNat m, KnownNat n, HiddenClock dom, HiddenEnable dom)
  => SNat n
  -> FilePath
  -> Signal dom (Unsigned n)
  -> Signal dom (BitVector m)
```

dramforever, 2020-07-03:
The generated code is pretty bad too, at least for Verilog... `c$i` is the super long address which I zero-extended from a nice and short `c$arg`, otherwise it wouldn't compile. Looks like clash arbitrarily just truncates it to 64 bits and indexes into the ROM with that.

Haskell code is essentially `romFile (SNat @3261) "file.txt" . fmap extend`, with 16 bit data width.

<details>

<summary>Verilog, cleaned up a bit</summary>

```verilog
module bad_rom_test
    ( // Inputs
      input  clk // clock
    , input  rst // reset
    , input [11:0] c$arg

      // Outputs
    , output wire [15:0] result
    );
  reg [15:0] x;
  wire signed [63:0] wild;
  wire signed [63:0] c$wild_app_arg;
  wire [3260:0] c$i;

  // romFile begin
  reg [15:0] ROM [0:3261-1];

  initial begin
    $readmemb("file.txt",ROM);
  end

  always @(posedge clk) begin : bad_rom_test_romFile
    x <= ROM[(wild)];
  end
  // romFile end

  assign result = x;

  assign wild = $signed(c$wild_app_arg);

  assign c$i = ({{(3261-12) {1'b0}},c$arg});

  assign c$wild_app_arg = $unsigned(c$i[0+:64]);
endmodule
```

</details>

christiaanb, 2020-07-03:
Right... it seems we only fixed `Clash.Explicit.Rom.File.romFile`, but not `Clash.Rom.File.romFile`.

gergoerdi, 2020-11-28:
Has this been given up on? Since the current `romFile` is unusable, I don't think there's any backwards compatibility to be maintained.

christiaanb, 2020-11-28:
No... simply forgotten. We'll fix it soon

gergoerdi, 2021-03-27:
It seems this bug has made it to 1.4.0. Since in its present form, `romFile` is unusable, there should be no compatibility issues with fixing this in 1.4.1.

PR #1728 (2021-03-27): `Clash.Prelude.romFile` takes a `Enum addr => addr` address line
Making it behave on par with `Clash.Explicit.Prelude.romFile` and thereby actually useful.

Fixes #407
commit 33175279 (2021-03-27): `Clash.Prelude.romFile` takes a `Enum addr => addr` address line

Making it behave on par with `Clash.Explicit.Prelude.romFile` and
thereby actually useful.

Fixes #407
commit 07b6f4c3 (2021-03-27): `Clash.Prelude.romFile` takes a `Enum addr => addr` address line

Making it behave on par with `Clash.Explicit.Prelude.romFile` and
thereby actually useful.

Fixes #407

(cherry picked from commit 33175279d5b6f43bd0b53c4ec3790c08a887f7c3)
closed 2021-03-27 by christiaanb