clash-lang/clash-compiler #413: generic map for netlist declarations

cchalmers, 2018-11-12:
Currently `InstDecl` in `Clash.Netlist.Types.Declaration` doesn't have a field for the "generic map" which means it can't be used to generate code that instantiates a generic component.

I propose an extra field is added to `InstDecl` of type `[(Expr, Expr)]` (or whatever this should be for generic map). So that `inst` will generate:

```
my_component : component my_component
  generic map (
   param => val
  )
  port map (...);
```

for the VHDL backend and the equivalent `#(parameters ...)` syntax for the Verilog backend.

oliverbunting, 2018-11-12:
I agree. Parameterised components are pretty vital

leonschoorl, 2018-11-13:
Do you have ideas on how to make this accessible for clash users?

cchalmers, 2018-11-13:
Well my use case is using it in the new blackbox haskell functions (#340). So I'd be directly using `blockDecl` from the `Backend` class in my blackbox template function.

PR #457 (2019-01-07): Inst parameters
Closes #413
closed 2019-01-07 by christiaanb