clash-lang/clash-compiler #420: Unknown Warnings while generating VHDL using DSignals

syadav1994, 2018-11-19:
While using DSignal (delayed signals) and generating VHDL for a simple design, i get warnings like this :

WARNING: file src/Clash/Core/Subst.hs, line 345
inScope {a3530822107858468865}
  tenv [3530822107858468865 :-> a3530822107858468865]
  tys k03530822107858468864
  -> a3530822107858468865 -> GHC.Prim.TYPE GHC.Types.LiftedRep
  needsInScope {k03530822107858468864}
WARNING: file src/Clash/Core/Subst.hs, line 345
inScope {a3530822107858468865}
  tenv [3530822107858468865 :-> a3530822107858468865]
  tys k03530822107858468864
  -> a3530822107858468865 -> GHC.Prim.TYPE GHC.Types.LiftedRep
  needsInScope {k03530822107858468864}
WARNING: file src/Clash/Core/Subst.hs, line 345
inScope {a3530822107858468865}
  tenv [3530822107858468865 :-> a3530822107858468865]
  tys k03530822107858468864
  -> a3530822107858468865 -> GHC.Prim.TYPE GHC.Types.LiftedRep
  needsInScope {k03530822107858468864}
WARNING: file src/Clash/Core/Subst.hs, line 345
inScope {a3530822107858468865}
  tenv [3530822107858468865 :-> a3530822107858468865]
  tys k03530822107858468864
  -> a3530822107858468865 -> GHC.Prim.TYPE GHC.Types.LiftedRep
  needsInScope {k03530822107858468864}

Code being synthesized is ::::::

function vx vy = res
        where
            res            = if' (vel_sqr !>=! (ad_Off zero_DS vel_sqr)) vel_sqr (ad_Off zero_DS vel_sqr)
            vel_sqr     = (sqr_DS vx) !+! (sqr_DS vy)

******* There are a lot of function definitions used here. These functions are mentioned below :
---------------------------------------------------------------------------------------------------------------------------
class DS_Num t where
    type Lat_Add t      :: Nat
    type Lat_Sqr t      :: Nat

    -- DSignal Addition operator (both DSignal)
    (!+!) :: forall domain n1 n2 n3 n4 gated sync.
             (HiddenClockReset domain gated sync, n3 ~ (Max n1 n2),
             (n2 <= n3), (n1 <= n3), n4 ~ (n3 + Lat_Add t),
             KnownNat n1, KnownNat n2, KnownNat n3)
          => DSignal domain n1 t
          -> DSignal domain n2 t
          -> DSignal domain n4 t

    -- DSignal Square operator
    sqr_DS :: forall domain n1 gated sync.
              (HiddenClockReset domain gated sync, KnownNat n1)
           => DSignal domain n1 t
           -> DSignal domain (n1 + Lat_Sqr t) t

class DS_Inequality t where
    type Lat_ge t       :: Nat

    -- DSignal Greater than equal to operator (both DSignal)
    (!>=!) :: forall domain n1 n2 n3 n4 gated sync.
              (HiddenClockReset domain gated sync, n3 ~ (Max n1 n2),
              (n2 <= n3), (n1 <= n3), n4 ~ (n3 + Lat_ge t),
              KnownNat n1, KnownNat n2, KnownNat n3)
           => DSignal domain n1 t
           -> DSignal domain n2 t
           -> DSignal domain n4 Bool

---------------------------------------------------------------------------------------------------------------------------

instance DS_Num Float where
    type Lat_Add Float    = 3
    type Lat_Sqr Float    = 3

    (!+!)      = fp_Add_single_elem
    sqr_DS    = fp_Sqr_single_elem

instance DS_Inequality Float where
    type Lat_ge Float = 0

    (!>=!)  = comp_function (ge_float)

ge_float :: Signal domain Float -> Signal domain Float -> Signal domain Bool
ge_float a b = a .>=. b

{-# NOINLINE ge_float #-}

---------------------------------------------------------------------------------------------------------------------------

fp_Add_single_elem a b = unsafeFromSignal out
  where
    (x,y) = sync_2signals a b
    x'    = toSignal x
    y'    = toSignal y
    out   = fpadd_single x' y'

fp_Sqr_single_elem a = unsafeFromSignal out
  where
    a'    = toSignal a
    out   = fpsqr_single a'

fpadd_single a b = register 0.0 (
                   register 0.0 (
                   register 0.0 (a+b)))
{-# NOINLINE fpadd_single #-}

fpsqr_single a = register 0.0 (
                 register 0.0 (
                 register 0.0 ((a*a))))

---------------------------------------------------------------------------------------------------------------------------

Some supporting functions used above are also mentioned below ::

-- Function used to adjust the offset of a DSignal with respect to other signal
ad_Off :: KnownNat n2 
       => (forall n1. KnownNat n1 => DSignal domain n1 b)  
       -> DSignal domain n2 a
       -> DSignal domain n2 b
ad_Off x y = x

-- if statement working on DSignals
-- automatically synchronizes the input signals
if' :: forall a domain gated sync n1 n2 n3 n4.
       (HiddenClockReset domain gated sync,
       n4 ~ (Max n3 (Max n1 n2)),
       (n1 <= n4), (n2 <= n4), (n3 <= n4),
       Undefined a, KnownNat n1,
       KnownNat n2, KnownNat n3, KnownNat n4)
    => DSignal domain n1 Bool
    -> DSignal domain n2 a
    -> DSignal domain n3 a
    -> DSignal domain n4 a
if' c t f = unsafeFromSignal out
    where
        (c', t', f') = sync_3signals c t f
        cond         = toSignal c'
        t_val        = toSignal t'
        f_val        = toSignal f'
        out          = mux cond t_val f_val

-- Function used to synchronize three signals using registers on either signal
sync_3signals :: forall a b c domain gated sync n1 n2 n3 n4.
                 (HiddenClockReset domain gated sync,
                 n4 ~ (Max n3 (Max n1 n2)),
                 (n1 <= n4), (n2 <= n4), (n3 <= n4),
                 Undefined a, KnownNat n1,
                 Undefined b, KnownNat n2,
                 Undefined c, KnownNat n3,
                 KnownNat n4)
              => DSignal domain n1 a
              -> DSignal domain n2 b
              -> DSignal domain n3 c
              -> (DSignal domain n4 a, DSignal domain n4 b, DSignal domain n4 c)
sync_3signals as bs cs = (delayedI' @a @(n4-n1) as, delayedI' @b @(n4-n2) bs, delayedI' @c @(n4-n3) cs)

zero_DS :: forall n1 domain a.
           (KnownNat n1, Num a)
        => DSignal domain n1 a
zero_DS = unsafeFromSignal (pure 0)

Any idea what might be the possible cause of these warnings !!!!

martijnbastiaan, 2019-02-12:
@syadav1994 I can't get your example to compile. Do you still happen to have the definitions of: `comp_function`, `sync_2signals`, and `delayedI'`?

syadav1994, 2019-02-12:
```Function definitions are as follows:

-- Function to delay a signal
delayedI' :: (Undefined a, KnownNat d, HiddenClockReset domain gated synchronous)
           => DSignal domain n1 a
           -> DSignal domain (n1 + d) a
delayedI' = (hideClockReset E.delayed) (repeat undefined)

-- Function used to synchronize two signals using registers on either signal
sync_2signals :: forall a b domain gated sync n1 n2.
                 (HiddenClockReset domain gated sync,
                 Undefined a, KnownNat n1,
                 Undefined b, KnownNat n2)
              => DSignal domain n1 a
              -> DSignal domain n2 b
              -> (DSignal domain (Max n1 n2) a, DSignal domain (Max n1 n2) b)
sync_2signals as bs = (delayedI' @a @((Max n1 n2)-n1) as, delayedI' @b @((Max n1 n2)-n2) bs)``

comp_function f a b = unsafeFromSignal out
  where
    (x,y) = sync_2signals a b
    x'    = toSignal x
    y'    = toSignal y
    out   = x' `f` `y'```

*** Some modifications are made to the type of functions. For e.g. in delayedI' n1+d should be equal to n2, and, and additional constraint of n1 <= n2 should be added in the type. Other functions should be edited in similar way to get this piece of code working.

martijnbastiaan, 2019-02-15:
This has been fixed, I don't get any warnings on the latest Clash. This has probably been fixed by one of the following commits:

* https://github.com/clash-lang/clash-compiler/commit/c67cc1519af3492b9734785a258b6e71987028c7
* https://github.com/clash-lang/clash-compiler/commit/e446c85ff286d61e67e7af285cf44eea7ea653fe

Thanks for the report!

closed 2019-02-15 by martijnbastiaan