clash-lang/clash-compiler #424: Clash is much slower when compiled using Nix

expipiplus1, 2018-11-22:
## Profiling results

- Output of `clash +RTS -p -RTS` after compiling with executable profiling enabled: https://gist.github.com/expipiplus1/4d83906df8048f8fec8ec1bc879d885c

- Output of `clash -dverbose-core2core`; observe that most time is spent in `Renamer/typechecker`: https://gist.github.com/expipiplus1/7f0149de2ebbc6f58ce493d544d7ffc3

## To reproduce

This occurs with at least ghc 844 and 862, and Clash versions since at least 82c4d9d063e8390aaf12f3399a0755fc2ca7e60c (Aug 10 2018)

### Clash built with nix

- Download a derivation for the latest version of Clash: `curl -O https://gist.githubusercontent.com/expipiplus1/9886b13ffad88109a207c4d282c44129/raw/a6ba97386d987eefac8cd11f671eaa58fd9ee693/default.nix`
- Create a simple example file `printf 'import Prelude\nmain = pure ()' > Blink.hs`
- Enter the nix shell with `nix-shell`
- Observe that Clash takes several seconds to compile this file: `time clash Blink.hs`
  - ```
$ time clash Blink.hs
[1 of 1] Compiling Main             ( Blink.hs, Blink.o )
Linking Blink ...
clash Blink.hs  64.37s user 1.09s system 99% cpu 1:05.48 total
```
- Observe that GHC takes very little time to compile this file: `time ghc Blink.hs`
  - ```
$ time ghc Blink.hs
[1 of 1] Compiling Main             ( Blink.hs, Blink.o )
Linking Blink ...
ghc Blink.hs  1.44s user 0.36s system 100% cpu 1.789 total
```

### Clash built with cabal

- Fetch clash: `git clone https://github.com/clash-lang/clash-compiler && cd clash-compiler`
- Download ghc and cabal `nix-shell -p haskell.compiler.ghc844 haskell.packages.ghc844.cabal-install`
- `cabal new-configure`
- `cabal new-build clash-ghc`
- Create a simple example file `printf 'import Prelude\nmain = pure ()' > Blink.hs`
- Observe that this Clash compiles compiles Blink.hs very quickly: `time cabal new-exec clash -- Blink.hs`

christiaanb, 2018-11-22:
@expipiplus1 did you compile `clash-ghc` with `executable-dynamic: True` on Nix? We do that for cabal builds: https://github.com/clash-lang/clash-compiler/blob/67d36495cc108afad767e92d1ee21e12411cc9e2/cabal.project#L34-L35

Taneb, 2018-11-22:
To do that with `nix`, you need to pass `enableExecutableSharing = true` to the Haskell builder function. It fixes this issue for me.

expipiplus1, 2018-11-23:
Thanks, this does inded solve the issue. Do you know why there is such a large difference in runtimes though?

closed 2018-11-23 by expipiplus1

martijnbastiaan, 2018-11-23:
We haven't figured that out yet, unfortunately.

expipiplus1, 2018-11-23:
Is it possible to detect the linking style at runtime (and issue a
warning)?

martijnbastiaan, 2018-11-23:
Absolutely. I'll add it as a feature request on the issue tracker.