clash-lang/clash-compiler #431: Generate sum types and use user-supplied record field names in VHDL

hesiod, 2018-11-25:
VHDL supports sum (as enums) and product (record) types, e.g.
```vhdl
type example_enum_type is (value, another_value);
```
and
```vhdl
type example_record is record
   member: std_logic;
   another_member: std_logic;
end record;
```
Most VHDL simulators/waveform viewers display enum values, aiding in debugging. If Clash could map Haskell sum types to enums, the generated VHDL would be more readable and viewing waveforms of Clash-generated VHDL would be more pleasant.
Clash already generates records for product types, however, the individual field names have generic names (`<name>_sel0`, `<name>_sel1` etc). If Clash could use field names of Haskell record types to give VHDL record types more appropriate names, the resulting VHDL would be even more readable.

I'm not quite sure how involved implementing these features would be or whether it is at all possible to realize them. If I understand the VHDL backend correctly, sum types are currently [normalized to bit vectors](https://github.com/clash-lang/clash-compiler/blob/67d36495cc108afad767e92d1ee21e12411cc9e2/clash-lib/src/Clash/Backend/VHDL.hs#L318-L337).
It should be possible to remove this normalization and instead map sum types to VHDL enum types, changing code generation in other place where appropriate, but I don't know enough about the Backend to tell whether that's enough.
Regarding field names, Clash's [`DataCon` type](https://hackage.haskell.org/package/clash-lib-0.99.3/docs/Clash-Core-DataCon.html#t:DataCon) as well as [the `Product` constructor of `HWType`](https://hackage.haskell.org/package/clash-lib-0.99.3/docs/Clash-Netlist-Types.html#t:HWType) does not contain the field names (in GHC, the field labels are [saved in `DataCon`s](https://downloads.haskell.org/~ghc/8.6.2/docs/html/libraries/ghc-8.6.2/DataCon.html#v:dataConFieldLabels)), so right now the backend doesn't know about them. Afterwards, changing the generated field names seems to be quite doable (e.g. [here](https://github.com/clash-lang/clash-compiler/blob/67d36495cc108afad767e92d1ee21e12411cc9e2/clash-lib/src/Clash/Backend/VHDL.hs#L367), [here](https://github.com/clash-lang/clash-compiler/blob/67d36495cc108afad767e92d1ee21e12411cc9e2/clash-lib/src/Clash/Backend/VHDL.hs#L1098-L1099) and in some other places).

martijnbastiaan, 2018-12-19:
>  Clash already generates records for product types, however, the individual field names have generic names (<name>_sel0, <name>_sel1 etc). If Clash could use field names of Haskell record types to give VHDL record types more appropriate names, the resulting VHDL would be even more readable.

FYI, this has been implemented in https://github.com/clash-lang/clash-compiler/pull/443

I don't think we do VHDL enums yet, so I'm leaving this issue open.

alex-mckenna, 2021-08-04:
Closing, as we now generate enums in VHDL by default in master

closed 2021-08-04 by alex-mckenna