clash-lang/clash-compiler #439: TH splice fails to find bindings from local modules when compiled to HDL from clashi

leonschoorl, 2018-12-02:
Thanks to Erik van Raalte for finding this.

LocalModule.hs:
```haskell
module LocalModule where
import Clash.Prelude

mysucc = succ
```

Main.hs
```haskell
import Clash.Prelude
import LocalModule

topEntity = $(lift $ mysucc (4 :: Int))
```

```
$ cabal new-run -- clashi Main.hs 
Up to date
Clashi, version 0.99 (using clash-lib, version 0.99):
http://www.clash-lang.org/  :? for help
Loaded package environment from /home/leon/dev/qbaylogic/clash-compiler-new/.ghc.environment.x86_64-linux-8.6.2
[1 of 2] Compiling LocalModule      ( LocalModule.hs, interpreted )
[2 of 2] Compiling Main             ( Main.hs, interpreted )
Ok, two modules loaded.
*Main> :vhdl
clashi: ^^ Could not load 'LocalModule_mysucc_closure', dependency unresolved. See top entry above.

ByteCodeLink.lookupCE
During interactive linking, GHCi couldn't find the following symbol:
  LocalModule_mysucc_closure
This may be due to you not asking GHCi to load extra object files,
archives or DLLs needed by your current session.  Restart GHCi, specifying
the missing library using the -L/path/to/object/dir and -lmissinglibname
flags, or simply by naming the relevant files on the GHCi command line.
Alternatively, this link failure might indicate a bug in GHCi.
If you suspect the latter, please send a bug report to:
  glasgow-haskell-bugs@haskell.org

*Main> :q
Leaving Clashi.
$ cabal new-run -- clashi Main.hs 
Up to date
Clashi, version 0.99 (using clash-lib, version 0.99):
http://www.clash-lang.org/  :? for help
Loaded package environment from /home/leon/dev/qbaylogic/clash-compiler-new/.ghc.environment.x86_64-linux-8.6.2
[2 of 2] Compiling Main             ( Main.hs, interpreted )
Ok, two modules loaded.
*Main> :vhdl
Loading dependencies took 1.202510007s
Parsing and compiling primitives took 0.368810688s
Compiling: Main.topEntity
Applied 2 transformations
Normalisation took 0.369981836s
Netlist generation took 0.000164701s
Total compilation took 1.590875759s
Clash.Prelude Main> :quit
Leaving Clashi.
$ 
```

Interestingly if you quit clashi and try again, it will work.
And also `clash --vhdl` works without problems.

leonschoorl, 2018-12-02:
Actually this might be related to #285

martijnbastiaan, 2018-12-03:
So I guess we have to lookup which modules are currently `interpreted`, and load them explicitly in `loadModules`. We can query GHC with [getContext](http://hackage.haskell.org/package/ghc-8.6.1/docs/GHC.html#v:getContext), according to [this blogpost](http://www.stephendiehl.com/posts/ghc_01.html).

christiaanb, 2019-08-14:
Bumping to 1.1 for a proper fix

leonschoorl, 2019-08-16:
Here's an example that still fails (from #707)

```haskell
module PortNameHacks where
import Clash.Prelude
import GHC.Exts

instance IsString PortName where
    fromString = PortName

instance IsList PortName where
    type Item PortName = PortName
    fromList = PortProduct []
    toList = error "toList for PortName is not implemented"
```

```haskell
{-# LANGUAGE OverloadedStrings,OverloadedLists #-}
module Ports where
import Clash.Prelude
import PortNameHacks

{-# ANN f
   (Synthesize
      { t_name   = "f"
      , t_inputs = ["a", [ "b", "c" ] ]
      , t_output = "q" }) #-}
f :: Bool -> (Bool,Bool) -> Bool
f a (b,c) = a || b && c
```

```
Clash.Prelude Ports> :vhdl
clashi: ^^ Could not load 'PortNameHacks_zdfIsListPortName2_closure', dependency unresolved. See top entry above.

ByteCodeLink.lookupCE
During interactive linking, GHCi couldn't find the following symbol:
  PortNameHacks_zdfIsListPortName2_closure
```

PR #852 (2019-10-15): Unload linked object files
Solves https://github.com/clash-lang/clash-compiler/issues/439#issuecomment-522015868
Fixes #439
commit bcc50303 (2019-10-15): Unload linked object files

Solves https://github.com/clash-lang/clash-compiler/issues/439#issuecomment-522015868
Fixes #439
commit ef8d841f (2019-10-15): Unload linked object files

Solves https://github.com/clash-lang/clash-compiler/issues/439#issuecomment-522015868
Fixes #439
commit 3985e42a (2019-10-15): Unload linked object files

Solves https://github.com/clash-lang/clash-compiler/issues/439#issuecomment-522015868
Fixes #439
closed 2019-10-15 by christiaanb