clash-lang/clash-compiler #452: Compilation error

adamwalker, 2018-12-17:
I pulled the latest master and tried to build a design. Clash failed with the error below. I don't have a minimal example because it can take quite a bit of work to make one. Instead I did some Clash compiler bisecting and discovered that if I revert the merge commit 7171f my design builds successfully. 

```
Loading dependencies took 5.763616313s
Parsing and compiling primitives took 0.432465223s
Compiling: Mod.topEntity
Applied 32893 transformations
Normalisation took 5.464824032s
Netlist generation took 0.134063316s

<no location info>: error:
    Clash error call:
    :1:24: error: expected: "~AND",
"~ARG",
"~ARGN",
"~COMPNAME",
"~DEPTH",
"~DEVNULL",
"~EARG",
"~ENDGENERATE",
"~ERESULT",
"~ERROR",
"~ERRORO",
"~FILE",
"~FROMBV",
"~GENERATE",
"~GENSYM",
"~INCLUDENAME",
"~INDEXTYPE",
"~ISGATED",
"~ISLIT",
"~ISSYNC",
"~ISVAR",
"~IW64",
"~LENGTH",
"~LIT",
"~NAME",
"~OTHERSYN",
"~OUTPUTWIREREG",
"~REPEAT",
"~RESULT",
"~SEL",
"~SIGD",
"~SIGDO",
"~SIZE",
"~STRCMP",
"~SYM",
"~TEMPLATE",
"~TOBV",
"~TYP",
"~TYPEL",
"~TYPM",
"~TYPMO",
"~TYPO",
"~VAR",
"~VARS",
"~VIVADO"
  assign ~RESULT = ~IF [( ~SIZE[~TYP[1]] 
                       ^                 
```

christiaanb, 2018-12-17:
@adamwalker could you try `master` with this patch:

```
--- a/clash-lib/src/Clash/Netlist/BlackBox/Util.hs
+++ b/clash-lib/src/Clash/Netlist/BlackBox/Util.hs
@@ -667,7 +667,7 @@ prettyElem (IF b esT esF) = do
      string "~FI")
 prettyElem (And es) =
   (renderLazy . layoutCompact) <$>
-  (brackets (tupled $ mapM (string <=< prettyElem) es))
+  (string "~AND" <> brackets (tupled $ mapM (string <=< prettyElem) es))
 prettyElem IW64 = return "~IW64"
 prettyElem (HdlSyn s) = case s of
   Vivado -> return "~VIVADO"
```

adamwalker, 2018-12-18:
No, but I get a different error:
```
<no location info>: error:
    Clash error call:
    :1:29: error: expected: "]",
"~AND",
"~ARG",
"~ARGN",
"~COMPNAME",
"~DEPTH",
"~DEVNULL",
"~EARG",
"~ENDGENERATE",
"~ERESULT",
"~ERROR",
"~ERRORO",
"~FILE",
"~FROMBV",
"~GENERATE",
"~GENSYM",
"~INCLUDENAME",
"~INDEXTYPE",
"~ISGATED",
"~ISLIT",
"~ISSYNC",
"~ISVAR",
"~IW64",
"~LENGTH",
"~LIT",
"~NAME",
"~OTHERSYN",
"~OUTPUTWIREREG",
"~REPEAT",
"~RESULT",
"~SEL",
"~SIGD",
"~SIGDO",
"~SIZE",
"~STRCMP",
"~SYM",
"~TEMPLATE",
"~TOBV",
"~TYP",
"~TYPEL",
"~TYPM",
"~TYPMO",
"~TYPO",
"~VAR",
"~VARS",
"~VIVADO"
  assign ~RESULT = ~IF ~AND[( ~SIZE[~TYP[1]] 
                            ^                
```

christiaanb, 2018-12-18:
Could you try commit bc90da3709cd2f532a5a8984c0b4c35051eddb7c from https://github.com/clash-lang/clash-compiler/pull/453?

adamwalker, 2018-12-18:
Slightly different error. The last line is:
```
assign ~RESULT = ~IF ~AND[ ~SIZE[~TYP[1]] 
```

christiaanb, 2018-12-18:
Okay, found a minimal example:

```haskell
module Test where

import Clash.Prelude

topEntity
  :: Vec 4 (Unsigned 4)
  -> Vec 4 (Unsigned 5)
topEntity = map (add (3 :: Unsigned 4))
```

I'll try to work from there

christiaanb, 2018-12-18:
@adamwalker hopefully fixed by: 4114a9eeb345d2f2b00570190c212406e5698ca6

adamwalker, 2018-12-18:
Yep. Works now. Thanks!

commit 1aa25180 (2018-12-18): Merge pull request #453 from clash-lang/fix-prettyelem-and

Fix `Clash.Netlist.BlackBox.Util.prettyElem` for `And`

Fixes #452
closed 2018-12-18 by adamwalker