clash-lang/clash-compiler #459: How do I install CLaSH from Git?

gergoerdi, 2019-01-04:
I am trying to install CLaSH from Git, since I would like to start using it in situations where I cannot change the cwd to the CLaSH source dir. Unfortunately, there is no `stack.yaml` file in the repo yet, so my only option seems to be Cabal.

So I did `cabal new-install` in the main CLaSH dir, which in turn installed all the packages. The problems start when I try to use it.

For reference, here's how I can use CLaSH 0.99.3, installed from Stackage via `stack`, to compile my code:

```
stack exec --resolver=lts-12.4 -- \
  clash -isrc-clash -outputdir _build -i../../clash-utils/src-clash \
  -Wno-partial-type-signatures \
  --vhdl \
  src-clash/Serial.hs
```

Now let's try the same after running `cabal new-install all` . Right off the bat, it fails because `cabal new-install` didn't install the library dependencies globally:

```
$ ~/.cabal/bin/clash -isrc-clash -outputdir _build -i../../clash-utils/src-clash \
  -Wno-partial-type-signatures \
  --vhdl \
  src-clash/Serial.hs

<no location info>: error:
    Other error:
    <command line>: Could not find module ‘GHC.TypeLits.Normalise’

```

OK so I try `cabal new-install all --lib`, which then does install the library dependencies, but still doesn't work because all those packages are hidden:

```
<no location info>: error:
    Other error:
    <command line>: Could not load module ‘GHC.TypeLits.Normalise’
It is a member of the hidden package ‘ghc-typelits-natnormalise-0.6.2’.
You can run ‘:set -package ghc-typelits-natnormalise’ to expose it.
(Note: this unloads all the modules in the current scope.)
```

Also, at this point `cabal` managed to break my `stack`-installed GHC, not a small feat:

```
$ ghc-env 8.6.3 
ghc-env: The GHC located at $HOME/.stack/programs/x86_64-linux/ghc-8.6.3/bin/ghc failed to compile a sanity check. Please see:

    http://docs.haskellstack.org/en/stable/install_and_upgrade.html

for more information. Exception was:
Running $HOME/.stack/programs/x86_64-linux/ghc-8.6.3/bin/ghc /tmp/stack-sanity-check19014/Main.hs -no-user-package-db in directory /tmp/stack-sanity-check19014/ exited with ExitFailure 1

[1 of 1] Compiling Main             ( /tmp/stack-sanity-check19014/Main.hs, /tmp/stack-sanity-check19014/Main.o )

Loaded package environment from $HOME/.ghc/x86_64-linux-8.6.3/environments/default

/tmp/stack-sanity-check19014/Main.hs:1:1: error:
    Could not load module ‘Distribution.Simple’
    It is a member of the hidden package ‘Cabal-2.4.0.1’.
    You can run ‘:set -package Cabal’ to expose it.
    (Note: this unloads all the modules in the current scope.)
    Use -v to see a list of the files searched for.
  |
1 | import Distribution.Simple
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^
```

gergoerdi, 2019-01-04:
I have also tried running the Cabal-installed `clash` inside the `stack` environment (after deleting `~/.ghc/x86_64-linux-8.6.3` to get out of the `Cabal-2.4.0.1` mess), i.e.

```
stack exec -- \
  ~/.cabal/bin/clash -isrc-clash -outputdir _build -i../../clash-utils/src-clash \
  -Wno-partial-type-signatures \
  --vhdl \
  src-clash/Serial.hs
```

which, promisingly, does find the installed version of the GHC type-level computation plugins, but still fails on the Clash-specific modules:

```
../../clash-utils/src-clash/Cactus/Clash/Explicit/Clock.hs:6:1: error:
    Could not find module ‘Clash.Prelude’
    Use -v to see a list of the files searched for.
  |
6 | import Clash.Prelude hiding (clkPeriod)
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

../../clash-utils/src-clash/Cactus/Clash/Explicit/Clock.hs:7:1: error:
    Could not find module ‘Clash.Signal.Internal’
    Use -v to see a list of the files searched for.
  |
7 | import Clash.Signal.Internal (Clock(..))
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

```

martijnbastiaan, 2019-01-04:
The following works (but is not ideal):

In the source directory run:

```
cabal new-run clash -v2
```

Note the clash binary location, in my case:

```
/home/martijn/Code/clash-compiler/dist-newstyle/build/x86_64-linux/ghc-8.6.2/clash-ghc-0.99/x/clash/build/clash/clash
```

Change to whatever directory you like:

```
cd $(mktemp -d)
```

Copy the ghc environment file (there's probably some better solution):

```
cp /home/martijn/Code/clash-compiler/.ghc.environment.x86_64-linux-8.6.2 .
```

And change it such that the `package-db` paths are absolute, in my case I had to change line 10:

```
package-db /home/martijn/code/clash-compiler/dist-newstyle/packagedb/ghc-8.6.2
```

Set datadirs:

```
export SRCDIR=/home/martijn/code/clash-compiler
export clash_ghc_datadir=${SRCDIR}/clash-ghc
export clash_lib_datadir=${SRCDIR}/clash-lib
export clash_prelude_datadir=${SRCDIR}/clash-prelude
export clash=${SRCDIR}/dist-newstyle/build/x86_64-linux/ghc-8.6.2/clash-ghc-0.99/x/clash/build/clash/clash
```

And finally run clash:

```
${clash} BlockRamTest.hs --vhdl
```

This is the best way I've found so far. And yes, we should probably improve it :-).

martijnbastiaan, 2019-01-04:
I've marked this as `enhancement` as there is obviously quite a bit to improve.

gergoerdi, 2019-01-04:
I now have made progress on this by just adding three `stack.yaml` files to the three directories `clash-prelude`, `clash-lib` and `clash-ghc`, with the following contents:

```
resolver: lts-12.14

extra-deps:
  - ghc-typelits-natnormalise-0.6.2
  - ghc-typelits-knownnat-0.6
  - ghc-typelits-extra-0.3
  - ../clash-prelude
  - ../clash-lib
```

with this setup, `stack install` seems to work.

gergoerdi, 2019-01-04:
No, I take it back -- it's not perfect yet. While all three `stack install`s succeed, importing `Clash.Driver.Types` afterwards in my project fails with no such module...

martijnbastiaan, 2019-01-04:
What about the workaround I posted. Does that work for you?

gergoerdi, 2019-01-04:
I haven't tried that one, I'm still fighting to get it working with Stack...

gergoerdi, 2019-01-05:
I almost have it working with an external `stack.yaml` file. I've added the following to the directory of my project:

```
resolver: lts-12.14

extra-deps:
  - ghc-typelits-natnormalise-0.6.2
  - ghc-typelits-knownnat-0.6
  - ghc-typelits-extra-0.3
  - git: git@github.com:clash-lang/clash-compiler
    commit: f2060a077
    subdirs:
    - clash-prelude
    - clash-lib
    - clash-ghc
```

and this would work, except that `stack build` barfs on the lack of a `packages` field. But if you manage to `stack build` somehow (e.g. by adding temporarily an unrelated directory to `packages`), then afterwards `stack exec clash` etc. works fine with the above file.

I've posted this question to [/r/haskell](https://reddit.com/r/haskell/comments/ab8ypl/monthly_hask_anything_january_2019/ed9rrl4/), hopefully someone who knows Stack better will have a solution.

martijnbastiaan, 2019-01-07:
/r/haskell indicates it's not possible, so that's a bummer. We'll work on binary distribution soon(tm), which will hopefully make building/installing Clash as a debian package easy.

gergoerdi, 2019-01-07:
I've come up with a workaround I feel I could live with. `stack build` and `stack run` all works with. Its only downside is that all projects using CLaSH this way will have their own copy of the source tree and the build artifacts.

1. Create a directory somewhere with the following [`package.yaml`](https://github.com/sol/hpack#readme) file:

```
name: empty-clash
version: 0.1.0
category: acme

dependencies:
  - clash-prelude
  - clash-lib
  - clash-ghc

source-dirs:
  - src
  
executables:
  empty:
    main: Main.hs
```

2. Use the following `stack.yaml` in your projects:
```
resolver: lts-12.14

packages:
  - /home/cactus/prog/clash/empty-project

extra-deps:
  - ghc-typelits-natnormalise-0.6.2
  - ghc-typelits-knownnat-0.6
  - ghc-typelits-extra-0.3
  - git: git@github.com:clash-lang/clash-compiler
    commit: b4e3d5e79ee5
    subdirs:
    - clash-prelude
    - clash-lib
    - clash-ghc
```

gergoerdi, 2019-01-07:
Turns out we can do one better: using a Git repo (containing a dummy package) instead of a local dummy package:
```
resolver: lts-12.14

packages:
  - location:
      git: https://github.com/gergoerdi/empty-clash
      commit: f5ce90ce1
    extra-dep: false

extra-deps:
  - ghc-typelits-natnormalise-0.6.2
  - ghc-typelits-knownnat-0.6
  - ghc-typelits-extra-0.3
  - git: git@github.com:clash-lang/clash-compiler
    commit: b4e3d5e79ee5
    subdirs:
    - clash-prelude
    - clash-lib
    - clash-ghc
```

martijnbastiaan, 2019-02-27:
Related: https://github.com/clash-lang/clash-compiler/issues/512

gergoerdi, 2020-04-04:
FWIW, these days I am using Stack resolvers:

```

```
resolver: lts-15.2
name: clash-1.2

packages:
  - clash-ghc-1.2.0
  - clash-prelude-1.2.0
  - clash-lib-1.2.0

  - concurrent-supply-0.1.7
```

```
name: clash-head
resolver: lts-15.2

packages:
  - github: clash-lang/clash-compiler
    commit: 702fea65584f57e0ff6cd19f3e80af92a27ed6e7    
    subdirs:
      - clash-ghc
      - clash-prelude
      - clash-lib

  - concurrent-supply-0.1.8

flags:
    clash-prelude:
      multiple-hidden: false
```

martijnbastiaan, 2020-11-09:
This has been solved by the [Clash starter project](https://github.com/clash-lang/clash-compiler/tree/master/clash-starters/clash-example-project). I'm still looking to improve the experience, one of the possibilities being support for [Stack templates](https://github.com/clash-lang/clash-compiler/issues/1582). When using git dependencies this does mean multiple locally installed Clash versions, like you noted. I think this is fair, though. Projects shouldn't rely on unspecified, global dependencies IMO.

 Thanks for being helpful as always @gergoerdi  :).

closed 2020-11-09 by martijnbastiaan