clash-lang/clash-compiler #460: Static division survives to VHDL

gergoerdi, 2019-01-05:
In the following minimized example, the `divider` argument passed to `foo` in `topEntity` is statically known:

```
{-# LANGUAGE PartialTypeSignatures #-}
module Serial where

import Clash.Prelude hiding (clkPeriod)
import qualified Cactus.Clash.Explicit.Clock as E
import Data.Word
import Data.Maybe (fromMaybe, isJust)
import Data.Proxy

type Dom32 = Dom "CLK_32MHZ" 31250

{-# NOINLINE topEntity #-}
topEntity
    :: Clock Dom32 Source
    -> Reset _ Asynchronous
    -> Signal _ Bit
    -> Signal _ Bit
topEntity = exposeClockReset board
  where
    board rxIn = foo (fromIntegral clkRate `div` serialRate) rxIn

serialRate :: Word32
serialRate = 9600

clkRate :: (HiddenClock domain gated, domain ~ Dom s ps, KnownNat ps) => Integer
clkRate = hideClock E.clkRate

foo :: (HiddenClockReset domain gated synchronous) => Word32 -> Signal domain Bit -> Signal domain Bit
foo divider = mealy step 0
  where
    step s x | s == divider = (0, x)
             | otherwise = (s + 1, low)
```

but, as of f2060a077, the division survives all the way to the VHDL, which is a problem on devices that don't have DSPs for division (e.g. the Papilio One running on a Spartan-3). And even on devices that do have hardware for division, I would be worried that the FPGA synthesis tools are not smart enough to eliminate them, and end up actually using up DSP resources.

VHDL output of the above (with `-fclash-intwidth=32`):

```
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.serial_types.all;

entity serial_topentity is
  port(-- clock
       clk  : in serial_types.clk_clk_32mhz;
       -- asynchronous reset: active high
       rst  : in serial_types.rst_clk_32mhz;
       eta1 : in std_logic;
       x    : out std_logic);
end;

architecture structural of serial_topentity is
  signal \#tup_case_alt\   : serial_types.tup2;
  signal \#tup_case_scrut\ : boolean;
  signal \#tup_app_arg\    : signed(31 downto 0);
  signal x1                : unsigned(31 downto 0);
  signal y                 : unsigned(31 downto 0);
  signal s                 : unsigned(31 downto 0);
  signal x_0               : unsigned(31 downto 0);
  signal \#w\              : unsigned(31 downto 0);
  signal \#w_0\            : unsigned(31 downto 0);
  signal y_projection      : unsigned(31 downto 0);

begin
  x <= \#tup_case_alt\.tup2_sel1_std_logic;

  \#w\ <= (x1 + to_unsigned(1,32));

  \#tup_case_alt\ <= ( tup2_sel0_unsigned => resize(to_unsigned(0,32),32)
                     , tup2_sel1_std_logic => eta1 ) when \#tup_case_scrut\ else
                     ( tup2_sel0_unsigned => resize((resize(\#w\(31 downto 0),32)),32)
                     , tup2_sel1_std_logic => '0' );

  \#tup_case_scrut\ <= tagToEnum(\#tup_app_arg\);

  \#tup_app_arg\ <= to_signed(1,32) when x1 = y else to_signed(0,32);

  x1 <= s;

  \#w_0\ <= (unsigned(std_logic_vector(to_signed(32000000,32))));

  y_projection <= resize(((resize(\#w_0\(31 downto 0),32)) / to_unsigned(9600,32)),32);

  y <= y_projection;

  -- register begin 
  serial_topentity_register : process(clk,rst)
  begin
    if rst = '1' then
      s <= (resize(to_unsigned(0,32),32))
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    elsif rising_edge(clk) then
      s <= x_0
      -- pragma translate_off
      after 1 ps
      -- pragma translate_on
      ;
    end if;
  end process;
  -- register end

  x_0 <= \#tup_case_alt\.tup2_sel0_unsigned;
end;
```

The money shot:

```
  \#w_0\ <= (unsigned(std_logic_vector(to_signed(32000000,32))));

  y_projection <= resize(((resize(\#w_0\(31 downto 0),32)) / to_unsigned(9600,32)),32);
```

gergoerdi, 2019-01-05:
Oh sorry, left out the definition of `E.clkRate`:

```
module Cactus.Clash.Explicit.Clock
    ( clkPeriod
    , clkRate
    ) where

import Clash.Prelude hiding (clkPeriod)
import Clash.Signal.Internal (Clock(..))

clkPeriod :: forall s ps gated. (KnownNat ps) => Clock (Dom s ps) gated -> Integer
clkPeriod clk = snatToInteger (SNat @ps)

clkRate :: (KnownNat ps) => Clock (Dom s ps) gated -> Integer
clkRate clk = 10^12 `div` clkPeriod clk

```

martijnbastiaan, 2019-01-07:
Thanks for the report!

As a workaround, you can use something like:

```haskell
$([| fromIntegral clkRate `div` serialRate |])
```

gergoerdi, 2019-01-19:
OK I managed to get it working, but in a way that takes the wind out of the whole type-directed clock rate matching idea.

1. Simplify `clkRate` so that it takes no term-level parameters:

```
clkRate :: forall proxy s ps gated. (KnownNat ps) => proxy (Clock (Dom s ps) gated) -> Integer
clkRate p = 10^12 `div` clkPeriod p

clkPeriod :: forall proxy s ps gated. (KnownNat ps) => proxy (Clock (Dom s ps) gated) -> Integer
clkPeriod _ = snatToInteger (SNat @ps)
```

2. Use TH to force its evaluation at the use site, passing an explicit `Proxy` of the right clock:

```
type Dom32 = Dom "CLK_32MHZ" 31250

{-# NOINLINE topEntity #-}
topEntity
    :: Clock Dom32 Source
    -> Reset _ Asynchronous
    -> Signal _ Bit
    -> Signal _ Bit
topEntity = exposeClockReset board
  where
    board rxIn = foo $(lift $ fromIntegral (clkRate (Proxy :: Proxy (Clock (Dom "CLK_32MHZ" 31250) Source))) `div` serialRate) rxIn
```

This "works" in the sense that the generated VHDL no longer contains a division; but now the connection to `Dom32` is severed (for example, one can easily change the type-level natural in the `clkRate` call and it will still typecheck).

gergoerdi, 2019-01-19:
We can enforce (but, AFAICT, not infer) the correct clock rate by using TH to generate a term with the right implicit argument:

```
import Clash.Prelude
import Data.Word
import Language.Haskell.TH

staticClockDivider :: Word32 -> Word32 -> Q Exp
staticClockDivider clk wantedRate = [|hideClock ((\_ -> $(lift rate)) :: Clock (Dom s $clk') gated -> Word32)|]
  where
    clk' = litT . numTyLit . fromIntegral $ clk
    rate = clk `div` wantedRate
```

and then it can be used in another module like this:

```
type Dom32 = Dom "CLK_32MHZ" 31250

topEntity
    :: Clock Dom32 Source
    -> Reset _ Asynchronous
    -> Signal _ Bit
    -> Signal _ Bit
topEntity = exposeClockReset board
  where
    board rxIn = foo $(staticClockDivider 31250 serialRate) rxIn
```

As desired, this fails to typecheck if the literal `31250` is changed.

gergoerdi, 2019-01-19:
We could also change `staticClockDivider` so that it can be passed `Dom32` but it feels like we're just digging ourselves into deeper and deeper holes:

```
staticClockDivider' :: Name -> Word32 -> Q Exp
staticClockDivider' dom wantedRate = do
    info <- reify dom
    clk <- case info of
        TyConI (TySynD _ _ (AppT _ (LitT (NumTyLit clk)))) -> return clk
        _ -> fail "don't know how to get clock rate"
    let rate = fromInteger clk `div` wantedRate
    [| let f :: Clock $(conT dom) gated -> Word32
           f _ = $(lift rate)
       in hideClock f
     |]
```

this allows for

```
topEntity = exposeClockReset board
  where
    board rxIn = foo $(staticClockDivider' ''Dom32 serialRate) rxIn
```

but of course [now `Dom32` has to be in a preceding declaration block](https://stackoverflow.com/a/37088312/477476), so we need to separate it with a

    $(return [])

TH splice.

Clearly, the juice is not worth the squeeze at this point. I thought one of the main ideas behind CLaSH is to do this kind of static reduction at compile time?

martijnbastiaan, 2019-01-30:
> I thought one of the main ideas behind CLaSH is to do this kind of static reduction at compile time?

Well, it is. The evaluator (https://github.com/clash-lang/clash-compiler/blob/master/clash-ghc/src-ghc/Clash/GHC/Evaluator.hs and https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/Clash/Core/Evaluator.hs) should probably be extended such that it correctly constant folds your case. We're interested in fixing this bug, and have therefore marked it as such.

Thanks for the report and your effort!

gergoerdi, 2019-01-30:
OK so I looked at this in detail based on your helpful comment above.

According to `-ddump-simpl`, the division coming from ```fromIntegral clkRate `div` serialRate``` looks like this:

         GHC.Word.W32#
           (GHC.Prim.quotWord# (GHC.Prim.narrow32Word# wild_aoHF) 9600##)

The evaluator in https://github.com/clash-lang/clash-compiler/blob/9ce093c2db1acb30247e9da61f4bef101adf1ae1/clash-ghc/src-ghc/Clash/GHC/Evaluator.hs does have implementations for `GHC.Prim.quotWord#` and `GHC.Prim.narrow32Word#`, so the pieces are there. But for some reason, this combination doesn't trigger them?

If I change my code to use `Word64`, GHC's output becomes simpler since there's no narrowing needed:

         GHC.Word.W64# (GHC.Prim.quotWord# wild_aoHG 9600##)

But even with this version, I can see the division in the VHDL  generated by CLaSH.

However, if I use `Int` in the types of `serialRate` and `foo`, the generated Core looks like this:

         case GHC.Classes.divInt# wild_aorV 9600# of ww4_ao3O { __DEFAULT ->
         GHC.Types.I# ww4_ao3O
         }

and this one is correctly reduced by CLaSH to

    \#tup_app_arg\ <= to_signed(1,64) when x1 = to_signed(3333,64) else to_signed(0,64);

gergoerdi, 2019-01-30:
As an aside, I've found a much nicer way to do what I was trying to do here, by doing all the computation on the type level. I'm putting it here in case someone else finds it useful:

```
type FromHz rate = 1000000000000 `Div` rate

type family ClockDivider (n :: Nat) dom where
    ClockDivider n (Dom s ps) = n `Div` ps

foo
    :: forall n domain gated synchronous.
       (KnownNat n, KnownNat (ClockDivider n domain), HiddenClockReset domain gated synchronous)
    => SNat n -> Signal domain Bit -> Signal domain Bit
foo _ = mealy step 0
  where
    step :: Index (ClockDivider n domain) -> _ -> _
    step s x | s == maxBound = (0, x)
             | otherwise = (s + 1, low)

```

Then it can be used like this:

```
type Dom32 = Dom "CLK_32MHZ" (FromHz 32000000)

{-# NOINLINE topEntity #-}
topEntity
    :: Clock Dom32 Source
    -> Reset _ Asynchronous
    -> Signal _ Bit
    -> Signal _ Bit
topEntity = exposeClockReset board
  where
    board rxIn = foo serialRate rxIn

serialRate :: SNat (FromHz 9600)
serialRate = SNat

```

martijnbastiaan, 2019-01-30:
While `wild_aoHF` is a constant, it's not a literal. I believe that's why the Evaluator doesn't catch it. I'm not sure how simple this is to solve. I'm currently working on something related (register power up values that HAVE to be a literal on HDL level), so maybe that will make it easier to solve this issue.

christiaanb, 2019-01-30:
Seems the evaluator is missing a rule for `GHC.Integer.Type.integerToWord`

commit 480cc137 (2019-02-22): Evaluator: add rule for integerToWord

Fixes #460
commit 7c5f9644 (2019-02-25): Evaluator: add rule for integerToWord

Fixes #460
commit 372a7863 (2019-03-08): Evaluator: add rule for integerToWord

Fixes #460
closed 2019-03-08 by martijnbastiaan