clash-lang/clash-compiler #468: No blackbox found for: GHC.Natural.plusNatural

gergoerdi, 2019-01-21:
I am using CLaSH 949a05a1d8b1e4b3aabf224248c3e0e124c8fc39. My full standalone code is:

```
module Serial where

import Clash.Prelude
import Control.Monad.State

topEntity
    :: Clock System Source
    -> Reset System Asynchronous
    -> Signal System Bit
    -> Signal System Bit
topEntity = exposeClockReset board
  where
    board btn0 = boolToBit <$> click
      where
        btn = debounce d16 False . fmap (bitToBool . complement) $ btn0
        click = isRising maxBound btn

mealyState :: (HiddenClockReset domain gated synchronous, Undefined s)
           => (i -> State s o) -> s -> (Signal domain i -> Signal domain o)
mealyState f = mealy step
  where
    step s x = let (y, s') = runState (f x) s
               in (s', y)

unsigned :: (KnownNat n) => SNat n -> Unsigned n -> Unsigned n
unsigned n = id

debounce
    :: (HiddenClockReset domain gated synchronous, KnownNat n, Eq a, Undefined a)
    => SNat n -> a -> Signal domain a -> Signal domain a
debounce n x0 = mealyState step (unsigned n 0, x0, x0)
  where
    step this = do
        (counter, last, prev) <- get
        let stable = counter == maxBound
            changing = this /= prev
            counter' = if changing then 0 else counter `boundedAdd` 1
            last' = if counter' == maxBound then this else last

        put (counter', last', this)
        return last'
```

The error message I get with the above, when using `clash --vhdl`:

```
<no location info>: error:
    Clash.Netlist.BlackBox(337): No blackbox found for: GHC.Natural.plusNatural
```

gergoerdi, 2019-01-21:
Adding an `INLINE` pragma to `debounce` also seems to work as an effective workaround.

christiaanb, 2019-01-21:
@gergoerdi what version of GHC are you using?

gergoerdi, 2019-01-21:
Should be 8.6.3 if I'm not mistaken (since I'm using `lts-13.3` as per the `stack.yaml` file in the repo).

martijnbastiaan, 2019-01-21:
The problem is three-fold:

1. We don't have blackboxes for `GHC.Natural.*`
2. .. but even if we did we should have constant folded it in `Clash.Core.Evaluator` thus not needing the blackboxes
3. ..but even if we did we shouldn't have used `~LIT` on `KnownNat`s in our blackboxes, triggering the need to constant fold naturals. 

I'll go over all of these.

martijnbastiaan, 2019-01-21:
GHC Naturals in evaluator: https://github.com/clash-lang/clash-compiler/commit/007b65fa7ed3e4ef80974dde44d48b92b04b8c98

christiaanb, 2019-01-29:
Fixed by #473

closed 2019-01-29 by christiaanb