clash-lang/clash-compiler #475: Compilation error

G-Pereira, 2019-01-25:
I am getting the following error:
`*** Exception: Expected: BlackBox or Primitive object
CallStack (from HasCallStack):
  error, called at src/Clash/Primitives/Types.hs:69:17 in clash-lib-0.99.3-FUua9GHL8Gd1xsTjreXzaa:Clash.Primitives.Types`

G-Pereira, 2019-01-25:
only when doing ´:vhdl´

leonschoorl, 2019-01-25:
Something is failing during the parsing of the HDL primitives json files.

Some more context about what you're doing would be helpful:
* are you trying to write your own primitive? if so could you post its source?
* only `:vdhl` fails, but does `:verilog` work, or `clash --vhdl`?

G-Pereira, 2019-01-25:
```
$ clash --vhdl Documents/Polynomial/Polynomial/Polynomial.hs 

<no location info>: error:
    Clash error call:
    Expected: BlackBox or Primitive object
```

Polynomial.hs:
```
module Polynomial where

import Clash.Prelude

-- type System is defined in 'Clash.Explicit.Signal' as: Dom "system" 10000
-- Hence System is a domain identified by 'system' and clock period 10ns
-- Ungated (due to Source) system (read domain) clock
type Clk = Clock System Source

-- Asynchronous reset for system domain
type Rst = Reset System Asynchronous
type Sig = Signal System

type Value = Signed 16
type Vector = Vec 3 Value

-- Student information:
--  Student 1
--    Pereira
--    s
--  Student 2
--    Jonkman
--    s1563599

-- NOTE: topentity functions are at the bottom
-- comment and uncomment the functions for each assignment

-----------------------------------------------------------
-- Assignment 1
-- f0p f0m
-----------------------------------------------------------
f0p :: Value -> Value
f0p x = x^4 + 22*x^3 + 172*x^2 + 552*x + 576

f0m :: Value -> Value
f0m x = x*x*x*x + 22*x*x*x + 172*x*x + 552*x + 576

f0mb :: Value -> Value
f0mb x = (x*x*x*x) + 22*(x*x*x) + 172*(x*x) + 552*x + 576
-----------------------------------------------------------
-- Assignment 2
-- f0f
-----------------------------------------------------------
f0f :: Value -> Value
f0f x = (x + 2) * (x + 6) * (x + 6) * (x + 8)
-----------------------------------------------------------
-- Assignment 3
-- f0hof
----------------------------------------------------------
helper x y z = y*(x+z)

--f0hof :: Value -> Value
f0hof x = foldl (helper x) ((2+x) :: Value) (6:>6:>8:>Nil)

-----------------------------------------------------------
-- Assignment 4
-- f0mealy
----------------------------------------------------------
f0helper s (x, y) = (s', z)
  where
    s'= z*(x+y)
    z = s'

f0mealy x = mealy f0helper 1 x

-----------------------------------------------------------
-- topEntity's
-----------------------------------------------------------

-- Assignment 1
--topEntity = f0p

-- Assignment 2
--topEntity = f0f

-- Assignment 3
--topEntity = f0hof

-- Assignment 4
topEntity :: Clk -> Rst -> Sig (Value, Value) -> Sig Value
topEntity clk rst x = exposeClockReset f0mealy clk rst x
```

G-Pereira, 2019-01-25:
verilog gives the same error

leonschoorl, 2019-01-25:
Are there maybe any `.json` files in your current directory?

G-Pereira, 2019-01-25:
No, not at all

leonschoorl, 2019-01-25:
I've added some better error reporting in this branch:
https://github.com/clash-lang/clash-compiler/tree/better-json-parse-errors

Could you try:
```
git clone https://github.com/clash-lang/clash-compiler.git
cd clash-compiler
git checkout better-json-parse-errors
cabal new-run -- clash --vhdl PATH_TO_YOUR_DESIGN.hs
```

G-Pereira, 2019-01-25:
```
/home/gpereira/clash-compiler/dist-newstyle/build/x86_64-linux/ghc-8.2.2/clash-ghc-0.99/doc/html/clash-ghc/index.html
Configuring executable 'clash' for clash-ghc-0.99..
Preprocessing executable 'clash' for clash-ghc-0.99..
Building executable 'clash' for clash-ghc-0.99..
[1 of 1] Compiling Main             ( src-ghc/Batch.hs, /home/gpereira/clash-compiler/dist-newstyle/build/x86_64-linux/ghc-8.2.2/clash-ghc-0.99/x/clash/build/clash/clash-tmp/Main.dyn_o )
Linking /home/gpereira/clash-compiler/dist-newstyle/build/x86_64-linux/ghc-8.2.2/clash-ghc-0.99/x/clash/build/clash/clash ...
Loading dependencies took 1.467311412s

<no location info>: error:
    Clash error call:
    Encountered compilation errors while compiling blackbox template function "BBFN<Clash.Primitives.Intel.ClockGen.altpllTF>" for function "Clash.Intel.ClockGen.altpll". Compilation reported: 

<no location info>: error:
    module ‘Clash.Primitives.Intel.ClockGen’ cannot be found locally
```

G-Pereira, 2019-01-25:
but it doesn't synthesize in Quartus 18.1

G-Pereira, 2019-01-25:
I can give Some more detail about that:
```
Info: *******************************************************************
Info: Running Quartus Prime Analysis & Synthesis
	Info: Version 18.1.0 Build 625 09/12/2018 SJ Lite Edition
	Info: Processing started: Fri Jan 25 16:31:31 2019
Info: Command: quartus_map --read_settings_files=on --write_settings_files=off Polynomial -c Polynomial
Warning (18236): Number of processors has not been specified which may cause overloading on shared machines.  Set the global assignment NUM_PARALLEL_PROCESSORS in your QSF to an appropriate value for best performance.
Info (20030): Parallel compilation is enabled and will use 4 of the 4 processors detected
Info (12021): Found 2 design units, including 0 entities, in source file polynomial_types.vhdl
	Info (12022): Found design unit 1: polynomial_types
	Info (12022): Found design unit 2: polynomial_types-body
Info (12021): Found 2 design units, including 1 entities, in source file polynomial_topentity.vhdl
	Info (12022): Found design unit 1: polynomial_topentity-structural
	Info (12023): Found entity 1: polynomial_topentity
Info (12127): Elaborating entity "polynomial_topentity" for the top level hierarchy
```
It hangs here at 9% of "Compile Design"

leonschoorl, 2019-01-25:
> worked. What does that mean?

Either something is wrong with the old version of clash and it's fixed in git, or something is wrong with your setup.
It seems there is something wrong with a json file, you could try:
```
strace -f clash --vhdl Documents/Polynomial/Polynomial/Polynomial.hs |& grep json
```
To see which ones clash is reading.

leonschoorl, 2019-01-25:
As to why it's not synthesizing:
```haskell
f0helper s (x, y) = (s', z)
  where
    s'= z*(x+y)
    z = s'
```
inlining 'z' we get:
```haskell
    s'= s'*(x+y)
```
So `s'` is recursive in itself

Which gets translated to VHDL:
```vhdl
  \#s'_rec\ <= resize(\#s'_rec\ * (x_2 + y), 16);
```

You probably want something like:
```haskell
f0helper s (x, y) = (s', z)
  where
    s'= z*(x+y)
    z = s
```

G-Pereira, 2019-01-25:
> Either something is wrong with the old version of clash and it's fixed in git, or something is wrong with your setup.
> It seems there is something wrong with a json file, you could try:
>
> ```
> strace [...]

```
<no location info>: error:
    Other error:
    user interrupt
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Control_Exception_Base.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 3
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_TypeNats.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 4
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_GHC_GHC2Core.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 5
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_Prim.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 6
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Debug_Trace.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 7
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_CString.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 8
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_Typelits.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 9
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_Explicit_Signal.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 10
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Unsafe_Coerce.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 11
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_Magic.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 12
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_IO_Exception.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 13
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_XException.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 14
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_Promoted_Symbol.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 15
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_Real.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 16
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_Err.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 17
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_Class_BitPack.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 18
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_Promoted_Nat_Unsafe.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 19
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_Transformations.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 20
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/GHC_Natural.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 21
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/common/Clash_Signal_Bundle.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 22
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Explicit_BlockRam.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 23
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/GHC_Types.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 24
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Explicit_RAM.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 25
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/GHC_Integer_Type.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 26
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Sized_Internal_Index.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 27
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/GHC_Int.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 28
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/GHC_Base.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 29
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/GHC_Word.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 30
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/GHC_Integer_Logarithms.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 31
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/GHC_Prim.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 32
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Xilinx_ClockGen.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 33
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Prelude_ROM.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 34
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Promoted_Nat.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 35
[... cut ...]
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Explicit_ROM.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 49
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Intel_ClockGen.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 50
openat(AT_FDCWD, "/home/gpereira/.cabal/share/x86_64-linux-ghc-8.2.2/clash-lib-0.99.3/prims/vhdl/Clash_Prelude_ROM_File.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 51
openat(AT_FDCWD, "./.v8flags.7.0.276.38-node.13.6d3fc6f20180e213bd88f0f2e682c7c4.json", O_RDONLY|O_NOCTTY|O_NONBLOCK) = 52
```