clash-lang/clash-compiler #477: `any (x ==) xs`  vs. `any (== x) xs`

gergoerdi, 2019-01-26:
With CLaSH 949a05a1d8b1e4b3aabf224248c3e0e124c8fc39 the behavior of the following code is quite strange in at least the Intel/Altera ModelSim:

```
i2cMaster
    :: forall domain gated synchronous. (HiddenClockReset domain gated synchronous)
    => Word32 -> Signal domain Bit
i2cMaster halfDiv = scl
  where
    phase = regEn (0 :: Unsigned 2) (countTo halfDiv) (phase + 1)
    scl = boolToBit . (\phi -> any (== phi) [0, 1]) <$> phase
```

with this version, the RTL-level simulation of `SCL` produces a waveform containing a 1ps glitch at time 360001 (and then repeated at later times): https://gist.github.com/gergoerdi/07d601acbf28d46ae6e9b2d6399a58d1#file-eq_phi-vcd

If I replace the definition of `scl` with

```
    scl = boolToBit . (\phi -> any (phi ==) [0, 1]) <$> phase
```

then there is no glitch visible in the waveform: https://gist.github.com/gergoerdi/07d601acbf28d46ae6e9b2d6399a58d1#file-phi_eq-vcd

gergoerdi, 2019-01-26:
Actually this can be simplified even further: `(\phi -> 0 == phi || 1 == phi) <$> phase` already causes the weird blip.

gergoerdi, 2019-01-26:
Generated VHDL for `any (== phi)`:

```
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.i2ctest_types.all;

entity I2CTest is
  port(-- clock
       CLK_50MHZ : in i2ctest_types.clk_system;
       -- asynchronous reset: active high
       RESET     : in i2ctest_types.rst_system;
       SCL       : out std_logic);
end;

architecture structural of I2CTest is
  signal \#case_alt\               : std_logic;
  signal phase                     : unsigned(1 downto 0);
  signal \#phase_app_arg\          : i2ctest_types.clk_gated_system;
  signal \#phase_app_arg_0\        : boolean;
  signal \$d(%,%)1\                : i2ctest_types.product;
  signal \#SCL_selection_res\      : boolean;
  signal \#SCL_sel_alt_res\        : std_logic;
  signal \#i_0\                    : signed(63 downto 0);
  signal \#i\                      : signed(63 downto 0);
  signal \#case_alt_selection_res\ : boolean;
  signal \#case_alt_sel_alt_res\   : std_logic;
  signal \#i_2\                    : signed(63 downto 0);
  signal \#i_1\                    : signed(63 downto 0);
  signal \#case_alt_sel_alt_res_0\ : std_logic;
  signal \#i_4\                    : signed(63 downto 0);
  signal \#i_3\                    : signed(63 downto 0);

begin
  \#SCL_selection_res\ <= to_unsigned(0,2) = phase;

  \#i_0\ <= to_signed(0,64);

  \#i\ <= to_signed(1,64);

  \#SCL_sel_alt_res\ <= \#i\(0) when '0' = \#i_0\(0) else '-';

  SCL <= \#SCL_sel_alt_res\ when \#SCL_selection_res\ else
         \#case_alt\;

  \#case_alt_selection_res\ <= to_unsigned(1,2) = phase;

  \#i_2\ <= to_signed(0,64);

  \#i_1\ <= to_signed(0,64);

  \#case_alt_sel_alt_res\ <= \#i_1\(0) when '0' = \#i_2\(0) else '-';

  \#i_4\ <= to_signed(0,64);

  \#i_3\ <= to_signed(1,64);

  \#case_alt_sel_alt_res_0\ <= \#i_3\(0) when '0' = \#i_4\(0) else '-';

  \#case_alt\ <= \#case_alt_sel_alt_res_0\ when \#case_alt_selection_res\ else
                 \#case_alt_sel_alt_res\;

  -- register begin
  i2ctest_register : block
    signal clk : std_logic;
    signal ce : boolean;
  begin
    (clk,ce) <= \#phase_app_arg\;
    i2ctest_reg : process(clk,RESET)
    begin
      if RESET = '1' then
        phase <= to_unsigned(0,2);
      elsif rising_edge(clk) then
        if ce then
          phase <= (phase + to_unsigned(1,2))
          -- pragma translate_off
          after 1 ps
          -- pragma translate_on
          ;
        end if;
      end if;
    end process;
  end block;
  -- register end

  -- clockGate begin
  \#phase_app_arg\ <= (CLK_50MHZ,\#phase_app_arg_0\);
  -- clockGate end

  cactus_clash_util_countto_phase_app_arg_0 : entity cactus_clash_util_countto
    port map
      ( \#finished_case_alt\ => \#phase_app_arg_0\
      , \$d(%,%)\            => \$d(%,%)1\ );

  \$d(%,%)1\ <= ( product_sel0_clk_system => CLK_50MHZ
                , product_sel1_rst_system => RESET );
end;
```

Generated VHDL for `any (phi ==)`:

```
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.i2ctest_types.all;

entity I2CTest is
  port(-- clock
       CLK_50MHZ : in i2ctest_types.clk_system;
       -- asynchronous reset: active high
       RESET     : in i2ctest_types.rst_system;
       SCL       : out std_logic);
end;

architecture structural of I2CTest is
  signal phase                : unsigned(1 downto 0);
  signal \#phase_app_arg\     : i2ctest_types.clk_gated_system;
  signal \#phase_app_arg_0\   : boolean;
  signal \$d(%,%)1\           : i2ctest_types.product;
  signal \#SCL_sel_alt_res\   : std_logic;
  signal \#i_0\               : signed(63 downto 0);
  signal \#i\                 : signed(63 downto 0);
  signal \#SCL_sel_alt_res_0\ : std_logic;
  signal \#i_2\               : signed(63 downto 0);
  signal \#i_1\               : signed(63 downto 0);
  signal \#SCL_sel_alt_res_1\ : std_logic;
  signal \#i_4\               : signed(63 downto 0);
  signal \#i_3\               : signed(63 downto 0);

begin
  \#i_0\ <= to_signed(0,64);

  \#i\ <= to_signed(1,64);

  \#SCL_sel_alt_res\ <= \#i\(0) when '0' = \#i_0\(0) else '-';

  \#i_2\ <= to_signed(0,64);

  \#i_1\ <= to_signed(1,64);

  \#SCL_sel_alt_res_0\ <= \#i_1\(0) when '0' = \#i_2\(0) else '-';

  \#i_4\ <= to_signed(0,64);

  \#i_3\ <= to_signed(0,64);

  \#SCL_sel_alt_res_1\ <= \#i_3\(0) when '0' = \#i_4\(0) else '-';

  with (phase) select
    SCL <= \#SCL_sel_alt_res\ when "00",
           \#SCL_sel_alt_res_0\ when "01",
           \#SCL_sel_alt_res_1\ when others;

  -- register begin
  i2ctest_register : block
    signal clk : std_logic;
    signal ce : boolean;
  begin
    (clk,ce) <= \#phase_app_arg\;
    i2ctest_reg : process(clk,RESET)
    begin
      if RESET = '1' then
        phase <= to_unsigned(0,2);
      elsif rising_edge(clk) then
        if ce then
          phase <= (phase + to_unsigned(1,2))
          -- pragma translate_off
          after 1 ps
          -- pragma translate_on
          ;
        end if;
      end if;
    end process;
  end block;
  -- register end

  -- clockGate begin
  \#phase_app_arg\ <= (CLK_50MHZ,\#phase_app_arg_0\);
  -- clockGate end

  cactus_clash_util_countto_phase_app_arg_0 : entity cactus_clash_util_countto
    port map
      ( \#finished_case_alt\ => \#phase_app_arg_0\
      , \$d(%,%)\            => \$d(%,%)1\ );

  \$d(%,%)1\ <= ( product_sel0_clk_system => CLK_50MHZ
                , product_sel1_rst_system => RESET );
end;

```

christiaanb, 2019-01-28:
The 1ps blip is probably caused by the `after 1 ps` part of the blackbox for `register`. This is needed to work around delta-delay issues when packing clock lines into structs (tuples) and later unpacking them.

gergoerdi, 2019-01-28:
I have to admit I don't understand what delta-delay is; but, regardless of which behaviour is the correct one, shouldn't
    
    (\x -> any (== x) [0, 1]) <$> sig

and

    (\x -> any (x ==) [0, 1]) <$> sig

behave the same?

martijnbastiaan, 2019-01-30:
I think what @christiaanb is trying to say is that in order for simulation to work, we induce a 1ps delay in certain constructs. When synthesizing, these delay-pragmas will be stripped away, thus yielding the exact same behavior in hardware anyway. This has the unfortunate side-effect that simulation is not as pretty as it could be, as you just noted, but it shouldn't influence your circuit's behavior. 

To answer your question more directly: as long as the signal settles correctly before the next clock, the behavior *is* exactly the same as far as all registers are concerned. Since it's only a hiccup of 1ps, you shouldn't ever see behavioral changes, unless you're clocking at 1THz.

martijnbastiaan, 2019-02-12:
To solve this issue we should make sure that tuples don't occur in tuples (on an HDL level). This is a prerequisite for https://github.com/clash-lang/clash-compiler/issues/418 too. I'm adding this as a 1.1 milestone.

commit 554dfb83 (2019-10-18): Seperate out clock arguments from product types

Turns e.g.

> f :: (Clock System, Reset System) -> Signal System Int

into

> f :: Clock System -> Reset System -> Signal System Int

Fixes #477
Partially solves #418
commit a2914c37 (2019-10-23): Seperate out clock arguments from product types

Turns e.g.

> f :: (Clock System, Reset System) -> Signal System Int

into

> f :: Clock System -> Reset System -> Signal System Int

Fixes #477
Partially solves #418
commit 16ad94e0 (2019-10-25): Seperate out clock arguments from product types

Turns e.g.

> f :: (Clock System, Reset System) -> Signal System Int

into

> f :: Clock System -> Reset System -> Signal System Int

Fixes #477
Partially solves #418
commit c5f8919b (2019-10-25): Seperate out clock arguments from product types

Turns e.g.

> f :: (Clock System, Reset System) -> Signal System Int

into

> f :: Clock System -> Reset System -> Signal System Int

Fixes #477
Partially solves #418
closed 2019-11-19 by christiaanb