clash-lang/clash-compiler #496: panic! Please don't report as a ghc bug

martijnbastiaan, 2019-02-11:
Every once in a while a Clash bug rears its head telling user to file a bug at the Haskell bug tracker. For example:

```
          clash: panic! (the 'impossible' happened)
            (GHC version 8.6.3 for x86_64-unknown-linux):
          	Clash.Netlist.BlackBox(173): Forced to evaluate untranslatable type: (λ($dIP8214565720323859278 :: GHC.Classes.IP
                                         callStack
                                         GHC.Stack.Types.CallStack) ->
          λ(eta8214565720323859279 :: GHC.Types.[]
                                        GHC.Types.Char) ->
          case $dIP8214565720323859278 of
            _ ->
              Clash.XException.errorX
                @(Clash.Sized.Internal.Unsigned.Unsigned 4)
                (Clash.Transformations.removedArg
                   @(GHC.Classes.IP
                       callStack
                       GHC.Stack.Types.CallStack))
                (Clash.Transformations.removedArg
                   @(GHC.Types.[] GHC.Types.Char))
            GHC.Stack.Types.FreezeCallStack8214565720323839287 
              (ds18214565720323859282 :: GHC.Stack.Types.CallStack) ->
              Clash.XException.errorX
                @(Clash.Sized.Internal.Unsigned.Unsigned 4)
                (Clash.Transformations.removedArg
                   @(GHC.Classes.IP
                       callStack
                       GHC.Stack.Types.CallStack))
                (Clash.Transformations.removedArg
                   @(GHC.Types.[] GHC.Types.Char)) :: GHC.Classes.IP
            callStack
            GHC.Stack.Types.CallStack
          -> GHC.Types.[] GHC.Types.Char
          -> Clash.Sized.Internal.Unsigned.Unsigned 4)
          CallStack (from HasCallStack):
            error, called at src/Clash/Netlist/BlackBox.hs:173:21 in clash-lib-0.99-inplace:Clash.Netlist.BlackBox
          
          Please report this as a GHC bug:  http://www.haskell.org/ghc/reportabug
          
          
          Stdout was:
          Loading dependencies took 2.274103764s
          Parsing and compiling primitives took 7.041263846s
          Compiling: BlockRamTest.topEntity
          Normalisation took 7.053837786s
```

We should **never** redirect users to the Haskell bug tracker. Because we can't feasibly foresee when this message will be reported, we should simply filter our output and replace it with a less offensive message.

christiaanb, 2020-01-17:
@martijnbastiaan Is there any reproducible test case for this?

martijnbastiaan, 2020-01-17:
I think this might still happen if Clash throws an `error` instead of `ClashException`? Maybe you could try adding an error call somewhere. If we can't reproduce it I guess we can close this issue and tag it with 'non-reproducable'.

christiaanb, 2020-01-18:
[handleClashException](https://github.com/clash-lang/clash-compiler/blob/655c0d748c118dd05f9480126ba60f5ecd19d02b/clash-ghc/src-ghc/Clash/GHC/Util.hs#L15) should catch the exceptions.

To try to reproduce the error, I changed the blackbox for blockram to bogusly use its `HasCallStack` argument, but `handleClashException` picked it up correctly.

Hence I'm moving this to 1.4