clash-lang/clash-compiler #512: Building Fails with BlackBox Error for Register

paddytheplaster, 2019-02-27:
Howsagoin,

I've been trying to install Clash under bash on my laptop and on a PC at
work (on and off) for a couple of weeks now but every install fails due to
a blackbox missing error for a register. The error is triggered at
synthesis time for the blinker example. The installs on the PC are done
after a fresh, standard installation of ubuntu 18.04. I'm installing as
root as I need a global installation.

Below I include the scripts I use to install. I'd like to point out that
I wrote the scripts to ensure a consistent installation and that I've
tried to catch any possible error. After a lot of time, the scripts now
run to completion without any error. Please feel free to use.

The following is what happens when I synthesise my Blinker.lhs:

    $ clash --vhdl Blinker.lhs 
    Loading dependencies took 0.959986309s
    Compiling: Blinker.topEntity
    Applied 107 transformations
    Normalisation took 0.163003555s

    <no location info>: error:
        Clash.Netlist.BlackBox(240): No blackbox found for: Clash.Signal.Internal.register#
    [FEW LINES DELETED]

The following are my questions. I'd be much obliged for your help.

    * How do I resolve the black box error?
    * In my script I remove ~/.ghc and ~/.cabal before the install
      (re-install, really). Needless to say, removing a previous ghc
      and cabal installation in order to install clash defeats the
      purpose of a package manager such as cabal. I'd be much obliged
      if you could let me know if there's a better way to re-install.
    * The cabal install fails, unless I manually create ~/.cabal/bin.
      Why do I have to create this directory manually?

TIA for your help.

Paddy

    #!/bin/bash

    ###############################################################
    ## UTILITY FUNCTIONS
    ###############################################################

    function inform( ) {
        echo $@ 1>&2
    }

    function exit_if_failed( ) {
        if [ $? -ne 0 ]; then
            inform "fatal error: ${1}"
            exit ${1}
        else
            inform "SUCCESSFUL EXIT STATUS FOR EXIT CODE ${1}"
        fi
    }

    function run_as_root( ) {
        if [ `whoami` != "root" ]; then
            inform "Fatal error: you must run this script as root"
            exit ${#}
        fi
    }

    # REMOVE PREVIOUS cabal AND ghc INSTALL.
    rm -rf ${HOME}/.ghc ${HOME}/.cabal 1> /dev/null 2>&1
    apt-get purge --assume-yes \
            cabal-install-${CABAL_VERSION} \
            ghc-${GHC_VERSION} 2> /dev/null

    ###############################################################
    # EXIT STATUS DEFINITIONS
    ###############################################################

    USER_ERROR=1
    APT_ERROR=2
    CLASH_INSTALL_ERROR=3
    LINK_ERROR=4

    ###############################################################
    ## THIS IS WHERE IT REALLY STARTS
    ###############################################################

    run_as_root ${USER_ERROR}

    ###############################################################
    # DIRECTORY DEFINITIONS
    ###############################################################

    # contains cabal installation files: you can change this
    DIST=/usr/local/clash-ghc
    DIST_INSTALL=${DIST}/install
    CLASH_INSTALL=${DIST_INSTALL}/clash
    GHC_INSTALL=${DIST_INSTALL}/ghc
    CABAL_INSTALL=${DIST_INSTALL}/cabal
    CABAL_VERSION=2.2
    GHC_VERSION=8.4.3

    ###############################################################
    # CABAL AND GHC INSTALL
    ###############################################################

    CABAL_BIN=/opt/cabal/${CABAL_VERSION}/bin
    GHC_BIN=/opt/ghc/${GHC_VERSION}/bin
    CLASH_BIN=/opt/clash/bin
    CLASH_DATA=/opt/clash/data
    CLASH_DOC=/opt/clash/doc

    mkdir ${DIST} ${DIST_INSTALL}
    chmod 0755 -R ${DIST}

    add-apt-repository -y ppa:hvr/ghc
    exit_if_failed ${APT_ERROR}
    apt-get update
    exit_if_failed ${APT_ERROR}
    apt-get upgrade
    exit_if_failed ${APT_ERROR}

    for PACKAGE in libtinfo-dev \
                   cabal-install-${CABAL_VERSION} \
           	       ghc-${GHC_VERSION} ; do
        inform "ATTEMPT INSTALL OF ${PACKAGE}"
        apt-get install --assume-yes ${PACKAGE}
        exit_if_failed ${APT_ERROR}
    done

    inform "PACKAGE INSTALLS COMPLETE"

    PATH=${GHC_BIN}:${CABAL_BIN}:${PATH}

    ###############################################################
    # CLASH INSTALL
    ###############################################################

    cabal update

    mkdir -p ${CLASH_INSTALL} ${CLASH_BIN} ${CLASH_DATA} ${CLASH_DOC}

    inform "ATTEMPTING TO INSTALL CLASH"

    # NOT SURE WHY THIS IS NEEDED (SEE QUESTIONS).
    mkdir ${HOME}/.cabal/bin -p 2> /dev/null

    cabal install clash-ghc \
           	--prefix=${CLASH_INSTALL} \
    	--bindir=${CLASH_BIN} \
    	--datadir=${CLASH_DATA} \
    	--docdir=${CLASH_DOC} \
            --enable-executable-dynamic \
           	--enable-documentation
    # PREVIOUS INSTAL FAILS, UNLESS ~/.cabal/bin IS CREATED.
    exit_if_failed ${CLASH_INSTALL_ERROR}

    ###############################################################
    # BYE
    ###############################################################

    chmod -R 0755 ${GHC_BIN} \
                  ${CABAL_BIN} \
                  ${CLASH_BIN} \
                  ${CLASH_DOC}

    chmod -R 0755   ${HOME}/.cabal ${HOME}/.ghc
    chown -R paddy  ${HOME}/.cabal ${HOME}/.ghc

    inform "Installation complete."
    inform "Add ${GHC_BIN}, ${CLASH_BIN}, and ${CABAL_BIN} to the PATH:"
    inform "   export PATH=${GHC_BIN}:${CLASH_BIN}:${CABAL_BIN}:\${PATH}"

martijnbastiaan, 2019-02-27:
I'm sorry you had to go through all that, and thanks for the write-up! I've tried your installation instructions, and they seem to work for me on a fresh Ubuntu 18.04 installation!

> How do I resolve the black box error?

For some reason, Clash is unable to find the json files where the blackboxes are stored. In your case,  you installed them in `/opt/clash/data/x86_64-linux-ghc-8.4.3/clash-lib-0.99.3/prims/`. Could you check if running Clash as such works?:

```bash
export PRIMDIR=/opt/clash/data/x86_64-linux-ghc-8.4.3/clash-lib-0.99.3/prims
clash -i${PRIMDIR}/vhdl -i${PRIMDIR}/common Blinker.hs --vhdl
```

(If Clash reports that it used a cached version of a previous compile, use `-fclash-nocache`.)

> In my script I remove ~/.ghc and ~/.cabal before the install (re-install, really). Needless to say, removing a previous ghc nd cabal installation in order to install clash defeats the purpose of a package manager such as cabal. I'd be much obliged if you could let me know if there's a better way to re-install.

Basically, `cabal install` is broken and there's no good solution, except for using `cabal new-install`. I don't think `new-install` supports global installations though, so you'd be back at square one. The *real* solution would be for the Clash team to create binary packages. [We plan to have Debian/Snap/Windows packages ready for Clash 1.0](https://github.com/clash-lang/clash-compiler/issues/20). 

> The cabal install fails, unless I manually create ~/.cabal/bin. Why do I have to create this directory manually?

This is a bug in Cabal, it should create it itself.

martijnbastiaan, 2019-02-27:
One other thing you might like! If you change the header of your installation script to:

```bash
#!/bin/bash
set -e
```

..Bash will stop at the first error it sees instead of carrying on mindlessly. Additionally, you might add `set -x` too, to have it print every command it executes.

```bash
#!/bin/bash
set -e
set -x
```

paddytheplaster, 2019-02-27:
hi Martijn,

Thaks for your reply, as well as for the `set -x`, which I may use.

Your instructions helped me find the cause of the problem. When I tried lookin at
the following directory, I got a permission denied!

    /opt/clash/data/x86_64-linux-ghc-8.4.3/clash-lib-0.99.3/prims

I then recursively changed permissions to 0755 and now `clash --vhdl` works fine,
even without eporting the directory explicitly as you suggested.

Thanks a mille. This has made my day.

Regards,

Paddy

martijnbastiaan, 2019-02-27:
Cheers!

closed 2019-02-27 by martijnbastiaan