clash-lang/clash-compiler #513: Emit warning if directory explicitely listed with `-idir` can't be accessed

martijnbastiaan, 2019-02-27:
If a directory is listed using `-idir` the user probably expects this directory to exist. It could be inaccessible due to various reasons however (permissions, typos, forgot-to-mount situations, ..). If so, we should warn the user.

mmzx, 2019-07-08:
As a low hanging fruit to learn to the source I have started looking into this issue... Currently, the user is notified about the case of _missing directories_ and clash goes on with processing.
Now,
- I would add further checks on these directories,
- Do it for ghc-8.4 and ghc-8.2, when the actual behavior is decided:
What do you think about the desired behavior of clash when a directory is missing or inaccessible?

martijnbastiaan, 2019-07-09:
Hey @mmzx, great to hear you want to contribute to Clash :). Welcome!

> I would add further checks on these directories,

What kind of checks do you have in mind?

> What do you think about the desired behavior of clash when a directory is missing or inaccessible?

Explicitly listed directories should exist/be accessible in my opinion. I've been stuck for _hours_ because I thought Clash was doing something funny, while all there was to it was a typo in `-i`.. I therefore think the default should be to error hard. Then, we could add a flag `-fclash-no-check-inaccessible-idirs` that would disable the check.

mmzx, 2019-07-14:
I have just created #670, to address this with a solution. 

Further checks could be e.g. on permissions or in some other but rare cases where an _IOError_ could provide more information why a given location cannot be accessed.

christiaanb, 2019-09-30:
@mmzx @martijnbastiaan Can we close this?

martijnbastiaan, 2019-09-30:
Yes :)