clash-lang/clash-compiler #516: Clash can't coveniently disable powerup values

martijnbastiaan, 2019-03-01:
With the addition of https://github.com/clash-lang/clash-compiler/pull/498, Clash now supports initial/powerup values. By default, register _reset_ values are also used as _initial_ values. For most designs, this is what we want. If a user doesn't want initial values, they can use `register#` to set a different (or no) initial value. This is, of course, not really a good answer: some designs can't support initial values. This report discusses some of the solutions to these problems.

-----------------------------------------------------

**Solution 1: implicit/default arguments**

We could use [implicit arguments in combinations with defaults](http://hackage.haskell.org/package/implicit-params) to specify whether a register should have a initial value. 

Pros:
 * Backwards compatible
 * Can change strategies within a design

Cons:
 * Easy to forget to propagate
 * Ugly, IMO
 * Doesn't feel clashy

---------------------------------------------------

**Solution 2: Add field to `Reset`**

Add an additional field `Initial` to `Reset`, such that it looks something like:

```haskell
data Reset (domain :: Domain) (synchronous :: ResetKind) (initial :: Bool) where
  Sync  :: Signal domain Bool -> Reset domain 'Synchronous initial
  Async :: Signal domain Bool -> Reset domain 'Asynchronous initial
```

Pros:
 * Can change init strategies within a design
 * Polymorphic; users don't have to decide on a strategy up to `topEntity`
 * Easy to implement
 * No real maintenance costs

Cons:
 * Not backwards compatible
 * `Undefined` is back baby

-------------------------------------------------

**Solution 3: Add `Initial`, besides `Clock` and `Reset`**

Similar to having `Clock` and `Reset` arguments, add an `Initial` argument deciding whether to use initial values in registers.

Pros:
 * Can change init strategies within a design
 * Polymorphic; users don't have to decide on a strategy up to `topEntity`

Cons:
 * We'd need twice as many `Hidden` variants. `HiddenClock`, `HiddenIntial`, `HiddenClockInitial`, `HiddenClockResetInitial` etc..
 * Not backwards compatible

-------------------------------------------------

**Solution 4: Flag to disable initial values in HDL**

Some users have signaled that they wouldn't care much about losing precise Clash simulation, as long as they have it their HDL tools. This is motivated by the fact that _as long as you reset your Clash circuit properly_ simulation will still yield the same results as HDL. HDL simulation is then used in the (rare) cases where a reset value caused an issue. We could therefore add a flag to Clash that omits initial values from HDL (however, Clash simulation would still act as if initial values were defined). 

Pros:
 * Easy to implement

Cons:
 * Breaks philosophy that users never have to venture into HDL in order to debug issues. **Eek!**

-------------------------------------------------

**Solution 5: Additional `Clash.Prelude` module**

Release a copy of `clash-prelude` on Hackage, with all relevant functions now taking a initial value.

Pros:
 * Backwards compatible

Cons:
 * Massive maintenance burden
 * Potentially confusion for users
 * Hard (impossible?) to keep design polymorphic in initial values

-------------------------------------------------

**Solution 6: Add specialized functions without initial values**

A zoo of functions taking initial values: `mealyI`, `mealyBI`, `mealyDFI`, ...

Pros:
 * Backwards compatible

Cons:
 * We now have a zoo of functions
 * Not polymorphic in initial values

-------------------------------------------------

**Solution 7: Add CPP compile option to Clash**

Add the ability to compile Clash with a CPP option enabled, adding enabling/disabling initial values.

Pros:
 * Backwards compatible

Cons:
 * Objectively ugly
 * Quite the maintenance burden
 * Need to recompile in order to switch
 * No strategy changes within design

-------------------------------------------------

Note that I haven't addressed the case of users wanting *different* initial and reset values. As far as I can see, this is not a common design choice. In fact, both Xilinx and Intel don't support it on their FPGA platforms.

For now, I'm in favor of solution 2, with solution 4 trailing at quite a distance. I don't feel much for the others, but I'm happy to be convinced otherwise.

christiaanb, 2019-03-01:
Can’t be 2 because registers without a reset can still have an initial value.

martijnbastiaan, 2019-03-02:
Alright, then we'll add it to `Clock`.

christiaanb, 2019-03-04:
Another option would be:

Produce HDL with a parameter(verilog)/generic(VDHL), with a default set to true, that sets an initinitial value or not. So for a register you'd get something along the line of:

```
register : if USE_INTIAL_VALUES generate
  signal reg : <type> := <initial value;
begin
  reg : proces (clk) begin
  if rising_edge(clk) then begin
    if rst then
       reg <= reset_val;
    else
       reg <= input;
    end if;
  end if;
  end process;
end generate;

register_no_initial : if !USE_INTIAL_VALUES generate
  signal reg : <type>;
begin
  proces (clk) begin
  if rising_edge(clk) then begin
    if rst then
       reg <= reset_val;
    else
       reg <= input;
    end if;
  end if;
  end process;
end generate;
```

and then you make sure that every module passes along its `USE_INITIAL_VALUE` to every module it calls. In Verilog this can be solved nicer, with out parameters, because it supports C-style macro preprocessing and you can use \``ifdef` and friends.

christiaanb, 2019-03-04:
Also there are other parameters to a register that are currently not configurable, such as:
1. what's the active clock flank
2. is the reset port positive or negated?

Just like "initial-value" aspect, I feel neither of these properties should really be part of the `Clock` nor `Reset` type; and really belong to a third "register configuration" argument. The reason why source vs gated clock and synchronous vs asynchronous reset are part of the clock and reset type, and not of this third "register configuration" argument, is because they affect meta-stability issues.

christiaanb, 2019-03-04:
So perhaps this "register configuration" could be implemented using **Solution 1**, with **Solution 8** on top as a failsafe.

martijnbastiaan, 2019-03-04:
Solution 8?

martijnbastiaan, 2019-03-04:
I agree that these properties do not necessarily belong to `Clock`, but all things considering, I think it's still the right place for a number of reasons:

1. Solution 1 would introduce yet another concept new Clash users have to learn. The implicitness/defaultness of it is a recipe for confusion, if not bugs. "Explicit" implicit (for the lack of a better term) arguments are tricky enough as it is.

2. We don't do clockless designs and probably won't for the foreseeable future. All relevant components can therefore access `Clock` and its associated configuration.

3. Adding yet another implicit argument might be detrimental to new users's experience. (I'm not feeling particularly strongly about this one though.)

Your first option:

> what's the active clock flank

Is tricky again, although I think I'd still be in favor of adding them to `Clock` for the reasons I listed above. Additionally, synchronizing two domains would need special treatmeant if active clock flanks differ. Adding it to `Clock` would ensure our synchronization functions can check for it.

As for your last option:

> is the reset port positive or negated?

Actually, I feel that this might be an excellent candidate for inclusion in `Reset`. This way, you won't be able to mix "high" reset compontents with "low" reset signals, and vice versa.

christiaanb, 2019-03-04:
I agee with you on the active flank and reset polarity. The initial value thing still feels it’s not really part of the clock though

martijnbastiaan, 2019-03-04:
I agree.

christiaanb, 2019-03-04:
But I guess adding it as a parameter to clk is the lesser of 2 evils... but maybe it should be part of the domain. So you can have a single clock, but e.g. on Intel FPGA you can have a static region with initial values, and reconfigurable region without, while still using the same clock

martijnbastiaan, 2019-06-25:
Fixed in https://github.com/clash-lang/clash-compiler/commit/1d4ef5ea23ee4f7948db246009a26ff546eb7639.

closed 2019-06-25 by martijnbastiaan