clash-lang/clash-compiler #517: Can't install using cabal, conflict with template-haskell-2.14.0.0

mgourd, 2019-03-03:
I'm using macOS. I get the following result when install clash-ghc:
```
cabal install clash-ghc --enable-documentation  
Warning: The install command is a part of the legacy v1 style of cabal usage.

Please switch to using either the new project style and the new-install
command or the legacy v1-install alias as new-style projects will become the
default in the next version of cabal-install. Please file a bug if you cannot
replicate a working v1- use case with the new-style commands.

For more information, see: https://wiki.haskell.org/Cabal/NewBuild

Resolving dependencies...
cabal: Could not resolve dependencies:
[__0] trying: clash-ghc-0.99.3 (user goal)
[__1] trying: transformers-0.5.5.0/installed-0.5... (dependency of clash-ghc)
[__2] trying: ghc-8.6.3/installed-8.6... (dependency of clash-ghc)
[__3] trying: template-haskell-2.14.0.0/installed-2.1... (dependency of ghc)
[__4] next goal: clash-prelude (dependency of clash-ghc)
[__4] rejecting: clash-prelude-0.99.3 (conflict: ghc =>
template-haskell==2.14.0.0/installed-2.1..., clash-prelude =>
template-haskell>=2.12.0.0 && <2.14)
[__4] rejecting: clash-prelude-0.99.2, clash-prelude-0.99.1,
clash-prelude-0.99, clash-prelude-0.11.2, clash-prelude-0.11.1,
clash-prelude-0.11, clash-prelude-0.10.14, clash-prelude-0.10.13,
clash-prelude-0.10.11, clash-prelude-0.10.10, clash-prelude-0.10.9,
clash-prelude-0.10.8, clash-prelude-0.10.7, clash-prelude-0.10.6,
clash-prelude-0.10.5, clash-prelude-0.10.4, clash-prelude-0.10.3,
clash-prelude-0.10.2, clash-prelude-0.10.1, clash-prelude-0.10,
clash-prelude-0.9.3, clash-prelude-0.9.2, clash-prelude-0.9.1,
clash-prelude-0.9, clash-prelude-0.8.1, clash-prelude-0.8,
clash-prelude-0.7.5, clash-prelude-0.7.4, clash-prelude-0.7.3,
clash-prelude-0.7.2, clash-prelude-0.7.1, clash-prelude-0.7,
clash-prelude-0.6.0.1, clash-prelude-0.6, clash-prelude-0.5.1,
clash-prelude-0.5.0.1, clash-prelude-0.5, clash-prelude-0.4,
clash-prelude-0.3, clash-prelude-0.2 (conflict: clash-ghc =>
clash-prelude>=0.99.3 && <1.0)
[__4] fail (backjumping, conflict set: clash-ghc, clash-prelude, ghc)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: clash-prelude, template-haskell,
clash-ghc, ghc, transformers, base, unbound
```
I find that `clash-prelude` specify that `template-haskell >=2.9.0.0 && <2.14`.
```
ghc --version 
The Glorious Glasgow Haskell Compilation System, version 8.6.3
```
```
cabal --version
cabal-install version 2.4.1.0
compiled using version 2.4.1.0 of the Cabal library 
```

mgourd, 2019-03-03:
I succeeded resolving dependencies using `--allow-older` flag, but resulted in another error.
```
Configuring clash-prelude-0.10.8...
Preprocessing library for clash-prelude-0.10.8..
Building library for clash-prelude-0.10.8..

src/CLaSH/Promoted/Nat.hs:27:66: error: parse error on input ‘*’
   |
27 | import GHC.TypeLits    (KnownNat, Nat, type (+), type (-), type (*), type (^),
   |                                                                  ^
```

leonschoorl, 2019-03-03:
The current clash (0.99.3) on hackage is not compatible GHC-8.6
Easiest is probably install GHC-8.4.4 and use that.

Or you can use the development version of clash:
https://github.com/clash-lang/clash-compiler/wiki/FAQ#how-do-i-run-the-development-version-of-clash

closed 2019-03-03 by mgourd