clash-lang/clash-compiler #566: Can't match template for "Clash.Signal.Internal.register#"

adamwalker, 2019-03-31:
When compiling this file: https://gist.github.com/adamwalker/a5e89861de66a4b9fd48c5b917009dcb
With the latest master (c3047), I get the error below:
```
Warning: primitive Clash.Sized.Vector.length isn't using all its arguments, some might be optimized away.

Warning: primitive Clash.Sized.Vector.dfold isn't using all its arguments, some might be optimized away.
Clash: Applied 3174 transformations

aesBug.hs:108:1: error:
    Clash.Netlist.BlackBox(150): Can't match template for "Clash.Signal.Internal.register#" :

// register begin
reg ~TYPO ~GENSYM[~RESULT_reg][0] = ~CONST[2];~IF ~ISGATED[0] ~THEN
always @(posedge ~ARG[0][1]~IF ~ISSYNC[1] ~THEN ~ELSE or posedge ~ARG[1]~FI) begin : ~GENSYM[~RESULT_register][1]
  if (~ARG[1]) begin
    ~SYM[0] <= ~CONST[3];
  end else if (~ARG[0][0]) begin
    ~SYM[0] <= ~ARG[4];
  end
end~ELSE
always @(posedge ~ARG[0]~IF ~ISSYNC[1] ~THEN ~ELSE or posedge ~ARG[1]~FI) begin : ~SYM[1]
  if (~ARG[1]) begin
    ~SYM[0] <= ~CONST[3];
  end else begin
    ~SYM[0] <= ~ARG[4];
  end
end~FI
assign ~RESULT = ~SYM[0];
// register end

with context:

Context {bbResult = (Identifier "result_2" Nothing,Vector 8 Bool), bbInputs = [(Identifier "clk" Nothing,Clock "system" 10000 Source,False),(Identifier "rst" Nothing,Reset "system" 10000 Synchronous,False),(DataCon (Vector 8 Bool) VecAppend [Identifier "result_2_res" Nothing,DataCon (Vector 7 Bool) VecAppend [Identifier "result_2_res_0" Nothing,DataCon (Vector 6 Bool) VecAppend [Identifier "result_2_res_1" Nothing,DataCon (Vector 5 Bool) VecAppend [Identifier "result_2_res_2" Nothing,DataCon (Vector 4 Bool) VecAppend [Identifier "result_2_res_3" Nothing,DataCon (Vector 3 Bool) VecAppend [Identifier "result_2_res_4" Nothing,DataCon (Vector 2 Bool) VecAppend [Identifier "result_2_res_5" Nothing,DataCon (Vector 1 Bool) VecAppend [Identifier "result_2_res_6" Nothing]]]]]]]],Vector 8 Bool,False),(DataCon (Vector 8 Bool) VecAppend [Identifier "result_2_res_7" Nothing,DataCon (Vector 7 Bool) VecAppend [Identifier "result_2_res_8" Nothing,DataCon (Vector 6 Bool) VecAppend [Identifier "result_2_res_9" Nothing,DataCon (Vector 5 Bool) VecAppend [Identifier "result_2_res_10" Nothing,DataCon (Vector 4 Bool) VecAppend [Identifier "result_2_res_11" Nothing,DataCon (Vector 3 Bool) VecAppend [Identifier "result_2_res_12" Nothing,DataCon (Vector 2 Bool) VecAppend [Identifier "result_2_res_13" Nothing,DataCon (Vector 1 Bool) VecAppend [Identifier "result_2_res_14" Nothing]]]]]]]],Vector 8 Bool,False),(Identifier "result_3" Nothing,Vector 8 Bool,False)], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 0, bbCompName = "Main_topEntity"}
    
    NB: The source location of the error is not exact, only indicative, as it is acquired after optimisations.
    The actual location of the error can be in a function that is inlined.
    To prevent inlining of those functions, annotate them with a NOINLINE pragma.
    |
108 | topEntity clk rst = withClockReset @System @Source @Synchronous clk rst aes
    | ^^^^^^^^^
```

leonschoorl, 2019-04-01:
Clash needs the initial values of registers to be constants and right now it doesn't think `unpack 1 :: Vec 8 Bool` is a constant.
As a workaround you could change it to something like: `repeat False :< True`

leonschoorl, 2019-04-01:
Minimal testcase:
```haskell
topEntity :: HiddenClockReset System Source Synchronous => Signal System (Vec 8 Bool) -> Signal System (Vec 8 Bool)
topEntity = register (unpack 1)
```

martijnbastiaan, 2019-04-03:
I believe this is due to a missing evaluator rule for `Clash.Sized.Internal.BitVector.checkUnpackUndef`.

martijnbastiaan, 2019-04-03:
@adamwalker Could we add the file you posted to the Clash's testsuite? Seems like a nice real-world use of Clash.

adamwalker, 2019-04-03:
Yep, thats fine

christiaanb, 2019-04-03:
Weird, all mentions of `checkUnpackUndef` should have been removed in the ghc2core process: https://github.com/clash-lang/clash-compiler/commit/6370c76b5da36d2e1e838510d3ce6f19d7c5fe5a#diff-c66685a9c1c6b8db5782ddab901dd784

martijnbastiaan, 2019-04-03:
The problem is that a primitive definition exists for [`checkUnpackUndef`](https://github.com/clash-lang/clash-compiler/blob/0dacb17d8406eaabb12ff7672f2145c1432500d0/clash-lib/prims/common/Clash_Sized_Internal_BitVector.json#L9). This causes the code linked to not consider it.

martijnbastiaan, 2019-04-03:
Fixed, thanks @adamwalker !

closed 2019-04-03 by martijnbastiaan

adamwalker, 2019-04-04:
No probs. Thanks for the quick fix!

commit b6aa8fae (2019-04-03): Remove BlackBox definition for `checkUnpackUndef`. Fix #566