clash-lang/clash-compiler #581: Trouble running clash development version

mitch1993, 2019-04-11:
Hey!
New to Clash and Git, so I'm sorry if this question is out of place here.
I'm trying to run the development version of clash on Ubuntu 18.04 with cabal 
version 2.4.1.0 installed

I installed the development version of clash following the [FAQ](https://github.com/clash-lang/clash-compiler/wiki/FAQ#how-do-i-run-the-development-version-of-clash) 

works fine until I try the :vhdl command which throws the following error:
`*** Exception: Clash only works with the exact clash-prelude it was built with.
Clash was built with: clash-prelude-0.99-inplace
So can't run with:    clash-prelude-0.99.3-7mMLqVZqcoR9k3qLFEE3WD`

Following the directions in the  [Getting Started Tab](https://github.com/clash-lang/clash-compiler/wiki/Getting-started) i get 
`error: missing binary operator before token "("
     #if MIN_VERSION_ghc(8,6,0)
`
when trying to run ./clash-dev
Any ideas what i could be doing wrong?

martijnbastiaan, 2019-04-11:
Hey @mitch1993 !

Clash is probably confused by some (temporary) files. It unfortunately happens during development. Could you try a clean clone and running it with `cabal`?

```
git clone https://github.com/clash-lang/clash-compiler.git
cd clash-compiler
cabal new-build clash clashi
```

You should now be able to run `clashi`:

```
cabal new-run clashi
```

Or Clash directly:

```
cabal new-run -- clash
```

Compiling a file would look like:

```
cabal new-run -- clash Test.hs --vhdl
```

martijnbastiaan, 2019-04-11:
Oh and `clash-dev` is only useful if you intend on hacking on Clash itself!

mitch1993, 2019-04-11:
Hmm clean clone (deleted the old folder and then just used the commands you posted). 
Same errors. I have a released version of Clash installed. Could that cause my problems?

christiaanb, 2019-04-11:
As a work-around, could you try changing the version in the `clash-lib.cabal`, `clash-ghc.cabal`, and `clash-prelude.cabal` files to `0.999`; and the running `cabal new-build all`.

Hopefully, Clash will then pick the `0.999` version over the installed `0.99.3` version.

mitch1993, 2019-04-11:
`*** Exception: Clash only works with the exact clash-prelude it was built with.
Clash was built with: clash-prelude-0.999-inplace
So can't run with:    clash-prelude-0.99.3-7mMLqVZqcoR9k3qLFEE3WD`

same error :-/

martijnbastiaan, 2019-04-11:
What happens if you write:

```
cabal new-run --write-ghc-environment-files=always clashi
```

mitch1993, 2019-04-11:
works with 

`cabal new-run --write-ghc-environment-files=always clashi`

thank you very much!

martijnbastiaan, 2019-04-11:
Great! Have fun!

closed 2019-04-11 by martijnbastiaan