clash-lang/clash-compiler #586: compile fail for clash-ghc 0.99.3 on 32-bit linux debian (Evaluator.hs)

pbreuer, 2019-04-22:
Everything else compiled (clash-prelude, clash-lib, various supporting modules). I guess this is a 32-bit issue. You maybe are developing on a 64-bit machine and assuming Int is 64-bit?
Linux debian unstable with ghc 8.4.4. 
Glasgow Haskell Compiler, Version 8.4.4, stage 2 booted by GHC version 8.4.4
Using binary package database: /usr/lib/ghc/package.conf.d/package.cache
package flags []
loading package database /usr/lib/ghc/package.conf.d
wired-in package ghc-prim mapped to ghc-prim-0.5.2.0
wired-in package integer-gmp mapped to integer-gmp-1.0.2.0
wired-in package base mapped to base-4.11.1.0
wired-in package rts mapped to rts
wired-in package template-haskell mapped to template-haskell-2.13.0.0
wired-in package ghc mapped to ghc-8.4.4
wired-in package dph-seq not found.
wired-in package dph-par not found.

The problem arrives here:

Building library for clash-ghc-0.99.3..
[ 1 of 13] Compiling Clash.GHC.ClashFlags ( src-ghc/Clash/GHC/ClashFlags.hs, dist-ghc/build/Clash/GHC/ClashFlags.o )
[ 2 of 13] Compiling Clash.GHC.GHC2Core ( src-ghc/Clash/GHC/GHC2Core.hs, dist-ghc/build/Clash/GHC/GHC2Core.o )
[ 3 of 13] Compiling Clash.GHC.Evaluator ( src-ghc/Clash/GHC/Evaluator.hs, dist-ghc/build/Clash/GHC/Evaluator.o )

src-ghc/Clash/GHC/Evaluator.hs:514:67: error:
    * Couldn't match expected type `Int#' with actual type `Int64#'
    * In the first argument of `I#', namely `p'
      In the second argument of `($)', namely `I# p'
      In the first argument of `Left', namely
        `(Literal . IntLiteral . toInteger $ I# p)'
    |
514 |                    [ Left (Literal . IntLiteral  . toInteger $ I# p)
    |                                                                   ^
make: *** [/usr/share/cdbs/1/class/hlibrary.mk:147: build-ghc-stamp] Error 1
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -d -us -uc -ui failed

The debuild with -d is because debuild thinks I need libpthread-dev, whereas I have it but with another name or something, and anyway that'd be a linker issue and I'm not there yet so it's irrelevant. This is a haskell  types issue. The indicated code is:

-- decodeDouble_Int64# :: Double# -> (#Int#, Int##)
  "GHC.Prim.decodeDouble_Int64#" | [i] <- doubleLiterals' args
    -> let (_,tyView -> TyConApp tupTcNm tyArgs) = splitFunForallTy ty
           (Just tupTc) = HashMap.lookup (nameOcc tupTcNm) tcm
           [tupDc] = tyConDataCons tupTc
           !(D# a) = fromRational i
           !(# p, q #) = decodeDouble_Int64# a
       in reduce $
          mkApps (Data tupDc) (map Right tyArgs ++
                   [ Left (Literal . IntLiteral  . toInteger $ I# p)  <------ HERE
                   , Left (Literal . IntLiteral  . toInteger $ I# q)])

The only sense I can make of that is that decodeDouble_Int64# is supplying an Int64 as the first of the pair result two lines higher up, not an Int. 

That guess would mean that decodeDouble_Int64#  needs retyping. I don't know where it comes from, so can't check or do much. If that is really a 32-bit int in 64-bit clothing, everything would be alright if I did some toInteger, fromIntegering, which I will try next. That won't help if the range of p is really 64-bit, not 32-bit.

Clues?

(I have been forming debian packages out of the haskell modules with cabal-debian and a bit of editing of the resulting control files before debuild, and that has gone OK, except that 1<= Div 18 3 failed (last test in clash-prelude), and I got the famous sporadic ghci problem with _GLOBAL_OFFSET_TABLE_ in the doctest sequence of clash-lib, and it wouldn't go away, just shifted, as I moved .o files out of the way, which is supposed to be the cure according to google, so I disabled the test sequence entirely for clash-lib with nocheck in DEB_BUILD_OPTIONS. That's all I have to confess to.)

martijnbastiaan, 2019-04-23:
> I guess this is a 32-bit issue. You maybe are developing on a 64-bit machine 

That is correct!

> and assuming Int is 64-bit?

Yes, and no. We might be doing that in some parts of our code, but that's not on purpose. We'll need to add 32-bit to our CI. 

Thanks for the bug report!

pbreuer, 2019-04-23:
I replaced the I# with I64# on line 514 and Evaluator compiled and the build went all the way OK (I'm not running the tests ... I'll try again later with them enabled).
Obviously that is just a flaky, syntactic bodge to get past the immediate problem and a more careful, systematic and semantically based approach is required. Thank you for adding 32-bit to your testing.

martijnbastiaan, 2019-04-23:
Keep in mind running Clash in 32 bits mode might change your HDL, as `Integer`s are now represented in HDL as 32 bit values. It's therefore probably best to run the testsuite as follows:

```cabal new-run testsuite -- -p VHDL.clash```

This will test if Clash succeeds to compile, but won't try to run any simulators.

martijnbastiaan, 2019-04-23:
If you want to run Clash in 64-bit mode on a 32-bit machine anyway, use:

```clash -fclash-intwidth=64```

pbreuer, 2019-04-23:
I used cabal only to download, then moved the tar.gz files to a src compile area, untared them, and ran cabal-debian in each directory to produce debian/ subdirs with debian/control, edited them, then ran debuild. That does not seem to produce a test sequence for clash-ghc, though it did for clash-prelude and clash-lib.
``. /usr/share/haskell-devscripts/Dh_Haskell.sh && check_recipe
Running debian/hlibrary.setup test --builddir=dist-ghc --show-details=direct
Package has no test suites.
touch check-ghc-stamp``

The cabal line you suggested produces "cabal new-run testsuite -- -p VHDL.clash
cabal: Cannot run the package testsuite, it is not in this project (either
directly or indirectly). If you want to add it to the project then edit the
cabal.project file."  (clash-ghc.cabal?)
Perhaps it wants me to search in the cabal download area for "testsuite" and go to work on that?
[Cabal seems to want me to work as root, in order to access /var/cache/cabal - I don't know why, it seems to have made a subdir for me as user-, but that also puts all the cabal download in /root/.cabal/, where it also seems to end up compiling. I'm not keen on that. Whatever gets made seems to stay in root at install, and not go to /usr/local where I expect it to go. So I've been making debian packages from the source tars and installing them and that looks much better integrated with the whole system.]

pbreuer, 2019-04-23:
I should add before I forget that I would recommend you prepare at least a debian/ subdir (with suitable content) in your source tar.gz, which is all anyone requires to run fakeroot make -f debian/rules binary or debuild (which does the former with knobs on) and thus build a .deb package in the directory above, which one can install with the apt or whatever package manager. It should then be immune to weirdness from the system updating beyond it, and so on. Deb binary packages can be made by anyone then for further distribution.
If it is of interest, I can provide the debian/ subdirs I built with the aid of cabal-debian, but I already removed all the foo-prof entries and anything else I didn't like to build so they're a bit manky.
I note that the .cabal file needs to be purged of non-ascii characters for cabal-debian to work (there is a nonascii copyright sign, and some nonascii chars in the apologies for why clash is called clash, and maybe the quotes used are nonascii too). Maybe that's my fault - who knows if I have i8n installed. Cabal-debian incorrectly produced some libghc-ghc-foo dependencies that have to be changed to libghc-foo, but the control file is alright otherwise. The correct cabal-debian options are a --debian-name-base foo where foo is what one wants in the libghc-foo-dev package name that results. You also need a random --maintainer 'NAME <EMAIL>' option on the cabal-debian command line. Presumably somebody who cares would do things with sigs too. No other cabal-debian options are advisable in a first instance (I used --disable-haddock ; docs, who needs 'em).
I did not need to compile any more haskell modules as support (the rest were available via apt in debian's repository) EXCEPT concurrent-supply, unbound-generics, ghc-typelits-natnormalise, ghc-tcplugins-extra,  ghc-typelits-knownnat, ghc-typelits-extra. Those would need to be provided as support along with clash debian packages if you were to distribute .debs. At least these instructions might be useful.
That's about all that strikes me as possibly useful information for you from my own exploration so far. The haskell modules I pulled via apt in debian unstable were all within a minor or subminor of what cabal would have pulled so I don't think there's anything fundamental to fear there. I can list the versions if it is of interest (I don't think so).

pbreuer, 2019-04-23:
Finally getting round to trying what was built, I see clashi does not run:

% clashi
CLaSHi, version 0.99.3 (using clash-lib, version 0.99.3):
http://www.clash-lang.org/  :? for help
clashi: /usr/lib/ghc/ghc-prim-0.5.2.0/HSghc-prim-0.5.2.0.o: unknown symbol `_GLOBAL_OFFSET_TABLE_'
clashi: unable to load package `ghc-prim-0.5.2.0'

That did give me trouble at compile. It's in the init section of libpthread.so:

% nm /lib/i386-linux-gnu/libpthread.so.0 | grep TABLE
0001e000 d _GLOBAL_OFFSET_TABLE_

Preloading that library does not help.

% setenv LD_PRELOAD /lib/i386-linux-gnu/libpthread.so.0
% clashi
CLaSHi, version 0.99.3 (using clash-lib, version 0.99.3):
http://www.clash-lang.org/  :? for help
clashi: /usr/lib/ghc/ghc-prim-0.5.2.0/HSghc-prim-0.5.2.0.o: unknown symbol `_GLOBAL_OFFSET_TABLE_'
clashi: unable to load package `ghc-prim-0.5.2.0'

That is maybe unsurprising because ldd says the executable dynamically links to libphread anyway, and it didn't help it:
% ldd `which clashi`
        linux-gate.so.1 (0xb7f1e000)
        libm.so.6 => /lib/i386-linux-gnu/libm.so.6 (0xb7df6000)
        libpthread.so.0 => /lib/i386-linux-gnu/libpthread.so.0 (0xb7dd5000)
        libtinfo.so.6 => /lib/i386-linux-gnu/libtinfo.so.6 (0xb7dac000)
        librt.so.1 => /lib/i386-linux-gnu/librt.so.1 (0xb7da2000)
        libutil.so.1 => /lib/i386-linux-gnu/libutil.so.1 (0xb7d9d000)
        libdl.so.2 => /lib/i386-linux-gnu/libdl.so.2 (0xb7d97000)
        libgmp.so.10 => /usr/lib/i386-linux-gnu/libgmp.so.10 (0xb7d08000)
        libatomic.so.1 => /usr/lib/i386-linux-gnu/libatomic.so.1 (0xb7cfe000)
        libffi.so.6 => /usr/lib/i386-linux-gnu/libffi.so.6 (0xb7cf4000)
        libc.so.6 => /lib/i386-linux-gnu/libc.so.6 (0xb7b16000)
        /lib/ld-linux.so.2 (0xb7f1f000)

So I'm not sure what to make of that. Apparently HSghc-prim-0.5.2.0.o comes from a part ghc-prim-0.5.2.0 of ghc itself, so I am not responsible for having produced it or really to blame for it wanting libpthread and not getting it.

ghc: /usr/lib/ghc/ghc-prim-0.5.2.0
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/CString.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/CString.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Classes.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Classes.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Debug.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Debug.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/IntWord64.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/IntWord64.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Magic.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Magic.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/PrimopWrappers.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/PrimopWrappers.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Tuple.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Tuple.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Types.dyn_hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/GHC/Types.hi
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/HSghc-prim-0.5.2.0.o
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/libHSghc-prim-0.5.2.0-ghc8.4.4.so
ghc: /usr/lib/ghc/ghc-prim-0.5.2.0/libHSghc-prim-0.5.2.0.a
ghc: /var/lib/ghc/package.conf.d/ghc-prim-0.5.2.0.conf

I see that ibHSghc-prim-0.5.2.0-ghc8.4.4.so wants to load libpthread, so if clashi wants to load libHSghc-prim then it may want to load libpthread as a result, but I do not know why the linker (which? ghci?) fails to find it.

Anyway, there is an issue here. If I can find the point just before clashi is linked together to form the executable, I may be able to add the libpthread.a into it, but I do not know if that is enough.

martijnbastiaan, 2019-04-24:
Right, didn't catch you were building Debian packages! We tried to do the same a while ago. 

I vaguely remember having to pass `--dep-map pthread:pthread` or something similar to cabal-debian. The exact options we used are stored on a computer a don't have access to at the moment. I'll update you ASAP.

martijnbastiaan, 2019-04-24:
I also think I remember you need to pass `--cabal-flag -dynamic` or something similar to `cabal-debian`. Again, I'll check it out ASAP.

pbreuer, 2019-04-24:
Thanks very much. I don't "know" how to make an apt repository either, but I think one just apes the structure of some other repository and/or uses some repository-making debian tool. I seem to recall "dput". The alternative is not to make a repository but offer the stuff to debian itself to host, which offloads the problem. There is a section devoted to electronics and there must be a maintainer there to whom one can pass the example packages and who will say "yes" on sight.
dput - Debian package upload tool
I have located the link line for clashi and it is
/usr/bin/ghc --make -fbuilding-cabal-package -O -static -outputdir dist-ghc/build/clashi/clashi-tmp -odir dist-ghc/build/clashi/clashi-tmp -hidir dist-ghc/build/clashi/clashi-tmp -stubdir dist-ghc/build/clashi/clashi-tmp -i -idist-ghc/build/clashi/clashi-tmp -i. -idist-ghc/build/clashi/autogen -idist-ghc/build/global-autogen -Idist-ghc/build/clashi/autogen -Idist-ghc/build/global-autogen -Idist-ghc/build/clashi/clashi-tmp -optP-include -optPdist-ghc/build/clashi/autogen/cabal_macros.h -lpthread -hide-all-packages -Wmissing-home-modules -no-user-package-db -package-db /var/lib/ghc/package.conf.d -package-db dist-ghc/package.conf.inplace -package-id base-4.11.1.0 -package-id clash-ghc-0.99.3-JZm7H7BJfw86gJ092MhIBk -XHaskell2010 ./src-ghc/Interactive.hs -o dist-ghc/build/clashi/clashi -Wall -optl-Wl,-Bsymbolic-functions -optl-Wl,-z,relro 
Replacing the -lpthread by an explicit /usr/..../libpthread.a does not help, nor does moving it later, nor does adding -threaded -optl-Wl,-rpath,/usr/lib/i386-linux-gnu . I'll try your suggestion.

martijnbastiaan, 2019-04-24:
Cool, let me know! 

Btw: You can format your post by using triple "ticks" as [documented here](https://help.github.com/en/articles/creating-and-highlighting-code-blocks). That makes posts here a bit easier to read!

pbreuer, 2019-04-24:
No improvement yet via the cabal-debian flags, but honestly I don't see how flags to cabal-debian should cure the problem. OK - maybe it's a common problem so they set up a solution, but it would have to affect the source "make" instructions not the debian package construction and I don't see it. Are you sure the trick wasn't to do something inside the source tar?
My interpretation of the issue is that there is a runtime load of a ghc-prim library that itself then wants to load a symbol from libpthreads but by then whatever it is that does the load doesn't know where or how to look for it. Strace of ```clashi``` shows me:
``` execve("/usr/sbin/i686-linux-gnu-gcc", ["i686-linux-gnu-gcc", "-fno-
stack-protector", "-DTABLES_NEXT_TO_CODE", "-fuse-ld=gold", "-B/usr/lib/ghc/ghc-
prim-0.5.2.0", "--print-file-name", "libatomic.so"], 0xbfb4320c /* 69 vars */) =
 -1 ENOENT (No such file or directory)
[pid 15807] execve("/usr/bin/i686-linux-gnu-gcc", ["i686-linux-gnu-gcc", "-fno-s
tack-protector", "-DTABLES_NEXT_TO_CODE", "-fuse-ld=gold", "-B/usr/lib/ghc/ghc-p
rim-0.5.2.0", "--print-file-name", "libatomic.so"], 0xbfb4320c /* 69 vars */ <un
finished ...>
```
which is where gcc itself is called to do late linking of libraries in the ghc-prim-0.5.2.0 directory. I find that weird.
Later on there is a search of the ghc-prim .so for the symbol that is in fact in libpthreads, which fails, causing general death.
```openat(AT_FDCWD, "/usr/lib/ghc/ghc-prim-0.5.2.0/HSghc-prim-0.5.2.0.o", O_RDONLY|O_LARGEFILE) = 3
...
write(2, "/usr/lib/ghc/ghc-prim-0.5.2.0/HSghc-prim-0.5.2.0.o: unknown symbol `_GLOBAL_OFFSET_TABLE_'", 90/usr/lib/ghc/ghc-prim-0.5.2.0/HSghc-prim-0.5.2.0.o: unknown symbol `_GLOBAL_OFFSET_TABLE_') = 90
```

martijnbastiaan, 2019-04-24:
We passed `--dep-map pthread:libc6-dev` to `cabal-debian`. Does that help?

The full script we used for dependencies was:

```
#!/bin/sh
set -e -x -u
PKGNAME="$1"
PKGVERSION="$2"
PKG="$PKGNAME-$PKGVERSION"
cabal unpack $PKG
cd $PKG
cabal-debian --official --native --disable-tests --dep-map pthread:libc6-dev
dpkg-buildpackage
cd ..
sudo dpkg -i libghc-${PKGNAME}*${PKGVERSION}_*.deb
```

And we used it as:

```
./auto-package.sh concurrent-supply 0.1.8
./auto-package.sh exceptions 0.10.0
./auto-package.sh ghc-typelits-natnormalise 0.6.2
./auto-package.sh ghc-typelits-knownnat 0.6
./auto-package.sh ghc-typelits-extra 0.3
```

pbreuer, 2019-04-24:
Yes, i.e., no, that's only going to change the build dependencies in the debian control file, putting in a libc6-dev (right) instead of pthread (wrong), so it won't make a fundamental change. I already eliminated teh bad "pthread" line, and it would only have stopped debuild running, easily overridden with debuild -d, and it would not have stopped the make that debuild should run from running, nor the package shlibs search from going successfully. With a fakeroot make -f debian/rules binary one can take it or leave it. The --native might do more. I'll try that.
I have also checked that a direct cabal install (--reinstall) clash-ghc doesn't do anything different. It makes a clashi in ~/.cabal/bin with the same "doesn't load libpthread when ghc-prim needs it to" problem.

martijnbastiaan, 2019-04-24:
Wew, I really have no clue. It looks like you're doing this with the latest release right? We were building it on `master` instead. Maybe you could try that? If _that_ doesn't work, it's probably some issue with 32 vs 64 bit. If it works, we've probably accidentally fixed the issue in the meantime.

christiaanb, 2019-04-24:
If/when your build from master doesn’t work could you try adding: `-optl-lpthread` to https://github.com/clash-lang/clash-compiler/blob/6891e6b1beaeefa5c46d98c54026f137bad67f6b/clash-ghc/clash-ghc.cabal#L67 and https://github.com/clash-lang/clash-compiler/blob/6891e6b1beaeefa5c46d98c54026f137bad67f6b/clash-ghc/clash-ghc.cabal#L76

pbreuer, 2019-04-27:
Thanks for all your comments above.
To answer: I believe the source comes via cabal. It contains clash-ghc-0.99.3.tar.gz that I took away, untarred, debianized, and compiled. That produces the compile error noted above on ia32. When corrected with a s/I/I64/ in one place as noted, it compiles, but clashi complains about the TABLE symbol from libpthread being missing as above (google has it that this is a sporadic problem with ghci, but offers no definitive help except "fiddle"). I forced that symbol into the clashi executable but the complaint is the same, hence my deduction that a spawned invocation of gcc or ghc is involved.
I have spent the last couple of days pivoting my 64-bit desktop around enough partitions to put a linux debian testing on (which has ghc 8.4.4 - I failed at compiling 8.4.4 on debian stable, which has 8.0.1, and didn't feel like trying again as changing the distro is easier than compiling those 700 odd modules 10 times) and have repeated the exact same compile process I went through under ia32 under 64 bit this time, to make sure it's not me. The result comes out right.
I am working through the tutorial with clashi now. I note only that :vhdl  dumps on the first example (topEntity = exposeClockReset mac) with 
```
:vhdl
Loading dependencies took 2.896773193s
Compiling: MAC.topEntity
Applied 4 transformations
Normalisation took 0.434049299s
*** Exception: Clash.Netlist.Util(86): Not in normal form: tyArgs
```
which would definitely be something somewhere else than my tiny source file.

leonschoorl, 2019-04-29:
I've tried installing clash using cabal on 32bit debian unstable, and without any options I got the same problem:
```CLaSHi, version 0.99.3 (using clash-lib, version 0.99.3):
http://www.clash-lang.org/  :? for help
clashi: /usr/lib/ghc/ghc-prim-0.5.2.0/HSghc-prim-0.5.2.0.o: unknown symbol `_GLOBAL_OFFSET_TABLE_'
clashi: unable to load package `ghc-prim-0.5.2.0'
```
But when passing cabal the `--enable-executable-dynamic` option like in the [tutorial](http://hackage.haskell.org/package/clash-prelude-0.99.3/docs/Clash-Tutorial.html#g:2) this problem disappears.

After patching the decodeDouble_Int64 in the Evaluator clash more or less works.
But any code code using Word64 or Int64 quickly runs into missing primitives, while I could add [some of them](https://github.com/leonschoorl/clash-compiler/commit/2d4c9f3e983f637b5de4097aa4b3c478b1c9288e).

But then we run in to crazy stuff like:
```
Clash.Netlist.BlackBox(224): No blackbox found for: _INTERNAL_.{__pkg_ccall ghc-prim-0.5.2.0 Word64#
                           -> Int# -> State# RealWorld -> (# State# RealWorld, Word64# #)}.
```
Because on a 32bit system most 64 bit operations are [implemented in C](https://gitlab.haskell.org/ghc/ghc/blob/master/libraries/ghc-prim/cbits/longlong.c) and called through the [foreign function interface](https://gitlab.haskell.org/ghc/ghc/blob/master/libraries/ghc-prim/GHC/IntWord64.hs).
This will probably be difficult to support.

Is there any specific reason you're using a 32bit system?

As for the error `*** Exception: Clash.Netlist.Util(86): Not in normal form: tyArgs`, this probably means that clash is trying to translate something polymorphic.
Check that your `topEntity` has a monomorphic type (it has no type variables).

martijnbastiaan, 2019-04-29:
> Maybe there is some way to do the same through cabal-debian?

Maybe I'm confused, but that's why we added the cabal flag `-dynamic`, right?

leonschoorl, 2019-04-30:
Right, forgot about that.
This was the exact situation we added that flag for.
Because we could only pass package flags through cabal-debian, not arbitrary cabal flags.

Although up to now we thought the dynamic setting was mostly a performance thing, it makes clashi start a lot faster.
Not that it was required to not crash.

@pbreuer: This flag isn't in 0.99.3, you could copy [this commit](b37b9f200f556588f0fb571c48e37fbb4f1b421b) and then use `--cabal-flag dynamic` or you could just add the `-dynamic` flag to the `GHC-options` of the executables in `clash-ghc.cabal`.

leonschoorl, 2019-04-30:
This seems to be distro specific.
I tried it on openSUSE Tumbleweed 32bit, there it works both with and without `-dynamic` when building with cabal.
(But it has ghc-8.6.5, so is might not be a fair comparison)

martijnbastiaan, 2019-08-01:
Fixed by https://github.com/clash-lang/clash-compiler/pull/690. Please reopen if #690 did not fix it.

closed 2019-08-01 by martijnbastiaan