clash-lang/clash-compiler #588: Clash seems to ignore primitive annotations

adamwalker, 2019-04-23:
Compiling this file: https://gist.github.com/adamwalker/f7d0afc30c416a43807b771509a91ba0
Results in this output: https://gist.github.com/adamwalker/5ab19410c8be6e67c9514f3a33025bbc

Notice that the arithmetic operation is a subtraction instead of an addition, as specified in the primitive annotation. 

This example is almost exactly the InlinePrimitive example from the documentation. The only difference is that I've swapped the addition for a subtraction in the clash code only. 

I'm using the latest master (6891e6)

leonschoorl, 2019-04-23:
When you don't explicitly name your module, GHC will give it the name `Main`.
(This is why clash will create a it's output in a directory called Main)

So the fully qualified name of example is `Main.example`:
```diff
@@ -7,5 +7,5 @@
 {-# ANN example (InlinePrimitive VHDL $ unindent [i|
   [ { "BlackBox" :
-      { "name" : "example"
+      { "name" : "Main.example"
       , "kind": "Declaration"
       , "template" :
```

leonschoorl, 2019-04-23:
Ideally you should need write the name field for inline prims.
Clash should be able to infer the name from the fact that it's an annotation attached to the function.

martijnbastiaan, 2019-04-23:
Alternatively, it should throw and error if it can't find `example`.

martijnbastiaan, 2019-04-23:
For inspiration, look at: https://github.com/clash-lang/clash-compiler/pull/514

adamwalker, 2019-04-23:
Aha. Thanks. Dumb mistake on my part. 

Yeah, it would be nice if it threw an error. The file I provided is broken, but compilation succeeds anyway.

christiaanb, 2019-08-14:
Fixed by #641

martijnbastiaan, 2019-08-15:
#641 isn't ready to be merged yet! I'll work on it shortly.

martijnbastiaan, 2019-08-16:
Fixed in https://github.com/clash-lang/clash-compiler/pull/641.

closed 2019-08-16 by martijnbastiaan