clash-lang/clash-compiler #590: Functions in `Clash.Sized.Vector` should also be available for `BitVector`

martijnbastiaan, 2019-04-26:
A user on the mailing list wanted to use the functions defined for `Vec` on `BitVector`. Unfortunately, on large vectors Clash can struggle to translate these constructs. We should therefore reimplement a lot of the functions defined for `Vec` for `BitVector`.

jhmcstanton, 2019-10-02:
I'll pick this up. `Clash.Sized.Vector` exports quite a few functions, is there anything specific to prioritize? 

> Unfortunately, on large vectors Clash can struggle to translate these constructs

Is there an example usage somewhere of a vector large enough to cause a problem?

christiaanb, 2019-10-02:
@jhmcstanton:

top of my head:

```haskell
replicateB :: SNat n -> Bit -> BitVector n
mapB :: (Bit -> Bit) -> BitVector n -> BitVector n
zipWithB :: (Bit -> Bit -> Bit) -> BitVector n -> BitVector n -> BitVector n
foldlB :: (Bit -> Bit -> Bit) -> Bit -> BitVector n -> Bit
foldrB :: (Bit -> Bit -> Bit) -> Bit -> BitVector n -> Bit
iterateB :: SNat n -> (Bit -> Bit) -> Bit -> BitVector n
concatMapB :: (Bit -> BitVector n) -> BitVector m -> BitVector (n*m)
reverseB :: BitVector n -> BitVector n
scanlB :: (Bit -> Bit -> Bit) -> Bit -> BitVector n -> BitVector (n+1)
scanrB :: (Bit -> Bit -> Bit) -> Bit -> BitVector n -> BitVector (n+1)
```

Note that some of these must be implemented as "primitives", see http://hackage.haskell.org/package/clash-prelude-1.0.0/docs/Clash-Tutorial.html#g:14, to avoid the compile time "explosion" @martijnbastiaan was referring too. You can draw inspiration from `Vec` (see https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/prims/vhdl/Clash_Sized_Vector.json) as to which functions must be actual primitives. Evaluation rules for these primitives must then be added to the evaluator, https://github.com/clash-lang/clash-compiler/blob/c5a432776a32c92e3e78c47a9c6249d9ab816977/clash-ghc/src-ghc/Clash/GHC/Evaluator.hs#L118, as well.

I could setup a branch tonight that just implements `mapB`, so you know what needs to be done for the rest of the functions. Unless you want to do it all by yourself, let me know.

jhmcstanton, 2019-10-02:
I won't have a chance to dig around until this evening myself. If/when I run into issues I can let you know. Thanks for the guidance!

jhmcstanton, 2019-10-02:
To clarify, don't worry about making your own branch just yet.

christiaanb, 2020-01-18:
Bumping to 1.4 because I want to release 1.2 soon which contains a lot of fixes, and the implementation of this would take some effort: adding blackboxes, evaluator rules, etc.