clash-lang/clash-compiler #595: Warn when compiling with MonoLocalBinds disabled

christiaanb, 2019-05-02:
Due to complex interactions with GHCs optimizations and Clash’ optimizations, turning off MonoLocalBinds can lead to highly surprising logic duplication. We should warn the user about this, until it is fixed.

martijnbastiaan, 2019-08-14:
@mmzx Did some work on this: Clash now warns whenever a user compiles with `-XNoMonoLocalBinds`. We should check for `{-# LANGUAGE NoMonoLocalBinds #-}` in modules too, so I'm leaving this issue open for now.

mmzx, 2019-08-14:
>  We should check for `{-# LANGUAGE NoMonoLocalBinds #-}` in modules too, so I'm leaving this issue open for now.

Oh yes! It seems I have the solution at hand already, I am going to make a PR soon for it.

martijnbastiaan, 2019-08-14:
Awesome! Looking forward to merging that one too!

martijnbastiaan, 2019-08-16:
@mmzx Would you think it's possible to make a PR in ~the coming week? We'd like to include a fix for this in Clash 1.0, which is to be release the 3rd of September. If that's too much, don't worry, we'll pick it up ourselves.

mmzx, 2019-08-16:
Absolutely, please stay tuned until Monday.

martijnbastiaan, 2019-08-26:
Fixed in https://github.com/clash-lang/clash-compiler/commit/1c93e09b8fc720f7dd0ae30d7ebd79125950b4b5, thanks to @mmzx

closed 2019-08-26 by martijnbastiaan