clash-lang/clash-compiler #601: Issue with signed numbers and Vivado

adamwalker, 2019-05-11:
If I compile this design: https://gist.github.com/adamwalker/7249127077b46b867018fb763774d2af with Vivado, it does not work correctly. 
The generated Verilog is also in the gist. 
If I replace lines 76 and 78 of the generated Verilog with 
```
assign c$app_arg = $signed(i1[15:8]) + $signed(\input [15:8]);
assign c$app_arg_0 = $signed(i1[7:0]) + $signed(\input [7:0]);
```
Or
```
assign c$app_arg = {i1[15], i1[15:8]} + {\input [15], \input [15:8]};
assign c$app_arg_0 = {i1[7], i1[7:0]} + {\input [7], \input [7:0]};
```
it works.
So, it looks like Clash doesn't do the length extending addition correctly. It's possible though, that it's a Vivado bug. I'm not sure what the exact semantics of signed addition are in Verilog.

christiaanb, 2019-05-11:
Right, when Clash translates the following projection to Verilog:

```haskell
case tup of (a :: Signed 8, _) -> a
```

it should add the `$signed( ... )` conversion to the bit-slice.

christiaanb, 2019-05-11:
This bug shows up because we recently decided to inline these projections, before they would be assigned to a wire first.

adamwalker, 2019-06-06:
Is there any way I can work around this? I'm keen to get some DSP stuff going in Clash.

leonschoorl, 2019-06-06:
You could compile to VHDL.
Their clash uses record types whose components are properly typed.

leonschoorl, 2019-06-06:
Or you could do explicit resizes yourself by replacing `add` with:

```haskell
add' :: (KnownNat a, KnownNat b) => Signed a -> Signed b -> Signed ((Max a b)+1)
add' a b = resize a + resize b
```

martijnbastiaan, 2019-06-11:
@leonschoorl has fixed this issue in https://github.com/clash-lang/clash-compiler/commit/81c0c00eac0cd78493e8fb434aeca1ebab6005fd. Thanks for the report @adamwalker !

commit 81c0c00e (2019-06-11): Verilog: cast projection result to $signed when needed

According to the Verilog spec the type of a part-select expression(like
a[15:8]) is always unsigned.

Fixes #601
closed 2019-06-11 by martijnbastiaan