clash-lang/clash-compiler #607: CLaSh tries to synthesize the string in errorX

gergoerdi, 2019-05-26:
With the latest version of CLaSH (b8440afa1289593772c60e49f04ec5a000b59bc3) it seems the string passed to `errorX` is not discarded before synthesis. This can cause problems if the string expression is not a literal. For example:

```
module Top where

import Clash.Prelude

{-# NOINLINE topEntity #-}
{-# ANN topEntity
  (Synthesize
    { t_name   = "Bug"
    , t_inputs =
          [ PortName "CLK_25MHZ"
          , PortName "RESET"
          , PortName "INPUT"
          ]
    , t_output = PortName "OUTPUT"
    }) #-}
topEntity
    :: Clock System Source
    -> Reset System Asynchronous
    -> Signal System Bit
    -> Signal System Bit
topEntity = exposeClockReset $ fmap $ \b -> errorX ("foo" <> show b)

```

The error message is:

```
Top.hs:21:1: error:
    Clash.Netlist.BlackBox(225): No blackbox found for: GHC.CString.unpackAppendCString#. Did you forget to include directories containing primitives? You can use '-i/my/prim/dir' to achieve this.
    
    NB: The source location of the error is not exact, only indicative, as it is acquired after optimisations.
    The actual location of the error can be in a function that is inlined.
    To prevent inlining of those functions, annotate them with a NOINLINE pragma.
   |
21 | topEntity = exposeClockReset $ fmap $ \b -> errorX ("foo" <> show b)
   | ^^^^^^^^^
```

martijnbastiaan, 2019-05-28:
I expected to find a missing blackbox for `errorX` or a missing `NOINLINE` pragma for it. The latter turns out to be true, but we also explicitly expect it to NOT have a `NOINLINE` pragma: 

https://github.com/clash-lang/clash-compiler/blob/7a1d2723f9123faef52bba7530db65a143d4da4f/clash-ghc/src-ghc/Clash/GHC/GenerateBindings.hs#L189-L191

I know I've been surprised by this before, but I forgot why we're doing it. @leonschoorl ?

martijnbastiaan, 2019-05-28:
Fixed, thanks @gergoerdi !

closed 2019-05-28 by martijnbastiaan

leonschoorl, 2019-05-29:
@martijnbastiaan `errorX` is excluded from those warnings because we know it will always generate the second warning (the one about being bottom/producing an error).

We should probably be more specific about that.
And also the `GHC.*` is way too broad.

PR #609 (2019-05-28): Count term instead of type arguments to calculate arity
`removeUnusedExpr` wasn't removing enough unused expresssions
because it counted type arguments instead of term arguments
to determine the arity of a primitive.

NB we need the arity so that `removeUnusedExpr` doesn't remove
arguments in over-applied expressions.

Fixes #607
commit 9a466bd9 (2019-05-28): Count term instead of type arguments to calculate arity (#609)

`removeUnusedExpr` wasn't removing enough unused expresssions
because it counted type arguments instead of term arguments
to determine the arity of a primitive.

NB we need the arity so that `removeUnusedExpr` doesn't remove
arguments in over-applied expressions.

Fixes #607