clash-lang/clash-compiler #624: Bad scaling from a branch

gergoerdi, 2019-06-07:
Please find a full standalone compilable code at the end of this post.

I am experiencing very bad scaling from the CLaSH synthesizer when it is compiling the instruction fetch/decode function for an Intel 8080. If I try to compile my code as-is, it just eats up all memory in a couple minutes. However, if I start commenting out most branches of `fetchInstr`, then it can finish quite quickly; for example, with just the following branches remaining, synthesis takes a mere 15 seconds:

```
fetchInstr :: (Monad m) => m (Unsigned 8) -> m Instr
fetchInstr fetch = do
    b1 <- fetch
    let b1'@(_ :> _ :> d2@r :> d1@p :> d0 :> s2 :> s1 :> s0 :> Nil) = bitCoerce b1 :: Vec 8 Bit
        d = decodeOp $ bitCoerce (d2, d1, d0)
        s = decodeOp $ bitCoerce (s2, s1, s0)
        rp = decodeRP r p
        rppp = decodeRPPP r p
        cond = decodeCond (d2 :> d1 :> d0 :> Nil)
    case b1' of
        0 :> 1 :> _ :> _ :> _ :> _ :> _ :> _ :> Nil -> return $ MOV d (Op s)
        0 :> 0 :> _ :> _ :> _ :> 1 :> 1 :> 0 :> Nil -> MOV d <$> Imm <$> fetch
        0 :> 0 :> _ :> _ :> 0 :> 0 :> 0 :> 1 :> Nil -> LXI rp <$> fetch16
        0 :> 1 :> 1 :> 1 :> 0 :> 1 :> 1 :> 0 :> Nil -> return HLT
        0 :> 0 :> 0 :> 0 :> 0 :> 0 :> 0 :> 0 :> Nil -> return NOP
        _ -> return NOP -- "Unofficial" NOP
  where
    fetch16 = do
        lo <- fetch
        hi <- fetch
        return $ bitCoerce (hi, lo)
```
But as I keep re-adding more, the synthesis speed goes down and memory usage goes up quickly. For an intermediate example, this takes 56 seconds to synthesize:

```
fetchInstr :: (Monad m) => m (Unsigned 8) -> m Instr
fetchInstr fetch = do
    b1 <- fetch
    let b1'@(_ :> _ :> d2@r :> d1@p :> d0 :> s2 :> s1 :> s0 :> Nil) = bitCoerce b1 :: Vec 8 Bit
        d = decodeOp $ bitCoerce (d2, d1, d0)
        s = decodeOp $ bitCoerce (s2, s1, s0)
        rp = decodeRP r p
        rppp = decodeRPPP r p
        cond = decodeCond (d2 :> d1 :> d0 :> Nil)
    case b1' of
        0 :> 1 :> _ :> _ :> _ :> _ :> _ :> _ :> Nil -> return $ MOV d (Op s)
        0 :> 0 :> _ :> _ :> _ :> 1 :> 1 :> 0 :> Nil -> MOV d <$> Imm <$> fetch
        0 :> 0 :> _ :> _ :> 0 :> 0 :> 0 :> 1 :> Nil -> LXI rp <$> fetch16

        0 :> 0 :> 1 :> 1 :> 1 :> 0 :> 1 :> 0 :> Nil -> LDA <$> fetch16
        0 :> 0 :> 1 :> 1 :> 0 :> 0 :> 1 :> 0 :> Nil -> STA <$> fetch16
        0 :> 0 :> 1 :> 0 :> 1 :> 0 :> 1 :> 0 :> Nil -> LHLD <$> fetch16
        0 :> 0 :> 1 :> 0 :> 0 :> 0 :> 1 :> 0 :> Nil -> SHLD <$> fetch16
        0 :> 0 :> _ :> _ :> 1 :> 0 :> 1 :> 0 :> Nil -> return $ LDAX rp
        0 :> 0 :> _ :> _ :> 0 :> 0 :> 1 :> 0 :> Nil -> return $ STAX rp
        1 :> 1 :> 1 :> 0 :> 1 :> 0 :> 1 :> 1 :> Nil -> return XCHG

        1 :> 0 :> 0 :> 0 :> 0 :> _ :> _ :> _ :> Nil -> return $ ALU ADD (Op s)
        1 :> 1 :> 0 :> 0 :> 0 :> 1 :> 1 :> 0 :> Nil -> ALU ADD <$> Imm <$> fetch
        1 :> 0 :> 0 :> 0 :> 1 :> _ :> _ :> _ :> Nil -> return $ ALU ADC (Op s)
        1 :> 1 :> 0 :> 0 :> 1 :> 1 :> 1 :> 0 :> Nil -> ALU ADC <$> Imm <$> fetch

        0 :> 1 :> 1 :> 1 :> 0 :> 1 :> 1 :> 0 :> Nil -> return HLT
        0 :> 0 :> 0 :> 0 :> 0 :> 0 :> 0 :> 0 :> Nil -> return NOP
        _ -> return NOP -- "Unofficial" NOP
```

Full code with no external dependencies:

```
{-# LANGUAGE RecordWildCards, ApplicativeDo #-}
{-# LANGUAGE GeneralizedNewtypeDeriving, DerivingStrategies #-}
module Bug where

import Clash.Prelude hiding (lift)
[... 353 lines cut ...]
          , PortName "RESET"
          ]
    , t_output = PortName "ADDR"
    }) #-}
topEntity
    :: Clock System Source
    -> Reset System Asynchronous
    -> Signal System Addr
topEntity = exposeClockReset mainBoard

mainBoard
    :: (HiddenClockReset domain gated synchronous)
    => Signal domain Addr
mainBoard = cpuOut
  where
    cpuOut = mealyState (runCPU defaultOut cpu) initState cpuIn

    progROM addr = unpack <$> romFile (SNat @13) "image.hex" (truncateB <$> addr)
    memRead = progROM
    read = memRead cpuOut

    cpuIn = do
        cpuInMem <- read
        pure CPUIn{..}
```

gergoerdi, 2019-06-07:
As an additional data point, keeping the following results in synthesis time of 5m34s, memory peaking somewhere around 8G:

```
fetchInstr :: (Monad m) => m (Unsigned 8) -> m Instr
fetchInstr fetch = do
    b1 <- fetch
    let b1'@(_ :> _ :> d2@r :> d1@p :> d0 :> s2 :> s1 :> s0 :> Nil) = bitCoerce b1 :: Vec 8 Bit
        d = decodeOp $ bitCoerce (d2, d1, d0)
        s = decodeOp $ bitCoerce (s2, s1, s0)
        rp = decodeRP r p
        rppp = decodeRPPP r p
        cond = decodeCond (d2 :> d1 :> d0 :> Nil)
    case b1' of
        0 :> 1 :> _ :> _ :> _ :> _ :> _ :> _ :> Nil -> return $ MOV d (Op s)
        0 :> 0 :> _ :> _ :> _ :> 1 :> 1 :> 0 :> Nil -> MOV d <$> Imm <$> fetch
        0 :> 0 :> _ :> _ :> 0 :> 0 :> 0 :> 1 :> Nil -> LXI rp <$> fetch16

        0 :> 0 :> 1 :> 1 :> 1 :> 0 :> 1 :> 0 :> Nil -> LDA <$> fetch16
        0 :> 0 :> 1 :> 1 :> 0 :> 0 :> 1 :> 0 :> Nil -> STA <$> fetch16
        0 :> 0 :> 1 :> 0 :> 1 :> 0 :> 1 :> 0 :> Nil -> LHLD <$> fetch16
        0 :> 0 :> 1 :> 0 :> 0 :> 0 :> 1 :> 0 :> Nil -> SHLD <$> fetch16
        0 :> 0 :> _ :> _ :> 1 :> 0 :> 1 :> 0 :> Nil -> return $ LDAX rp
        0 :> 0 :> _ :> _ :> 0 :> 0 :> 1 :> 0 :> Nil -> return $ STAX rp
        1 :> 1 :> 1 :> 0 :> 1 :> 0 :> 1 :> 1 :> Nil -> return XCHG

        1 :> 0 :> 0 :> 0 :> 0 :> _ :> _ :> _ :> Nil -> return $ ALU ADD (Op s)
        1 :> 1 :> 0 :> 0 :> 0 :> 1 :> 1 :> 0 :> Nil -> ALU ADD <$> Imm <$> fetch
        1 :> 0 :> 0 :> 0 :> 1 :> _ :> _ :> _ :> Nil -> return $ ALU ADC (Op s)
        1 :> 1 :> 0 :> 0 :> 1 :> 1 :> 1 :> 0 :> Nil -> ALU ADC <$> Imm <$> fetch

        1 :> 0 :> 0 :> 1 :> 0 :> _ :> _ :> _ :> Nil -> return $ ALU SUB (Op s)
        1 :> 1 :> 0 :> 1 :> 0 :> 1 :> 1 :> 0 :> Nil -> ALU SUB <$> Imm <$> fetch
        1 :> 0 :> 0 :> 1 :> 1 :> _ :> _ :> _ :> Nil -> return $ ALU SBB (Op s)
        1 :> 1 :> 0 :> 1 :> 1 :> 1 :> 1 :> 0 :> Nil -> ALU SBB <$> Imm <$> fetch

        1 :> 0 :> 1 :> 0 :> 0 :> _ :> _ :> _ :> Nil -> return $ ALU AND (Op s)
        1 :> 1 :> 1 :> 0 :> 0 :> 1 :> 1 :> 0 :> Nil -> ALU AND <$> Imm <$> fetch

        1 :> 0 :> 1 :> 1 :> 0 :> _ :> _ :> _ :> Nil -> return $ ALU OR (Op s)
        1 :> 1 :> 1 :> 1 :> 0 :> 1 :> 1 :> 0 :> Nil -> ALU OR <$> Imm <$> fetch

        1 :> 0 :> 1 :> 0 :> 1 :> _ :> _ :> _ :> Nil -> return $ ALU XOR (Op s)
        1 :> 1 :> 1 :> 0 :> 1 :> 1 :> 1 :> 0 :> Nil -> ALU XOR <$> Imm <$> fetch

        1 :> 0 :> 1 :> 1 :> 1 :> _ :> _ :> _ :> Nil -> return $ ALU CMP (Op s)
        1 :> 1 :> 1 :> 1 :> 1 :> 1 :> 1 :> 0 :> Nil -> ALU CMP <$> Imm <$> fetch

        0 :> 0 :> _ :> _ :> _ :> 1 :> 0 :> 0 :> Nil -> return $ INR d
        0 :> 0 :> _ :> _ :> _ :> 1 :> 0 :> 1 :> Nil -> return $ DCR d
        0 :> 0 :> _ :> _ :> 0 :> 0 :> 1 :> 1 :> Nil -> return $ INX rp
        0 :> 0 :> _ :> _ :> 1 :> 0 :> 1 :> 1 :> Nil -> return $ DCX rp

        0 :> 0 :> _ :> _ :> 1 :> 0 :> 0 :> 1 :> Nil -> return $ DAD rp
        0 :> 0 :> 1 :> 0 :> 0 :> 1 :> 1 :> 1 :> Nil -> return DAA

        0 :> 1 :> 1 :> 1 :> 0 :> 1 :> 1 :> 0 :> Nil -> return HLT
        0 :> 0 :> 0 :> 0 :> 0 :> 0 :> 0 :> 0 :> Nil -> return NOP
        _ -> return NOP -- "Unofficial" NOP
```

martijnbastiaan, 2019-06-07:
I'm guessing this is due to every `:>` representing a different (GADT) case. I don't think it'll be trivial to speed this case up. (I won't be able to look at in detail til at least after the weekend, though, so I'm not sure.) Could you try:

```haskell
    case b1 of
        --     mask          value
        ((.&.) 0b11000000 -> 0b01000000) -> return $ MOV d (Op s)
        ((.&.) 0b11000111 -> 0b01000110) -> MOV d <$> Imm <$> fetch
[..]
```

This should be _much_ quicker in simulation, as well as Clash compilation. If that works, I guess you could write a template haskell function as a workaround..

```haskell
    case b1 of
        --     mask          value
        $(casex "01xxxxxx") -> return $ MOV d (Op s)
        $(casex "01xxx110") -> MOV d <$> Imm <$> fetch
[..]
```

I'm more than happy to write it for you if the first example actually solves your issues.

gergoerdi, 2019-06-07:
So does that mean it should also work to deconstruct it into a `(Bit, Bit, Bit, Bit, Bit, Bit, Bit, Bit)` and pattern-match on that? I'd prefer that to manual bitmasking.

gergoerdi, 2019-06-07:
So a quick experiment shows the 8-tuple of `Bit`s has the same problem as the `Vec 8 Bit`, even though it's not a GADT. I guess because it is still compiled to nested pattern matches coordinate-by-coordinate.

Your TH idea looks like a pragmatic compromise, so I am going to go with that for now. For reference, my implementation is:

```
import Clash.Prelude
import Language.Haskell.TH
import Data.List as L
import Data.Maybe (fromMaybe)

fromBits :: [Bit] -> Integer
fromBits = L.foldl (\v b -> v `shiftL` 1 .|. fromIntegral b) 0

bitMask :: String -> Q Pat
bitMask s = [p| (($mask .&.) -> $pattern) |]
  where
    bs = parse <$> s

    mask = litE . IntegerL . fromBits $ maybe 0 (const 1) <$> bs
    pattern = litP . IntegerL . fromBits $ fromMaybe 0 <$> bs

    parse 'x' = Nothing
    parse '_' = Nothing
    parse '0' = Just 0
    parse '1' = Just 1

```

martijnbastiaan, 2019-06-07:
Thanks for posting the snippet @gergoerdi. Embarrassingly, I didn't know you could use `Q Pat`, so I was hacking on an uglier approach. Could you confirm the workaround works? I'll suggest it for inclusion in the Prelude if it does.

gergoerdi, 2019-06-07:
For my stripped-down test case from this ticket, it seems the bitmask view patterns do indeed work:

```
GHC+Clash: Loading modules cumulatively took 13.849s
Clash: Parsing and compiling primitives took 0.623s
Clash: Compiling Bug.topEntity
Clash: Applied 1762 transformations
Clash: Normalisation took 11.658s
Clash: Netlist generation took 0.292s
WARNING: file "image.hex" does not exist
Clash: Total compilation took 26.042s
```

Unfortunately, my original code is still synthesizing: it's been running for 10 minutes now, but its memory usage is steady at just below 4 GB so I'll keep it running for a bit.

I'm happy to prepare a pull request for `clash-prelude` with the `bitMask` macro; would `Clash.Prelude.TH.BitMask` work for the name? Or `Clash.Prelude.BitMask.TH`? Or just `Clash.Prelude.BitMask`?

gergoerdi, 2019-06-07:
```
GHC+Clash: Loading modules cumulatively took 24.404s
Clash: Parsing and compiling primitives took 0.647s
Clash: Compiling SpaceInvaders.topEntity
Clash: Applied 74909 transformations
Clash: Normalisation took 15m40.465s
Clash: Netlist generation took 5.340s
Clash: Total compilation took 16m15.785s
Build completed in 16m16s
```

martijnbastiaan, 2019-06-07:
I'd be fine with having it in `Clash.Sized.Internal.BitVector` and exporting it in `Clash.Sized.BitVector`, `Clash.Prelude`, and `Clash.Explicit.Prelude`.  To keep it consistent with `bLit`, we'd need to accept `.` instead of `_`/`x`. I'm wondering if there's a more obvious name than `bitMask`..

martijnbastiaan, 2019-06-07:
This module might also be interesting: https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/src/Clash/Annotations/BitRepresentation.hs

gergoerdi, 2019-06-07:
`bitPattern` maybe?

martijnbastiaan, 2019-06-07:
Perfect.

martijnbastiaan, 2019-06-07:
One option to actually address the core of this issue is to recognize when a pattern is "bit constant", and then replace it with a `BitPat` - which we'd need to add to `Pat`:

https://github.com/clash-lang/clash-compiler/blob/f4a5d1752e2ea5df2d104ce2e75c46f76db58f2a/clash-lib/src/Clash/Core/Term.hs#L65-L73

If _all_ alternatives of a case are `BitPat`s, we could even print a nice `casex` in the resulting HDL.