clash-lang/clash-compiler #629: Normalize: Expr belonging to bndr: elT0938307 not found

TheZoq2, 2019-06-07:
After adding a large chunk of code to my project, I started getting the following error:

```
Loading dependencies took 4.349083517s
Compiling: CompositeVideo.topEntity

<no location info>: error:
    Clash error call:
    Clash.Normalize(189): Expr belonging to bndr: elT098307 not found
```

The project in question is here: https://github.com/TheZoq2/OSD/tree/41c2e91b097e9234187791eacdef3f7da986773f

So far, i've traced it to https://github.com/TheZoq2/OSD/blob/41c2e91b097e9234187791eacdef3f7da986773f/clash/LineDrawer.hs#L191 which if replaced with a constant value, fixes the issue.

TheZoq2, 2019-06-07:
Looks like I linked to the wrong line before, it should be 178, not 191.

It also turns out that the issue is caused by the use of the `fold` function, as replacing that with `foldl1` fixes the issue

christiaanb, 2019-06-07:
It seems this bug is already fixed in `master`

TheZoq2, 2019-06-12:
Great, since I managed to work around it as well, I guess I'll close this issue :)

closed 2019-06-12 by TheZoq2