clash-lang/clash-compiler #630: Symbol `fromSLV` can't be used as a prefix in a selected name.

gergoerdi, 2019-06-08:
I have no compact way of reproducing this, yet, but when synthesizing a whole Space Invaders arcade machine into VHDL, I am getting the following error messages from the Xilinx ISE in the "check syntax" phase: 

```
SpaceInvaders.vhdl Line 57776. tup2_27_sel1_maybe_5 is not an element of fromSLV type tup3_11.
SpaceInvaders.vhdl Line 57776. Symbol fromSLV can't be used as a prefix in a selected name.
SpaceInvaders.vhdl Line 57776. Invalid selected name fromSLV.tup2_27_sel1_maybe_5.
SpaceInvaders.vhdl Line 57915. tup2_27_sel0_unsigned is not an element of fromSLV type tup3_11.
SpaceInvaders.vhdl Line 57915. Symbol fromSLV can't be used as a prefix in a selected name.
SpaceInvaders.vhdl Line 57915. Invalid selected name fromSLV.tup2_27_sel0_unsigned.
SpaceInvaders.vhdl Line 89918. tup2_27_sel1_maybe_5 is not an element of fromSLV type tup3_11.
SpaceInvaders.vhdl Line 89918. Symbol fromSLV can't be used as a prefix in a selected name.
SpaceInvaders.vhdl Line 89918. Invalid selected name fromSLV.tup2_27_sel1_maybe_5.
SpaceInvaders.vhdl Line 90057. tup2_27_sel0_unsigned is not an element of fromSLV type tup3_11.
SpaceInvaders.vhdl Line 90057. Symbol fromSLV can't be used as a prefix in a selected name.
SpaceInvaders.vhdl Line 90057. Invalid selected name fromSLV.tup2_27_sel0_unsigned.
SpaceInvaders.vhdl Line 122060. tup2_27_sel1_maybe_5 is not an element of fromSLV type tup3_11.
SpaceInvaders.vhdl Line 122060. Symbol fromSLV can't be used as a prefix in a selected name.
SpaceInvaders.vhdl Line 122060. Invalid selected name fromSLV.tup2_27_sel1_maybe_5.
SpaceInvaders.vhdl Line 122199. tup2_27_sel0_unsigned is not an element of fromSLV type tup3_11.
SpaceInvaders.vhdl Line 122199. Symbol fromSLV can't be used as a prefix in a selected name.
SpaceInvaders.vhdl Line 122199. Invalid selected name fromSLV.tup2_27_sel0_unsigned.
```

If I look at the generated files, `SpaceInvaders.vhdl` is 122829 lines long which looks like it could be the basis for season 2 of the Chernobyl miniseries. I have no idea how to start cutting down on the input CLaSH side to focus in on the problem. On the other hand, `spaceinvaders_types.vhdl` is a mere 1063 lines, and indeed it contains a lot of overloads for `fromSLV`, including ones for `tup2_27`:

```
  type tup2_27 is record
    tup2_27_sel0_unsigned : unsigned(7 downto 0);
    tup2_27_sel1_maybe_5 : spaceinvaders_types.maybe_5;
  end record;

  function toSLV (p : spaceinvaders_types.tup2_27) return std_logic_vector;
  function fromSLV (slv : in std_logic_vector) return spaceinvaders_types.tup2_27;

  function toSLV (p : spaceinvaders_types.tup2_27) return std_logic_vector is
  begin
    return (toSLV(p.tup2_27_sel0_unsigned) & toSLV(p.tup2_27_sel1_maybe_5));
  end;
  function fromSLV (slv : in std_logic_vector) return spaceinvaders_types.tup2_27 is
  alias islv : std_logic_vector(0 to slv'length - 1) is slv;
  begin
    return (fromSLV(islv(0 to 7)),fromSLV(islv(8 to 16)));
  end;
```

gergoerdi, 2019-06-08:
I've uploaded my full CLaSH source code to https://github.com/gergoerdi/clash-issue-630. The version of CLaSH I am using is 9a466bd9c08b8ff8fc215bfbca318873e6d57e4a.

I am happy to start cutting down on my code if I can get some guidance on where to even start.

gergoerdi, 2019-06-08:
I tried synthesizing to Verilog instead, but that output has now kept the Xilinx ISE busy for 2.5 hours by now... :scream:

gergoerdi, 2019-06-08:
I'm bailing out on the Verilog version after 4.5 hours.

gergoerdi, 2019-06-11:
One thing I've now found out, and will open a new ticket about it with a minimal way of reproducing it once I have time to prepare one, is that the 120,000 lines output is at least partially the result of a LOT of duplication that stems from the following part:

```
        (vgaR, vgaG, vgaB) = unbundle $ mux (bitToBool <$> pixel) fg bg
          where
            bg = pure (0x0, 0x0, 0x0)
            fg = pure (0xf, 0xf, 0xf)
```

(in a later version I plan to change `fg` and `bg` based on the screen coordinates, hence the weird way of writing it)

This seems to destroy sharing of the `pixel` signal, thus generating e.g. three copies of the video framebuffer, and everything dangling from it -- including the CPU itself...

If I replace it with

```
        vgaR = mux (bitToBool <$> pixel) 0xf 0x0
        vgaG = mux (bitToBool <$> pixel) 0xf 0x0
        vgaB = mux (bitToBool <$> pixel) 0xf 0x0
```

then the output is "only" 40,000 LOC and contains only one copy of the framebuffer block RAM and everything else.

christiaanb, 2019-06-25:
@martijnbastiaan @leonschoorl the solution for this is to change:

https://github.com/clash-lang/clash-compiler/blob/81163f7404a93531bc682760a0e3b81a227c6ca7/clash-lib/src/Clash/Backend/VHDL.hs#L1578-L1582

in such a way that __whenever__ the modifier of an identifier __contains__ a `Nested`, is to emit the following VHDL: `typemark'(fromSLV(slice(toSLV(xyz),start,end)))`, where this `slice` function is something we'll need to add to the `_types` package, which will do the actual bit-slicing. You can reuse: https://github.com/clash-lang/clash-compiler/blob/81163f7404a93531bc682760a0e3b81a227c6ca7/clash-lib/src/Clash/Backend/Verilog.hs#L550-L558 to calculate `start` and `end`.

christiaanb, 2020-04-20:
Fixed by https://github.com/clash-lang/clash-compiler/pull/1263

closed 2020-04-20 by christiaanb