clash-lang/clash-compiler #633: `unbundle` destroys sharing

gergoerdi, 2019-06-13:
(CLaSH version: 9a466bd9c08b8ff8fc215bfbca318873e6d57e4a)

In the following, heavily cut-down version of my code, the `unbundle` combinator used to set `vgaR`, `vgaG` and `vgaB` destroys the sharing of `pixel`, leading to three copies of the video RAM `vidRAM` and everything else dangling off it, in the generated VHDL file. Needless to say, this can have catastrophic consequences when e.g. three copies of a block RAM is more than what fits on a board etc.

Full code with no external dependencies:

```
{-# LANGUAGE RecordWildCards, ApplicativeDo #-}
module SpaceInvaders where

import Clash.Prelude
import Data.Maybe (fromMaybe, isJust)
import Control.Monad (guard)
import Control.Monad.State

mealyState :: (HiddenClockReset domain gated synchronous, Undefined s)
           => (i -> State s o) -> s -> (Signal domain i -> Signal domain o)
mealyState f s0 = mealy step s0
  where
    step s x = let (y, s') = runState (f x) s
               in (s', y)

runCPU :: (s -> CPUOut) -> (i -> State s CPUOut)
runCPU mkDef _ = gets mkDef

type Value = Unsigned 8
type Addr = Unsigned 16

data CPUState = CPUState
    { pc :: Addr
    }
    deriving (Show, Generic, Undefined)

initState :: CPUState
initState = CPUState
    { pc = 0x0000
    }

data CPUOut = CPUOut
    { cpuOutMemAddr :: Addr
    , cpuOutMemWrite :: Maybe Value
    }
    deriving (Generic)

defaultOut :: CPUState -> CPUOut
defaultOut CPUState{..} = CPUOut{..}
  where
    cpuOutMemAddr = pc
    cpuOutMemWrite = Nothing

type VidX = 256
type VidY = 256
type VidSize = VidX * VidY `Div` 8

{-# NOINLINE topEntity #-}
{-# ANN topEntity
  (Synthesize
    { t_name   = "SpaceInvaders"
    , t_inputs =
          [ PortName "CLK_25MHZ"
          , PortName "RESET"
          ]
    , t_output = PortProduct ""
            [ PortName "VGA_RED"
            , PortName "VGA_GREEN"
            , PortName "VGA_BLUE"
            ]
    }) #-}
topEntity
    :: Clock System Source
    -> Reset System Asynchronous
    -> ( ( Signal System (Unsigned 4)
        , Signal System (Unsigned 4)
        , Signal System (Unsigned 4)
        )
      )
topEntity = exposeClockReset board
  where
    board = (vgaR, vgaG, vgaB)
      where
        cnt = register (0 :: Unsigned 8) $ cnt + 1
        vgaX' = Just <$> cnt
        vgaY' = Just <$> cnt

        vidWrite = mainBoard
        vidRAM addr = blockRam (pure 0x00 :: Vec VidSize Value) addr vidWrite

        pixel = mux visible ((!) <$> vidRAM pixAddr <*> pixBit) (pure low)
          where
            visible = isJust <$> vgaX' .&&. isJust <$> vgaY'
            (pixAddr, pixBit) = unbundle $ do
                x <- fromMaybe 0 <$> vgaX'
                y <- fromMaybe 0 <$> vgaY'
                pure (bitCoerce (x, y) :: (Index VidSize, Unsigned 3))

        (vgaR, vgaG, vgaB) = unbundle $ mux (bitToBool <$> pixel) fg bg
          where
            bg = pure (0x0, 0x0, 0x0)
            fg = pure (0xf, 0xf, 0xf)

mainBoard
    :: (HiddenClockReset domain gated synchronous)
    => Signal domain (Maybe (Index VidSize, Value))
mainBoard = do
    addr <- cpuOutMemAddr <$> cpuOut
    memWrite <- cpuOutMemWrite <$> cpuOut
    pure $ do
        guard $ 0x2400 <= addr && addr <= 0x4000
        x <- memWrite
        return $ (fromIntegral (truncateB @_ @13 (addr - 0x2400)), x)
  where
    cpuOut = mealyState (runCPU defaultOut) initState (pure ())
```

Note that by changing it to

        vgaR = mux (bitToBool <$> pixel) (pure 0xf) (pure 0xf)
        vgaG = mux (bitToBool <$> pixel) (pure 0xf) (pure 0xf)
        vgaB = mux (bitToBool <$> pixel) (pure 0xf) (pure 0x0)

the un-sharing goes away and the output looks OK.

christiaanb, 2019-06-13:
Workaround, change:

```
            bg = pure (0x0, 0x0, 0x0)
            fg = pure (0xf, 0xf, 0xf)
```

to 

```
            bg = pure (0x0 :: Unsigned 4, 0x0 :: Unsigned 4, 0x0 :: Unsigned 4)
            fg = pure (0xf :: Unsigned 4, 0xf :: Unsigned 4, 0xf :: Unsigned 4)
```

christiaanb, 2019-06-13:
So what happened is that your code is more polymorphic than you might have imaged, and so the code the Clash compiler is getting from GHC is something along the lines of:

```
f :: (Num a, Num b, Num c) => ... -> (a,b,c)
f ... = ...
  where
    pixel = ...

board = (vgaR, vgaG, vgaB) 
 where
  (case f @(Unsigned 4) @Integer @Integer numUnsignedDict numIntegerDict ... of (x,y,z) -> x
  ,case f @Integer @(Unsigned 4) @Integer ... of (x,y,z) -> y
  ,case f @Integer @Integer @(Unsigned 4) ... of (x,y,z) -> z
  )
```
and so Clash creates 3 different copies of `f` because they get instantiated to different types. Subsequently, Clash' common-subexpression-elimination (CSE) can't recover the lost sharing.

I dunno exactly what's going on here, but the fact that GHC is picking `Integer` means there's probably some defaulting going on.

Perhaps, if you mean elements to have the same type, in a polymorphic setting, is touse `Vec`, now that we have GADT pattern matching support:

```
(vgaR :> vgaG :> vgaB :> Nil) = unbundle $ mux (bitToBool <$> pixel) fg bg
          where
            bg = pure (0x0 :> 0x0 :> 0x0 :> Nil)
            fg = pure (0xf :> 0xf :> 0xf :> Nil)
```            

Hopefully the above gives some guidance on how to work around the issue in your original code.

gergoerdi, 2019-06-15:
Interesting! Unfortunately, the vector solution is not going to work for all cases for me -- some of my target boards have 3/3/2 bits for red/green/blue, so they are not homogeneous in size.

The explanation made me hope that by turning on [`MonoLocalBinds`](https://downloads.haskell.org/~ghc/latest/docs/html/users_guide/glasgow_exts.html#extension-MonoLocalBinds) I could work around this, but alas, it doesn't apply in this case.

Just for reference, I ended up going with a type signature

      fg, bg: _ (Unsigned 4, Unsigned 4, Unsigned 4)

christiaanb, 2020-04-20:
Perhaps this issue can be solved by adding back the monomorphism restriction, which clash has currently disabled, by adding `{-# LANGUAGE MonomorphismRestriction #-}` to the file

gergoerdi, 2020-05-25:
Maybe worth linking to #1270.