clash-lang/clash-compiler #637: Compilation error: "Expected identifier"

adamwalker, 2019-06-14:
Compiling this file: https://gist.github.com/adamwalker/28cf5984e3564fc0d46a246784c76c0d, with the latest clash (ffc98a), results in this error:
```
<no location info>: error:
    Clash error call:
    Expected Identifier: BlackBoxE "Clash.Sized.Vector.unconcat" [] [] [] [Arg False 2] (Context {bbResult = (Identifier "\\c$_INTERNAL_.ws1_app_arg_6 " Nothing,Vector 8 (Vector 2 (Signed 18))), bbInputs = [(BlackBoxE "Clash.Transformations.removedArg" [] [] [] [Err Nothing] (Context {bbResult = (Identifier "\\c$_INTERNAL_.ws1_app_arg_6 " Nothing,Unsigned 64), bbInputs = [], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 0, bbCompName = "fir"}) True,Unsigned 64,True),(BlackBoxE "Clash.Transformations.removedArg" [] [] [] [Err Nothing] (Context {bbResult = (Identifier "\\c$_INTERNAL_.ws1_app_arg_6 " Nothing,Unsigned 64), bbInputs = [], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 0, bbCompName = "fir"}) True,Unsigned 64,True),(BlackBoxE "Clash.Sized.Vector.replicate" [] [] [] [Text "{",Lit 0,Text " {",Arg False 1,Text "}}"] (Context {bbResult = (Identifier "\\c$_INTERNAL_.ws1_app_arg_6 " Nothing,Vector 16 (Signed 18)), bbInputs = [(DataCon (Unsigned 64) (DC (Void Nothing,-1)) [Literal (Just (Unsigned 64,64)) (NumLit 16)],Unsigned 64,True),(BlackBoxE "Clash.Sized.Internal.Signed.fromInteger#" [] [] [] [IF (CmpLE (Size (Typ Nothing)) (Size (Typ (Just 1)))) [Text "$signed(",Var [Text "c$i_211"] 1,Text "[0+:",Size (Typ Nothing),Text "])"] [Text "$signed({{(",Size (Typ Nothing),Text "-",Size (Typ (Just 1)),Text ") {1'b0}},",Var [Text "c$i_211"] 1,Text "})"]] (Context {bbResult = (Identifier "\\c$_INTERNAL_.ws1_app_arg_6 " Nothing,Signed 18), bbInputs = [(Literal (Just (Unsigned 64,64)) (NumLit 18),Unsigned 64,True),(Literal (Just (Signed 64,64)) (NumLit 0),Signed 64,True)], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 0, bbCompName = "fir"}) True,Signed 18,True)], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 0, bbCompName = "fir"}) True,Vector 16 (Signed 18),True)], bbFunctions = fromList [], bbQsysIncName = [], bbLevel = 0, bbCompName = "fir"}) True
```
Furthermore, if I recompile, it succeeds, but produces broken verilog because it used a broken cached result from the failing run.

martijnbastiaan, 2019-06-14:
[This commit](https://github.com/clash-lang/clash-compiler/commit/eb70dfab4ac5926bd0adf7ffe5bd31026e19bcca) adds a special codepath for `index_int` used on constants when producing verilog. Apparently it's broken. As a workaround, you can try to generate vhdl or systemverilog.

> Furthermore, if I recompile, it succeeds, but produces broken verilog because it used a broken cached result from the failing run.

That's really bad..

martijnbastiaan, 2019-06-14:
Second workaround, use:

```haskell
    (c0 :> c1 :> c2 :> c3 :> c4 :> c5 :> c6 :> c7 :> Nil) = consts
```

and

```haskell
    = fmap (step 14 $ c7)
    $ register undefined
    $ fmap (step 12 $ c6)
    $ register undefined
    $ fmap (step 10 $ c5)
    $ register undefined
    $ fmap (step 8  $ c4)
    $ register undefined
    $ fmap (step 6  $ c3)
    $ register undefined
    $ fmap (step 4  $ c2)
    $ register undefined
    $ fmap (step 2  $ c1)
    $ register undefined
    $ fmap (step 0  $ c0)
```

PR #648 (2019-06-26): unconcat template needs to reference all arguments
Otherwise, `removeUnusedArguments` removes the arguments and the
compile-time evaluator can no longer fire because it's lacking
the constants.

Fixes #637
commit 89c0cbba (2019-06-26): unconcat template needs to reference all arguments

Otherwise, `removeUnusedArguments` removes the arguments and the
compile-time evaluator can no longer fire because it's lacking
the constants.

Fixes #637
closed 2019-06-26 by christiaanb