clash-lang/clash-compiler #647: SResetPolarity not exported by Prelude

gergoerdi, 2019-06-25:
I have to manually

```
import Clash.Signal.Internal (SResetPolarity(..))
```

to be able to define my own `KnownDomain`s. That is probably not intended.

martijnbastiaan, 2019-06-25:
Hmm, I'm not sure. I'd like people to use `createDomain`, as its much more ergonomic than creating your own instances. See:

https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/src/Clash/Signal/Internal.hs#L394

The docs at

https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/src/Clash/Signal.hs#L46-L49

could probably mention this.

martijnbastiaan, 2019-06-25:
Actually, it should be exported as it has a valid usecase in combination with `knownDomain` - just like the other GADTs are. I'll merge a patch soon.

commit 32d662c4 (2019-06-27): Export 'SResetPolarity' from "Clash.Signal" too. Fix #647
commit 0038a0b7 (2019-06-27): Export 'SResetPolarity' from "Clash.Signal" too. Fix #647
commit d5fd3a81 (2019-06-27): Export 'SResetPolarity' from "Clash.Signal" too. Fix #647
commit 017eb903 (2019-06-27): Export 'SResetPolarity' from "Clash.Signal" too. Fix #647
closed 2019-06-27 by martijnbastiaan