clash-lang/clash-compiler #656: Refutable pattern matching gives different result than `guard`

gergoerdi, 2019-06-30:
Unfortunately I don't have a repro for this yet, but I have noticed yesterday that the following code inside the definition of ` Signal _ (Maybe _)` :

```
    vidAddr = do
        x <- vgaX'
        y <- vgaY'
        pure $ do
            (True, x) <- x
            (_, y) <- y
            (addr, 0) <- return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3))
            return addr
```

this doesn't seem to work as expected; in particular, the refutable pattern `(addr,0)` doesn't become `Nothing` when the last 3 bits of `x` is not `0 :: Unsigned 3`.

However, the long-hand form

```
    vidAddr = do
        x <- vgaX'
        y <- vgaY'
        pure $ do
            (newPixel, x) <- x
            (_, y) <- y
            guard newPixel
            let (addr, idx) = bitCoerce (y, x) :: (Index VidSize, Unsigned 3)
            guard $ idx == 0
            return addr
```

works exactly as expected, i.e. the `guard`s will cause the signal to take on the value of `Nothing` unless `idx` is 0 etc.

leonschoorl, 2019-07-01:
I tried with:
```haskell
f :: Maybe (Unsigned 8, Bool) -> Maybe (Unsigned 8)
f inp = do
  (i,True) <- inp
  return i

g :: Maybe (Unsigned 8, Bool) -> Maybe (Unsigned 8)
g inp = do
  (i,b) <- inp
  guard b
  return i
```

But both just return `Nothing` when the Bool is `False`

christiaanb, 2019-07-02:
Using:

```haskell
{-# LANGUAGE ApplicativeDo #-}

module Test where

import Clash.Prelude

type VidSize = 32

vidAddr
  :: Signal dom (Maybe (Bool, Unsigned 5))
  -> Signal dom (Maybe (Bool, Index 6))
  -> Signal dom (Maybe (Index VidSize))
vidAddr vgaX' vgaY' = do
        x <- vgaX'
        y <- vgaY'
        pure $ do
            (True, x) <- x
            (_, y) <- y
            (addr, 0) <- return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3))
            return addr

vgaX0 = pure (Just (True, 8 :: Unsigned 5))
vgaX1 = pure (Just (True, 9 :: Unsigned 5))
vgaY0 = pure (Just (False, 4 :: Index 6))
```

I get:

```
christiaan@vm:~/devel/clash-compiler$ cabal new-run clashi -- Test.hs
Up to date
Clashi, version 0.9999 (using clash-lib, version 0.9999):
http://www.clash-lang.org/  :? for help
Loaded package environment from /home/christiaan/devel/clash-compiler/.ghc.environment.x86_64-linux-8.6.5
[1 of 1] Compiling Test             ( Test.hs, interpreted )
Ok, one module loaded.
*Test> sampleN @System 1 (vidAddr vgaX0 vgaY0)
[Just 17]
*Test> sampleN @System 1 (vidAddr vgaX1 vgaY0)
[Nothing]
*Test> 
```

gergoerdi, 2019-07-02:
OK so let's try three versions, I think all three should behave the same:

```
{-# LANGUAGE ApplicativeDo #-}
import Clash.Prelude
import Control.Monad (guard)

type VidX = 256
type VidY = 224
type VidSize = VidX * VidY `Div` 8

vgaX' = fromList [Just (True, i) | i <- [minBound..maxBound] ]
vgaY' = fromList [Just (True, i) | i <- [minBound..maxBound] ]

vidAddr1
    :: Signal System (Maybe (Bool, Index VidX))
    -> Signal System (Maybe (Bool, Index VidY))
    -> Signal System (Maybe (Index VidSize))
vidAddr1 vgaX' vgaY' = do
    x <- vgaX'
    y <- vgaY'
    pure $ do
        (newPixel, x) <- x
        (_, y) <- y
        guard newPixel
        (addr, 0) <- return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3))
        return addr

vidAddr2
    :: Signal System (Maybe (Bool, Index VidX))
    -> Signal System (Maybe (Bool, Index VidY))
    -> Signal System (Maybe (Index VidSize))
vidAddr2 vgaX' vgaY' = do
    x <- vgaX'
    y <- vgaY'
    pure $ do
        (True, x) <- x
        (_, y) <- y
        (addr, 0) <- return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3))
        return addr

vidAddr3
    :: Signal System (Maybe (Bool, Index VidX))
    -> Signal System (Maybe (Bool, Index VidY))
    -> Signal System (Maybe (Index VidSize))
vidAddr3 vgaX' vgaY' = do
    x <- vgaX'
    y <- vgaY'
    pure $ do
        (True, x) <- x
        (_, y) <- y
        (addr, idx) <- return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3))
        guard $ idx == 0
        return addr
```

```
*Main> sampleN 10 $ vidAddr1 vgaX' vgaY'
[Just 0*** Exception: Foo.hs:21:9-19: Non-exhaustive patterns in lambda
*Main> sampleN 10 $ vidAddr2 vgaX' vgaY'
[Just 0,Nothing,Nothing,Nothing,Nothing,Nothing,Nothing,Nothing,Just 257,Nothing]
*Main> sampleN 10 $ vidAddr3 vgaX' vgaY'
[Just 0,Nothing,Nothing,Nothing,Nothing,Nothing,Nothing,Nothing,Just 257,Nothing]
```

gergoerdi, 2019-07-02:
In fact, changing `vidAddr1` just slightly already seems to fix it:

```
vidAddr1'
    :: Signal System (Maybe (Bool, Index VidX))
    -> Signal System (Maybe (Bool, Index VidY))
    -> Signal System (Maybe (Index VidSize))
vidAddr1' vgaX' vgaY' = do
    x <- vgaX'
    y <- vgaY'
    pure $ do
        (newPixel, x) <- x
        guard newPixel
        (_, y) <- y
        (addr, 0) <- return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3))
        return addr
```

```
*Main> sampleN 10 $ vidAddr1' vgaX' vgaY'
[Just 0,Nothing,Nothing,Nothing,Nothing,Nothing,Nothing,Nothing,Just 257,Nothing]
```

leonschoorl, 2019-07-02:
> `*** Exception: Foo.hs:21:9-19: Non-exhaustive patterns in lambda`

Could the be a bug in GHC?
Because the pattern at that location `(_, y) <-` shouldn't fail.
If any pattern fails it should be the `(addr, 0)` that fail, right?

When I manually desugar the inner do block to:
```haskell
vidAddr1''
    :: Signal System (Maybe (Bool, Index VidX))
    -> Signal System (Maybe (Bool, Index VidY))
    -> Signal System (Maybe (Index VidSize))
vidAddr1'' vgaX' vgaY' = do
    x <- vgaX'
    y <- vgaY'
    pure $ do
        x >>= \(newPixel, x) ->
         y >>= \ (_, y) ->
          guard newPixel >>
           return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3)) >>= \(addr, 0) ->
            return addr
```
It does report the error on the `\(addr, 0) -> ...`.

leonschoorl, 2019-07-02:
But if we want to get `Nothing` when the pattern fails, we should desugar it like:
```haskell
vidAddr1''' vgaX' vgaY' = do
    x <- vgaX'
    y <- vgaY'
    pure $
        x >>= \(newPixel, x) ->
         y >>= \ (_, y) ->
          guard newPixel >>
           return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3)) >>= \case
             (addr, 0) -> return addr
             _         -> fail "pattern failed"
```

christiaanb, 2019-07-02:
Seems like a bug in `ApplicativeDo` desugaring, because if I do:

```haskell
{-# LANGUAGE ApplicativeDo #-}
import Clash.Prelude
import Control.Monad (guard)

type VidX = 256
type VidY = 224
type VidSize = VidX * VidY `Div` 8

vgaX' = fromList [Just (True, i) | i <- [minBound..maxBound] ]
vgaY' = fromList [Just (True, i) | i <- [minBound..maxBound] ]

vidAddr1
    :: (Maybe (Bool, Index VidX))
    -> (Maybe (Bool, Index VidY))
    -> (Maybe (Index VidSize))
vidAddr1 x y = do
        (newPixel, x) <- x
        (_, y) <- y
        guard newPixel
        (addr, 0) <- return (bitCoerce (y, x) :: (Index VidSize, Unsigned 3))
        return addr
```

with the `ApplicativeDo` pragma as enabled I get:

```
*Main> vidAddr1 <$> vgaX' <*> vgaY'
Just 0 Just *** Exception: Test.hs:18:9-19: Non-exhaustive patterns in lambda
```

however, when I remove the `ApplicativeDo` pragma I get:

```
*Main> vidAddr1 <$> vgaX' <*> vgaY'
Just 0 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 257 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 514 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 771 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 1028 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 1285 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 1542 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 1799 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 2056 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 2313 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 2570 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 2827 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 3084 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 3341 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 3598 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 3855 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 4112 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 4369 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 4626 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 4883 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 5140 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 5397 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 5654 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 5911 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 6168 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 6425 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 6682 Nothing Nothing Nothing Nothing Nothing Nothing Nothing Just 6939 Nothing Nothing Nothing Nothing Nothing Nothing Nothing *** Exception: X: finite list
CallStack (from HasCallStack):
  errorX, called at src/Clash/Signal/Internal.hs:1269:57 in clash-prelude-0.9999-inplace:Clash.Signal.Internal
```

leonschoorl, 2019-07-03:
It's this issue in GHC: https://gitlab.haskell.org/ghc/ghc/issues/15344