clash-lang/clash-compiler #664: Merge "Clash.Prelude.{BitIndex,BitReduction}" into "Clash.Sized.BitVector"

martijnbastiaan, 2019-07-04:
Functions such as `msb`, `lsb`, `reduceAnd`, etc. are hard to discover as recently discussed on the mailing list. Most users (including yours truly, who was blissfully unaware of the modules these functions reside in) will look in `Clash.Sized.BitVector`. Worse, some users have tried to use the primitives `msb#`, etc. in `Clash.Sized.Internal.BitVector` because they couldn't find the others!

We should therefore get rid of `Clash.Prelude.{BitIndex,BitReduction}` and merge them into  `Clash.Sized.Internal.BitVector`.

martijnbastiaan, 2019-08-27:
I should have rephrased this issue after having had a discussion about it with @leonschoorl. What it comes down to is the fact that `msb#` and friends are hard to discover. So it may not be necessary to move these modules, as long as they are re-exported at relevant places. Leon has put in some thought on how to best do that.

Still:

> how about moving them to `Clash.Class.BitPack`?

makes sense to me.

christiaanb, 2019-08-27:
@martijnbastiaan But those operations work on more types than `BitVector`, how about moving them to `Clash.Class.BitPack`?

alex-mckenna, 2021-06-03:
Closing: this was done in https://github.com/clash-lang/clash-compiler/commit/2b19b17bf929cd20556ff01c9f1c9791bc3213a8

closed 2021-06-03 by alex-mckenna