clash-lang/clash-compiler #678: Implement Bits instance for Index type

rowanG077, 2019-07-22:
I was refactoring some code and changing some Unsigned I'm using to Index since they better represent the use case. I'm using `shiftR x 1` as non-costly way to halve one of these unsigneds. But `Bits` is class is not implemented for index.

Is there any specific reason I'm overlooking why the `Bits` is not implemented for index?

leonschoorl, 2019-07-23:
Compared to other number types, `Index n` is somewhat special.

When `n` is not a power of 2 there exists bitpatterns which are not valid Index values.
In Haskell we check for these for out of range values and notify the user by throwing an exception:
```
Clash.Prelude> 9 + 1 :: Index 10
*** Exception: X: Clash.Sized.Index: result 10 is out of bounds: [0..9]
```
But in HDL there no such checks are done. You'll just get 10.

While this is a problem for arithmetic operations.
At least it's relatively easy to understand that `x + 9 :: Index n` is only  safe when `x <= (maxBound - 9)`. Also we have thing like [saturating arithmetic] to help.

But with bitwise operations it's very easy to get out of range.
When would it be safe to do `x .|. 9 :: Index n`?

A reason for not having an instance `Bits (Index n)` could be to discourage users from doing bitwise operation on Index values. 
Thereby hinting that maybe they'd better so another type like Unsigned if they're doing bitwise operations.

Of course it is perfectly safe to do `shiftR x 1 :: Index n`.
But I'm pretty sure you can rely on (almost all?) synthesis software to synthesize `shiftR x 1` and 
``x `div` 2`` to the same hardware implementation of a shift.

[Saturating arithmetic]: http://hackage.haskell.org/package/clash-prelude-0.99.3/docs/Clash-Class-Num.html#g:2

rowanG077, 2019-07-23:
Yeah you can get out of bounds very easily with `Index`. But I don't feel this is different then getting out of bounds with any other mathematical operation. Once you start doing bit operations you have to understand anyway what the underlying bit representation is. `Signed` has a `Bits` instance as well and there you also can get very weird result if you don't understand what is going on under the hood.

I'm not particularly married to having a bits instance if division will also work for my case. However I do think having the same API for `Unsigned` and `Index` would make sense because fundamentally they both are just unsigned number.

martijnbastiaan, 2019-07-23:
I agree with @rowanG077 here. Sure it's easy to define a bit-level operation that will result in an out-of-bound number, but that's true for nearly all operations defined for `Index`. In fact, other than saturating/wrapping ops I can't think of any that I regularly use that can't go out of bounds. Unless we're willing to remove all unsafe operations, it makes no sense to me to discourage bitwise ones.

martijnbastiaan, 2019-07-24:
Fixed by merging  #679. Thanks @rowanG077 !

closed 2019-07-24 by martijnbastiaan