clash-lang/clash-compiler #680: Make all operation on Index type safe

rowanG077, 2019-07-24:
Based on discussion in https://github.com/clash-lang/clash-compiler/pull/679 and https://github.com/clash-lang/clash-compiler/issues/678.

> We did have some internal discussion about this PR. Ideally, we would like to move to a situation where all operations on Index are safe. Unfortunately, Haskell legacy stands in our way (most notably, we have to implement Num and its unsafe operations to get index literals), so we can't really offer a nice API. We decided that the next best thing was to unify the API of Unsigned and Index, until we can implement it the "proper" way.

Currently the `Num` instance and `Bits` instance are unsafe for the `Index` type. Each operation on a value of type `Index n` is checked during simulation to ensure that the result will fit inside that index. If the operation results in a value that does not fit in the `Index n` type an exception is thrown.

The HDL implementation no such checking occurs which means simulation and the resulting HDL do not have the same functionality.

I would like to use this issue to discuss possible remedies. I'm not really that experienced with Haskell so correct me if I'm wrong but wouldn't it be possible to implement `Num` in terms of `SaturatingNum` with `SatWrap` as `SaturationMode`?

The `Bits` instance is a different beast where I will have to think a little harder.

martijnbastiaan, 2019-07-25:
> wouldn't it be possible to implement Num in terms of SaturatingNum with SatWrap as SaturationMode?

That would work I think.

> That would match the Unsigned behaviour but has additional cost in hardware since more logic is required for the wrapping behaviour(? I think, didn't verify).

Yes!

> The Bits instance is a different beast where I will have to think a little harder.

Wrapping may be a bit costly. I'd imagine you'd do the operations (internally) on an unsigned and return `result % maxBound`. Saturating would be much less costly, as you don't have to do modulo operations.

-------------

With that being said, I'm not a big fan of picking a default. I'd much rather see something like:

```haskell
newtype Index (sat :: SaturationMode) (n :: Nat) = ..
```

while maybe even extending `SaturationMode` with an `Unsafe` option, to replicate the current behavior. We would then make a few different instances for `Num`:

```haskell
instance KnownNat n => Num (Index 'SatWrap n)
instance KnownNat n => Num (Index 'SatBound n)
...
```

We could document the tradeoffs between them, and let the user pick a compromise.

leonschoorl, 2019-07-26:
If we would add an unsafe SaturationMode we might keep backwards compatibility with something like:
```haskell
newtype SafeIndex (sat :: SaturationMode) (n :: Nat) = ...
type Index n = SafeIndex 'SatUnsafe n
```

rowanG077, 2019-07-26:
@leonschoorl Yeah that is also an option. The thing to consider is do we want to introduce legacy API like this before the 1.0 release?

rowanG077, 2022-02-16:
Since this is my issue and this has been fixed now I mark it as closed

closed 2022-02-16 by rowanG077