clash-lang/clash-compiler #691: Promoted Nat inlining extremely slow?

mt-caret, 2019-08-01:
Hi, I've been working on a nice way to initialize registers in a similar way to `blockRamFile`, and I've got something working with TemplateHaskell by compiling listToVecTH and initMem:
```haskell
{-# Language AllowAmbiguousTypes #-}
module CreateVecFileTH (vecFromFileTH) where

import qualified Prelude as P
import Clash.Prelude
import Clash.Explicit.BlockRam.File (initMem)
import System.IO.Unsafe (unsafePerformIO)
import Language.Haskell.TH (ExpQ)

initMem' :: KnownNat n => SNat n -> FilePath -> IO [BitVector n]
initMem' _ = initMem

vecFromFileTH :: (KnownNat height, KnownNat width) => SNat height -> SNat width -> FilePath -> ExpQ
vecFromFileTH height width filepath =
    listToVecTH $ contents <> (P.replicate (height' - P.length contents) 0)
  where
    height' = snatToNum height
    contents = unsafePerformIO $ initMem' width filepath
```

This works perfectly fine except for the fact that it's TemplateHaskell, which means messing up the arguments/signatures when calling `vecFromFileTH' results in a horrific unreadable error message. So, I tried implemented it without TemplateHaskell and this is what I got:

```haskell
listToVecU :: Lift a => UNat n -> a -> [a] -> Vec n a
listToVecU UZero _ _ = Nil
listToVecU n x' [] = replicate (fromUNat n) x'
listToVecU (USucc s) x' (x:xs) = x :> (listToVecU s x' xs)

listToVecWithDefault :: Lift a => SNat n -> a -> [a] -> Vec n a
listToVecWithDefault = listToVecU . toUNat

listToVec :: (Lift a, Default a) => SNat n -> [a] -> Vec n a
listToVec n xs = listToVecWithDefault n def xs

vecFromFile :: (KnownNat height, KnownNat width) => FilePath -> Vec height (BitVector width)
vecFromFile filepath =
    unsafePerformIO $ withSNat listToVec <$> initMem filepath
```

Everything typechecks, but when CλaSH tries to synthesize the circuit containing a call to `vecFromFile`  I get the following message repeatedly:
```
Type of the subject is: Clash.Promoted.Nat.UNat8214565720323789186 64
Function Clash.Promoted.Nat.fromUNat585480 will not reach a normal form, and compilation might fail.
Run with '-fclash-inline-limit=N' to increase the inlining limit to N.
```
Passing `-flash-inline-limit=10000` to CλaSH seems to pacify it, but then it essentially hangs (I've been running it for a while, and top tells me it's doing something, but I've yet to see it complete execution). Interestingly, explicitly running this at compile-time using TemplateHaskell (i.e. `$(lift $ (vecFromFile "./memfile.bin" :: Vec 64 (BitVector 32)))`  instead of `vecFromFile "./memfile.bin"`) finishes promptly and gives me sane Verilog. My question is 1. is all of this expected behavior and 2. would you be interested in a PR packaging one and/or the other implementation into Clash.Prelude?

mt-caret, 2019-08-01:
After running a while, compilation of `vecFromFile "./memfile.bin"` without TemplateHaskell suddenly started to eat ~80% of memory on my 16GB memory notebook and I was forced to stop it.

martijnbastiaan, 2019-08-01:
I don't think Clash will ever try to translate `unsafePerformIO`. @christiaanb is working on / thinking about adding a feature that will let us do IO. That feature would let us write the equivalent of Verilog's `$fgetc`. My guess is that would help out in this situation too. Maybe @christiaanb could weigh in?

mt-caret, 2019-08-01:
`unsafePerformIO` shouldn't be the culprit here, since `vecFromFile` isn't doing anything really different from what `blockRamFile` (https://github.com/clash-lang/clash-compiler/blob/master/clash-prelude/src/Clash/Explicit/BlockRam/File.hs#L267) is doing, right?

christiaanb, 2019-08-01:
Clash doesn't look inside of the definition of `blockRamFile` as it is considered a primitive. It's considered a primitive because there's a BlackBox definition: https://github.com/clash-lang/clash-compiler/blob/35b97b0a1efd5274df3d8302077922fb9de50977/clash-lib/prims/verilog/Clash_Explicit_BlockRam_File.json#L1-L50. That's why  `blockRamFile` can call `unsafePerformIO` and never throws Clash into a spin.

mt-caret, 2019-08-01:
Ah, I see; that makes a lot of sense. So currently the only way to do the above is to do it at compile time with TemplateHaskell?

christiaanb, 2019-08-01:
As to the other point @martijnbastiaan was referring to: we will shortly be working on translating a subset of Haskell I/O to Verilog I/O, mostly for the purpose of synthesizable test benches. But only synthesizable insofar that Clash outputs Verilog, _not_ that a downstream synthesis tool can take that particular piece of Verilog I/O code and turn it into a working FPGA bitstream or ASIC.

christiaanb, 2019-08-01:
@mt-caret: and finally, for your use case: you can also make a new primitive! Simply take the current blackBox for `delay#`: https://github.com/clash-lang/clash-compiler/blob/35b97b0a1efd5274df3d8302077922fb9de50977/clash-lib/prims/verilog/Clash_Signal_Internal.json#L1-L27

but use the logic from `blockRamFile#` to initialize it with a File: https://github.com/clash-lang/clash-compiler/blob/35b97b0a1efd5274df3d8302077922fb9de50977/clash-lib/prims/verilog/Clash_Explicit_BlockRam_File.json#L21-L25

The tutorial describes where Clash looks for blackbox definitions: http://hackage.haskell.org/package/clash-prelude-0.99.3/docs/Clash-Tutorial.html#g:14

mt-caret, 2019-08-01:
Thanks for the info & help! I'll take a look. Feel free to close this issue.

closed 2019-08-01 by martijnbastiaan