clash-lang/clash-compiler #693: CircleCI/Clash regularly runs out of memory (`NumConstantFoldingTB`)

martijnbastiaan, 2019-08-01:
.. this most recently happened on [this build](https://circleci.com/gh/clash-lang/clash-compiler/4751?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link). We should either:

* Look into reducing Clash's memory consumption
* Split up `NumConstantFoldingTB`

leonschoorl, 2019-08-01:
Right now compiling `NumConstantFoldingTB` to VHDL uses about 3.7GB on my machine.
Which is close to the 4Gb available on Circle instances.
```
GHC: Parsing and optimising modules took: 51.156s
GHC: Loading external modules from interface files took: 3.569s
GHC: Parsing annotations took: 0.008s
GHC+Clash: Loading modules cumulatively took 1m2.956s
Clash: Parsing and compiling primitives took 0.605s

Clash: Compiling NumConstantFoldingTB.topEntity
Clash: Applied 21535 transformations
Clash: Normalisation took 2.551s
Clash: Netlist generation took 0.027s

Clash: Compiling NumConstantFoldingTB.testBench
Clash: Applied 28132 transformations
Clash: Testbench normalisation took 3.843s
Clash: Testbench netlist generation took 0.020s
Clash: Total compilation took 1m9.690s
```
Most of the time is spend in `GHC: Parsing and optimising modules`.
But this only takes about 1.6GB.
When we've done `GHC+Clash: Loading modules cumulatively` the memory has almost doubled to about 2.9GB.

This is just a guess, but maybe the GHC core representation is about 1.5Gb, and which gets translated to a clash core representation of about the same size and the GHC core is retained.
If this is the case maybe we could do something that would allow the GHC core to be garbage collected and save the 1.5Gb.

martijnbastiaan, 2019-08-01:
Only tangentially related to this issue report, but I noticed some interesting behavior today: test statuses are commit based. Therefore, if you clone an external PR and push it to a branch owned by `clash-lang` GitLab CI will run it and report on its status. GitHub shows this in the original PR! We could therefore easily make a small script that "approves" a certain PR:

* `./approve_pr 999`
  * Fetch `https://api.github.com/repos/clash-lang/clash-compiler/pulls/999.json`
  * Clone external PR
  * Push to branch `_test_pr_999`

With maybe a days worth of engineering we could probably get it to poll regularly, have trusted users, and garbage collect.

martijnbastiaan, 2019-08-01:
> This is just a guess, but maybe the GHC core representation is about 1.5Gb, and which gets translated to a clash core representation of about the same size and the GHC core is retained.

Sounds like a promising lead!

mmzx, 2019-08-16:
After splitting up `NumConstantFoldingTB` to two parts I have made some experimental tests with some sort of results, that are:
Executing the _testsuite_ on a machine which has 4G of memory, it was possible to have all tests successfully executed by limiting the amount of CPUs to 5.
Executing the testsuite with more then 5 threads increases overall memory consumption that causes `NumConstantFoldingTB`s to start failing again. Other tests are still fine.
Since I did not find these data so far, I thought it may be relevant at some point later.

martijnbastiaan, 2021-03-19:
Memory optimizations accumulated in master and setting CircleCI to only use one thread have solved/mitigated this issue.

closed 2021-03-19 by martijnbastiaan