clash-lang/clash-compiler #700: "specializing a function on a possibly non work-free cast"

gergoerdi, 2019-08-08:
With `09c8aa49aa2aa80b638a565c471ab05d853d98be` I am getting the following  output thousands and thousands of times:

> WARNING: Clash.Normalize.Transformations(852): specializing a function on a possibly non work-free cast.
> Generated HDL implementation might contain duplicate work.
> Please report this as a bug.
> 
> Expression where this occurs:
> ```
> GHC.Integer.Type.plusInteger
>   ((Clash.Sized.Internal.Index.fromInteger# @5 5
>       0) ▷
>     Clash.Sized.Internal.Index.Index8214565720323789241
>       5
>     ~
>     GHC.Integer.Type.Integer3674937295934324780)
> ``` 

Since it doesn't seem to be doing much more than just repeatedly printing this, I don't know if it's worth waiting for it or if it'll just keep looping (and eating up memory).

gergoerdi, 2019-08-09:
I don't have a fully standalone way of reproducing this, but I have located that the above error is caused by this part of my code:

```
data S = MkS
    { timers :: Vec 5 (Unsigned 8)
    }
    deriving (Show, Generic, Undefined)

foo :: State S ()
foo = mapM_ bar [0..4]

bar :: Index 5 -> State S ()
bar i = modify $ \s -> s{ timers = replace i 42 $ timers s }
```

gergoerdi, 2019-08-09:
This workaround seems to work:

```
foo :: State S ()
foo = mapM_ (bar . fromIntegral) [0..4]
```

martijnbastiaan, 2019-10-16:
@gergoerdi Sorry for the very slow response. I think you can safely ignore the warning, as the expression it warns about is indeed work-free. I'll fix it in a bit.

commit 11bef739 (2019-10-16): Check work content before warning in 'argCastSpec'. Fix #700
commit e9e9dd4f (2019-10-16): Check work content before warning in 'argCastSpec'. Fix #700
commit 62fcbf42 (2019-10-17): Check work content before warning in 'argCastSpec'. Fix #700
commit 11d69542 (2019-10-17): Check work content before warning in 'argCastSpec'. Fix #700